RinCat/RTL88x2BU-Linux-Driver #89: module parameter documentation

beroal, 2021-05-25:
Hi. Thank you for supporting this driver. It'd be great to include a documentation for some module parameters; see the [documentation for an alternative driver by `morrownr`](https://github.com/morrownr/88x2bu/blob/5.8.7.4/88x2bu.conf). I personally tried to enable a 80 MHz channel in the AP mode in the 5 GHz band for several months before discovering the parameter `rtw_vht_enable=2`. Other people may benefit from the knowledge of other parameters. Some parameters are documented in `/os_dep/linux/os_intfs.c` in the source code, so you can insert the documentation there. It's where I found the documentation for `rtw_dfs_region_domain`, for example.

First, I installed `morrownr`'s driver, but then I noticed that `rtw_vht_enable` works with your driver. I prefer your driver since you created an ArchLinux package. Thank you again.

I tested your driver with kernel 5.12.6, so you can update `README.md`.

RinCat, 2021-05-25:
I may add a few parameters docs, but you list them by using `modinfo 88x2bu`.

beroal, 2021-05-26:
> I may add a few parameters docs, but you can list them by using `modinfo 88x2bu`.

This is what I've done, but parameter names without docs aren't really useful. ☹