RinCat/RTL88x2BU-Linux-Driver #90: set_wiphy_netns is missing.

yananet, 2021-05-26:
Hello.

The driver works fine, but `set_wiphy_netns` is missing.
```
$ iw list
	Supported commands:
	...
		 * set_wiphy_netns
	...
```
I have one rtl8812 Wi-Fi, I could see this with with https://github.com/cilynx/rtl88x2bu,
however, in this driver, monitor mode and MIMO are missing...

Can you please let me know how I can enable this?

RinCat, 2021-05-26:
NS is not support in the official driver, I need add a patch.

RinCat, 2021-05-26:
Try the latest version.

yananet, 2021-05-27:
What do you mean by the latest version?

FYI,
I also issued this for 8814au as well...
https://github.com/morrownr/8814au/issues/23

RinCat, 2021-05-28:
git pull or download this driver again and install.

yananet, 2021-05-28:
It works! Thank you!

closed 2021-05-28 by yananet