RinCat/RTL88x2BU-Linux-Driver #94: Fail to compile on kernel 5.3.18-59

nagius, 2021-07-11:
Running OpenSuse Leap 15.3 (kernel 5.3.18-59.10-default), I got these build errors with current HEAD (d547cb6):

```
/usr/src/rtl88x2bu-git/os_dep/linux/ioctl_cfg80211.c:10246:3: error: ‘struct cfg80211_ops’ has no member named ‘mgmt_frame_register’
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
   ^~~~~~~~~~~~~~~~~~~
/usr/src/rtl88x2bu-git/os_dep/linux/ioctl_cfg80211.c:10246:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/rtl88x2bu-git/os_dep/linux/ioctl_cfg80211.c:10246:25: note: (near initialization for ‘rtw_cfg80211_ops.mgmt_tx_cancel_wait’)
cc1: some warnings being treated as errors
```

Look like there was some API changes in kernel version 5.8, they might have been backported by OpenSuse in earlier kernel versions.
There is similar reported issue with other Realtek drivers:

https://github.com/aircrack-ng/rtl8812au/issues/873
https://www.spinics.net/lists/linux-wireless/msg197852.html
https://bugs.launchpad.net/ubuntu/+source/rtl8812au/+bug/1900180

nagius, 2021-07-11:
I came up with a workaround that seems to work :

```patch
diff --git a/os_dep/linux/ioctl_cfg80211.c b/os_dep/linux/ioctl_cfg80211.c
index be2308b..0b9ed7e 100644
--- a/os_dep/linux/ioctl_cfg80211.c
+++ b/os_dep/linux/ioctl_cfg80211.c
@@ -7777,7 +7777,7 @@ exit:
    return ret;
 }
 
-#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 8, 0))
+#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 3, 0))
 static void cfg80211_rtw_mgmt_frame_register(struct wiphy *wiphy,
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 6, 0))
    struct wireless_dev *wdev,
@@ -7797,7 +7797,7 @@ static void cfg80211_rtw_update_mgmt_frame_register(struct wiphy *wiphy,
    _adapter *adapter;
 
    struct rtw_wdev_priv *pwdev_priv;
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 3, 0))
    u32 rtw_mask;
 #endif
 
@@ -7808,7 +7808,7 @@ static void cfg80211_rtw_update_mgmt_frame_register(struct wiphy *wiphy,
    pwdev_priv = adapter_wdev_data(adapter);

 #ifdef CONFIG_DEBUG_CFG80211
-#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 8, 0))
+#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 3, 0))
    RTW_INFO(FUNC_ADPT_FMT" frame_type:%x, reg:%d\n", FUNC_ADPT_ARG(adapter),
        frame_type, reg);
 #else
@@ -7817,7 +7817,7 @@ static void cfg80211_rtw_update_mgmt_frame_register(struct wiphy *wiphy,
 #endif
 #endif
 
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 3, 0))
    rtw_mask = BIT(IEEE80211_STYPE_AUTH >> 4);
    if ((upd->interface_stypes & rtw_mask)
        == (pwdev_priv->mgmt_mask & rtw_mask))
@@ -10240,7 +10240,7 @@ static struct cfg80211_ops rtw_cfg80211_ops = {
    .action = cfg80211_rtw_mgmt_tx,
 #endif
 
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 3, 0))
    .update_mgmt_frame_registrations = cfg80211_rtw_update_mgmt_frame_register,
 #elif (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 37)) || defined(COMPAT_KERNEL_RELEASE)
    .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
diff --git a/os_dep/linux/ioctl_cfg80211.h b/os_dep/linux/ioctl_cfg80211.h
index 9ab050c..06af06f 100644
--- a/os_dep/linux/ioctl_cfg80211.h
+++ b/os_dep/linux/ioctl_cfg80211.h
@@ -175,7 +175,7 @@ struct rtw_wdev_priv {
    u16 mgmt_tx_cookie;

    _mutex roch_mutex;
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 3, 0))
    u32 mgmt_mask;
 #endif
```

Obviously, this is a quick and dirty patch; and should not be merged. But that could unblock some people.

RinCat, 2021-07-12:
Sadly, I cannot do much if some distributions backport it.

boogiefromzk, 2021-07-29:
Thank you, this worked for me.

KierPrev, 2021-11-05:
Is there a way to make this work with 5.3 and 5.8?
Just to make a driver that works through every kernel.

RinCat, 2021-11-05:
This driver should works through every official kernels, but most users will use distribution modified kernels which I cannot do much about that.

KierPrev, 2021-11-05:
I just asked because I have exactly the same error on this driver https://github.com/McMCCRU/rtl8188gu  only in RHEL/openSUSE distributions, working on Arch/Debian, and I'm trying to fix it.

Thanks anyway!

RinCat, 2021-11-06:
You just need to change the kernel version macro to your kernel version.

schaefi, 2022-01-03:
@nagius thanks for providing the workaround patch. I stumbled over the same issue on Leap. It looks like with the current HEAD of the driver the patch can be reduced to:

```
diff --git a/os_dep/linux/ioctl_cfg80211.c b/os_dep/linux/ioctl_cfg80211.c
index cdd3043..4088338 100644
--- a/os_dep/linux/ioctl_cfg80211.c
+++ b/os_dep/linux/ioctl_cfg80211.c
@@ -8289,7 +8289,7 @@ exit:
        return ret;
 }
 
-#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 8, 0))
+#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 3, 0))
 static void cfg80211_rtw_mgmt_frame_register(struct wiphy *wiphy,
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 6, 0))
        struct wireless_dev *wdev,
@@ -10471,7 +10471,7 @@ static struct cfg80211_ops rtw_cfg80211_ops = {
 
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 37)) || defined(COMPAT_KERNEL_RELEASE)
        .mgmt_tx = cfg80211_rtw_mgmt_tx,
-#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 8, 0))
+#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 3, 0))
        .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
 #else
        .update_mgmt_frame_registrations = cfg80211_rtw_update_mgmt_frame_register,
```

This worked for me.

Thanks

5erotonin, 2022-03-18:
Just an FYI for anyone else trying to get this to work with openSUSE Leap 15.3 (or later) with a kernel lower than 5.8:  the newest version of the 8812au driver code referenced here has these if-else statements reversed and uses a >= comparator. Changing the 5,8,0 to 5,3,0 still works.