RinCat/RTL88x2BU-Linux-Driver #96: Can't connect to router

tomoverlund, 2021-07-20:
Product: Cudy WU1400
OS: Debian unstable (Linux debian 5.10.0-8-amd64 #1 SMP Debian 5.10.46-1 (2021-06-24) x86_64 GNU/Linux)
Router: Netgear R6080

Compiled and installed with dkms, no errors. Device is detected when plugged in and loads the module. But when I try to bring the network up with the "ifup" command, it cycles through six DHCPDISCOVERs before timing out.

I tried with both 5g and 2g bands, no luck. The same device works under Windows 10.

Log file with debugging turned on is attached. I inserted labels into the log to mark two sections, "plugin device:" and "ifup:"
[wifi_log.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/6850850/wifi_log.txt)

RinCat, 2021-07-21:
It looks like the configuration channel timeout. Would you like try use 40Mhz channel?

tomoverlund, 2021-07-21:
I tried both settings, rtw_vht_enable=0 and rtw_ht_enable=0, but it still doesn't work. Is there any other way to debug this?

RinCat, 2021-07-22:
I don't see any error here, would you like try use [iwd](https://wiki.debian.org/NetworkManager/iwd)?

tomoverlund, 2021-07-22:
I think the problem is, for some reason, the driver is very slow, causing timeouts. When I run a scan on my old, but working adapter, it finishes in under two seconds. This new adapter times out after 7 seconds:

root$ iw dev wlxb44bd624afa5 scan 
scan aborted!

I can see the device is doing it's thing in the logs, but it takes way too long. There's a bunch of commands where I'll see a gap of over 10ms, and those add up, for example:

Jul 20 15:13:55 debian kernel: [ 5671.404329] RTW: [HALMAC][TRACE]cfg_ch_bw_88xx <===
Jul 20 15:13:55 debian kernel: [ 5671.575367] RTW: [HALMAC][TRACE]chksum disable

RinCat, 2021-07-22:
Some gap over 10ms is common, I have the same things.
I can only confirm it successful connected to your router, but your system `wpa_supplicant` ask it to do a channel switch:
`wpa_supplicant[2531]: wlxb44bd624afa5: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5180 ht_enabled=1 ch_offset=1 ch_width=80 MHz cf1=5210 cf2=0`
and it timeout/failed for some reasons.

tomoverlund, 2021-07-22:
Do you have an example log of a working router connection with the current version that you can share? I still feel like this is related to things being too slow, especially considering the scan times out, which has nothing to do with connecting to the router or wpa_supplicant.

RinCat, 2021-07-22:
https://gist.github.com/RinCat/4c7354fc629c99f76520def4babbdaef

tomoverlund, 2021-07-22:
Thank you for the log. Comparing the two, my driver is running _extremely_  slow for some reason. Your entire process completes in under 3 seconds. My process eats up 3 seconds just getting to the "Download Firmware from array success" point.

Very weird. Clearly this is an issue specific to my end. I'll have to keep digging to see if I can pinpoint the trouble.  Thanks for your help.

tomoverlund, 2021-07-22:
By the way, could you tell me what kernel version you are using? If it's different, I could try using that to eliminate variables.

RinCat, 2021-07-22:
5.13.3 and 5.10.49.

tomoverlund, 2021-07-24:
I figured out the problem via the lsusb command. I was plugging the device into the front of my computer, which (unknown to me) shared the USB hub with a low-rate mouse and keyboard, even though those were plugged into the back. This worked fine with my old WiFi stick, but it made this new device interminably slow. Plugged it into an unshared port, and now it comes up in a jiffy. Thanks for your help, and sorry to waste your time on a non-coding issue.

closed 2021-07-24 by tomoverlund