RinCat/RTL88x2BU-Linux-Driver #98: airmon-ng will hang whole system

wkliang, 2021-07-26:
using airmon-ng command of aircrack-ng package will hang whole system and cpu usage keep going high. any clue?

# airmon-ng start wlan0

there is no message shown in log as I check

RinCat, 2021-07-26:
I just tested and did not see this issue.
```
phy1    wlp8s0f1u2      rtl88x2bu       Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
                (monitor mode enabled)
```

wkliang, 2021-07-26:
after following command
```
# airmon-ng start wlp0s18f2u4 

PHY	Interface	Driver		Chipset

phy0	wlp0s18f2u4	rtl88x2bu	Edimax Technology Co., Ltd Edimax AC1200 USB WiFi Adapter
```
running `ifconfig` and `iwconfig` will hang without good luck
cannot tell any useful message with `cat /proc/kmsg`
my linux is fedora running with kernel `5.13.4-200.fc34.x86_64`
is there any thing can help to clear up situation ? thanks in advance.

RinCat, 2021-07-27:
I had no idea, just tested `ifconfig` and `iwconfig`, both works fine under monitor mode. If there is no useful message in kmsg, maybe it is caused by other parts? like selinux etc.

wkliang, 2021-07-27:
I'd used another Edimax  product EW-7711UTn [Ralink RT2870] with airmon-ng without problem.
following is output of airmon-ng turnned on verbose and debug flag

```
# airmon-ng --verbose --debug start wlp0s18f2u4 

getStack mac80211
getBus usb
getdriver() rtl88x2bu
getchipset() Edimax Technology Co., Ltd Edimax AC1200 USB WiFi Adapter
BUS = usb
BUSINFO = 7392:F822
DEVICEID = 
getFrom() ?
getFirmware unavailable
?[phy0]wlp0s18f2u4	rtl88x2bu[mac80211]-unavailable	Edimax Technology Co., Ltd Edimax AC1200 USB WiFi Adapter	mode managed
```

RinCat, 2021-07-27:
I do have same logs but able to enable monitor mode without issues...

```
getStack mac80211
getBus usb
getdriver() rtl88x2bu
getchipset() Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
BUS = usb
BUSINFO = 0BDA:B812
DEVICEID = 
getFrom() ?
getFirmware unavailable
?[phy3]wlp8s0f1u2       rtl88x2bu[mac80211]-unavailable Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]                          mode managed
                (monitor mode enabled)
```

Caelestis94, 2021-08-11:
ill just chip in and say i having the same issues as this guy. System becomes unusable. 5.13.9 arch. My log is identical as the one you have @RinCat  (minus the monitor mode enabled)

Caelestis94, 2021-08-11:
Okay, downgrading to LTS 5.10.5 airmon-ng start wlan0 doesnt hangup the system anymore. So the freezing issue is most likely a kernel version problem.   Also only was successful in putting it to monitor mode with these : 
```
iwconfig wlan0
ifconfig wlan0 down
airmon-ng check kill
iwconfig wlan0 mode monitor
ifconfig wlan0 up
iwconfig wlan0
```

RinCat, 2021-08-12:
@Caelestis94 It works for me in 5.13.X, and it should support packet injections https://github.com/RinCat/RTL88x2BU-Linux-Driver/pull/51

wkliang, 2021-08-15:
@RinCat Are you using this device for packet monitoring and injection?

https://www.amazon.com/BrosTrend-1200Mbps-Linux-Adapter-Wi-Fi/dp/B07FCN6WGX

RinCat, 2021-08-15:
@wkliang I just tested it, and confirm both packet monitoring and injection are working for me.
```
> airmon-ng start wlp8s0f1u2

phy1    wlp8s0f1u2      rtl88x2bu       Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
                (monitor mode enabled)
```
```
> airodump-ng wlp8s0f1u2    

 CH  7 ][ Elapsed: 0 s ][ 2021-08-15 07:04 

 BSSID              PWR  Beacons    #Data, #/s  CH   MB   ENC CIPHER  AUTH ESSID
[classified]
```
```
> aireplay-ng -9 wlp8s0f1u2                                                                                                                                                                                                                                   

07:02:55  Trying broadcast probe requests...
07:02:57  Injection is working!
07:02:57  Found 7 APs

07:02:57  Trying directed probe requests...
07:02:57  [classified] - channel: 9 - '[classified]'
07:03:03  Ping (min/avg/max): 10.442ms/25.420ms/34.210ms Power: -72.50
07:03:03   4/30:  13%

.......
[classified]
```

wkliang, 2021-08-20:
got a new 'TP-Link Archer Ver.3` for testing,
whole system hung after running `airmon-ng start wlp...` just as Edimax AC1200

RinCat, 2021-08-20:
@wkliang do you able to get any kernel logs? I may try it in some live systems and see if I can reproducing it.

wkliang, 2021-09-01:
@RinCat  sorry for late reply

after running `airmon-ng check kill`, dmesg shown following:
```
[ 1053.442017] RTW: WARN _beamforming_leave: 40:9b:cd:a9:6b:5e is neither beamforming ee or er!!
[ 1053.442068] RTW: rtw_set_country_cmd country_code:"TW" mapping to chplan:0x76
[ 1053.442104] RTW: there is no any txpwr_regd
[ 1053.448432] RTW: WARN _beamforming_leave: 40:9b:cd:a9:6b:5e is neither beamforming ee or er!!
[ 1053.448441] RTW: ERROR Free disconnecting network of scanned_queue failed due to pwlan == NULL

[ 1053.451883] RTW: rtw_set_country_cmd country_code:"TW" mapping to chplan:0x76
[ 1053.451934] RTW: there is no any txpwr_regd
[ 1057.067432] RTW: nolinked power save enter
```
there is no further message emitted after running `airmon-ng start wlp0s18f2u4` 
in other window, running `strace ifconfig -a` shown
```
socket(AF_INET, SOCK_DGRAM, IPPROTO_IP) = 4
access("/proc/net/if_inet6", R_OK)      = 0
socket(AF_INET6, SOCK_DGRAM, IPPROTO_IP) = 5

/* some irrelevant messages deleted */

ioctl(4, SIOCGIFCONF, {ifc_len=1200 /* 30 * sizeof(struct ifreq) */
```
in other window, running `strace iwconfig` shown
```
socket(AF_INET, SOCK_DGRAM, IPPROTO_IP) = 3
brk(NULL)                               = 0x55b6e533a000
brk(0x55b6e535b000)                     = 0x55b6e535b000
openat(AT_FDCWD, "/proc/net/dev", O_RDONLY) = 4
newfstatat(4, "", {st_mode=S_IFREG|0444, st_size=0, ...}, AT_EMPTY_PATH) = 0
read(4, "Inter-|   Receive               "..., 1024) = 828
ioctl(3, SIOCGIWNAME
```
all commands hung on socket related operation and system cannot shutdown properly.

RinCat, 2021-09-01:
I cannot reproduce it in Debian 11, but able to do it in Arch. But since my Gentoo has same kernel version and it works fine, I am not sure what could cause it.

`Linux debian 5.10.0-8-amd64 #1 SMP Debian 5.10.46-4 (2021-08-03) x86_64 GNU/Linux`
`aircrack-ng  1:1.6+git20210130.91820bc-1`

```
root@debian:/home/user/RTL88x2BU-Linux-Driver# airmon-ng start wlx00e04c2948f5

PHY     Interface       Driver          Chipset
phy0    wlx00e04c2948f5 rtl88x2bu       Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
                (monitor mode enabled)

root@debian:/home/user/RTL88x2BU-Linux-Driver# iwconfig wlx00e04c2948f5
wlx00e04c2948f5  IEEE 802.11b  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Monitor  Frequency:2.457 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=-100 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

Arch:
```
[  736.965644] INFO: task iw:10568 blocked for more than 122 seconds.
[  736.965978]       Tainted: G           OE     5.13.13-arch1-1 #1
[  736.966310] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  736.966672] task:iw              state:D stack:    0 pid:10568 ppid: 10567 flags:0x00004000
[  736.966674] Call Trace:
[  736.966675]  __schedule+0x310/0x930
[  736.966677]  schedule+0x5b/0xc0
[  736.966679]  schedule_preempt_disabled+0x11/0x20
[  736.966680]  __mutex_lock.constprop.0+0x2f8/0x4e0
[  736.966683]  cfg80211_netdev_notifier_call+0x104/0x4f0 [cfg80211]
[  736.966709]  raw_notifier_call_chain+0x44/0x60
[  736.966711]  register_netdevice+0x4ee/0x5f0
[  736.966714]  cfg80211_rtw_add_virtual_intf+0x192/0x300 [88x2bu]
[  736.966783]  nl80211_new_interface+0x1b5/0x4b0 [cfg80211]
[  736.966815]  genl_family_rcv_msg_doit+0xfd/0x160
[  736.966818]  genl_rcv_msg+0xeb/0x1e0
[  736.966820]  ? nl80211_get_interface+0x90/0x90 [cfg80211]
[  736.966850]  ? genl_get_cmd+0xd0/0xd0
[  736.966852]  netlink_rcv_skb+0x5b/0x100
[  736.966854]  genl_rcv+0x24/0x40
[  736.966856]  netlink_unicast+0x23e/0x350
[  736.966858]  netlink_sendmsg+0x23a/0x470
[  736.966860]  ? __check_object_size+0x46/0x150
[  736.966862]  sock_sendmsg+0x5e/0x60
[  736.966864]  ____sys_sendmsg+0x258/0x2a0
[  736.966866]  ___sys_sendmsg+0xa3/0xf0
[  736.966870]  __sys_sendmsg+0x81/0xd0
[  736.966872]  do_syscall_64+0x61/0x80
[  736.966874]  ? handle_mm_fault+0xdb/0x2c0
[  736.966876]  ? do_user_addr_fault+0x1e8/0x690
[  736.966879]  ? do_syscall_64+0x6e/0x80
[  736.966880]  ? exc_page_fault+0x78/0x180
[  736.966882]  entry_SYSCALL_64_after_hwframe+0x44/0xae
[  736.966884] RIP: 0033:0x7f7c40a42cc7
[  736.966885] RSP: 002b:00007fffde235ab8 EFLAGS: 00000246 ORIG_RAX: 000000000000002e
[  736.966890] RAX: ffffffffffffffda RBX: 000055fa141e9390 RCX: 00007f7c40a42cc7
[  736.966891] RDX: 0000000000000000 RSI: 00007fffde235af0 RDI: 0000000000000003
[  736.966892] RBP: 000055fa141ee8c0 R08: 000055fa141e92a0 R09: 00007fffde235b5c
[  736.966893] R10: 00007fffde235dd8 R11: 0000000000000246 R12: 000055fa141ee780
[  736.966894] R13: 00007fffde235af0 R14: 000055fa141ee7d0 R15: 000055fa141ee8c0
```

RinCat, 2021-09-01:
The wiphy mutex lock is not control by the driver, so its something else hold it. I highly suspect it may be systemd or udev rules, as it still appears under a minimized arch live system.

RinCat, 2023-06-15:
Hi, someone has found a possible cause of the problem, if you are still using this, please update the driver to see if it is fixed. 
https://github.com/RinCat/RTL88x2BU-Linux-Driver/issues/198

wkliang, 2023-06-17:
@RinCat it work smoothly now! thanks for your diligent work

closed 2023-06-17 by wkliang