RinCat/RTL88x2BU-Linux-Driver #100: Can't build on kali linux(?)

MISTER-Unchained, 2021-08-17:
I tried building using dkms but I got this error:

KMS make.log for rtl88x2bu-git for kernel 5.10.0-kali7-amd64 (x86_64)
Tue 17 Aug 2021 05:25:17 PM CEST
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-kali7-amd64/build M=/var/lib/dkms/rtl88x2bu/git/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.0-kali7-amd64'
/bin/sh: 1: bc: not found
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.c: In function ‘dump_drv_version’:
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.c:45:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                                              ^
In file included from /var/lib/dkms/rtl88x2bu/git/build/include/drv_types.h:60,
                 from /var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.c:17:
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.c:45:44: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                            ^~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/include/rtw_debug.h:140:21: note: in definition of macro ‘RTW_PRINT’
  140 |      EX_INFO_ARG, ##arg);\
      |                     ^~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.c:45:2: note: in expansion of macro ‘RTW_PRINT_SEL’
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |  ^~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.c:45:54: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                                      ^~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/include/rtw_debug.h:140:21: note: in definition of macro ‘RTW_PRINT’
  140 |      EX_INFO_ARG, ##arg);\
      |                     ^~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.c:45:2: note: in expansion of macro ‘RTW_PRINT_SEL’
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |  ^~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.c:45:44: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                            ^~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/include/rtw_debug.h:254:25: note: in definition of macro ‘RTW_PRINT_SEL’
  254 |    _seqdump(sel, fmt, ##arg) /*rtw_warn_on(1)*/; \
      |                         ^~~
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.10.0-kali7-common/scripts/Makefile.build:284: /var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/src/linux-headers-5.10.0-kali7-common/Makefile:1822: /var/lib/dkms/rtl88x2bu/git/build] Error 2
make[1]: *** [/usr/src/linux-headers-5.10.0-kali7-common/Makefile:185: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-kali7-amd64'
make: *** [Makefile:2378: modules] Error 2

I installed both build-essential and kernel-headers. I'm not very familiar with this stuff tbh. 
Thanks in advance!

RinCat, 2021-08-17:
First, `/bin/sh: 1: bc: not found`, go install it `apt-get install bc`
Second, your system seems want to do reproducible builds, go to file `core/rtw_debug.c` and delete the line 45 `	RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);`

MISTER-Unchained, 2021-08-17:
Thanks for the quick reply!

MISTER-Unchained, 2021-08-17:
This solution worked for me on Kali Linux^ :D

samanfarzam, 2022-02-08:
How should I solve this problem?
kali)-[ /home/kali/Downloads/rtl88x2BU_WiFi_linux_v5.6.1.5_33902.20190604_COEX20180928-6a6a]
└─# make                                                                                                                                               2 ⨯
make: Warning: File 'Makefile' has modification time 17911003 s in the future
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-kali9-amd64/build M=/home/kali/Downloads/rtl88x2BU_WiFi_linux_v5.6.1.5_33902.20190604_COEX20180928-6a6a  modules
make[1]: *** /lib/modules/5.10.0-kali9-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2245: modules] Error 2