RinCat/RTL88x2BU-Linux-Driver #101: Device 0bda:b812 not in whitelist?

pauek, 2021-08-24:
I have a KagoLing Adapter (https://www.amazon.es/gp/product/B0894F897C/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) and the driver compiles but it does not detect the hardware. As you mention in the README, maybe its the hardware ID missing in the whitelist. This is the relevant line in the "lsusb" output:

    Bus 001 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. USB3.0 802.11ac 1200M Adapter

Using an older driver, I've managed to use it under Linux 5.4.0 but the sources for that driver are not compiling anymore on 5.8.0 and 5.11.0. Your driver seems to work but no WiFi adapter shows up.

Thanks in advance.

nextyang, 2022-01-13:
disagrees about version of symbol module_layout

nextyang, 2022-01-13:
> `0bda:b812`在列表中。 你能试试用吗 `modprobe 88x2bu`看看有什么错误 `dmesg`?

After 'dmesg' ,88x2bu: disagrees about version of symbol module_layout

RinCat, 2022-01-13:
> > `0bda:b812`在列表中。 你能试试用吗 `modprobe 88x2bu`看看有什么错误 `dmesg`?
> 
> After 'dmesg' ,88x2bu: disagrees about version of symbol module_layout

The version of the kernel you are using and the version you compiled do not match. Try reboot and install again?

nextyang, 2022-01-13:
> 
okay，I'll have a try.

nextyang, 2022-01-13:
> 您使用的内核版本和您编译的版本不匹配。 尝试重新启动并重新安装？

I tried again, but the problem still exists. Can you tell me what's wrong?

RinCat, 2022-01-13:
> > 您使用的内核版本和您编译的版本不匹配。 尝试重新启动并重新安装？
> 
> I tried again, but the problem still exists.

Can you post your `make` log and your kernel version `uname -a`?

nextyang, 2022-01-13:
> 

My version number is ’Linux kali 5.14.0-kali4-amd64 #1 SMP Debian 5.14.16-1kali1 (2021-11-05) x86_64 GNU/Linux‘ 
But what is the make log (I just touched it).

RinCat, 2022-01-13:
> My version number is ’Linux kali 5.14.0-kali4-amd64 #1 SMP Debian 5.14.16-1kali1 (2021-11-05) x86_64 GNU/Linux‘ But what is the make log (I just touched it).

The output when you run `make` in this driver directory. You may need run `make clean` first.

nextyang, 2022-01-13:
Is this?
make[1]: 进入目录“/usr/src/linux-headers-5.15.0-kali2-amd64”
  CC [M]  /usr/rtl88x2bu-git/core/rtw_cmd.o
make i  CC [M]  /usr/rtl88x2bu-git/core/rtw_security.o
nstall  CC [M]  /usr/rtl88x2bu-git/core/rtw_debug.o

  CC [M]  /usr/rtl88x2bu-git/core/rtw_io.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_ioctl_query.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_ioctl_set.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_ieee80211.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_mlme.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_mlme_ext.o
/usr/rtl88x2bu-git/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/usr/rtl88x2bu-git/core/rtw_mlme_ext.c:299:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  299 |                 if (rfctl->regd_name)
      |                    ^
/usr/rtl88x2bu-git/core/rtw_mlme_ext.c:302:9: note: here
  302 |         default:
      |         ^~~~~~~
/usr/rtl88x2bu-git/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/usr/rtl88x2bu-git/core/rtw_mlme_ext.c:1670:38: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1670 |                         ptable->func = &OnAuthClient;
      |                         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/usr/rtl88x2bu-git/core/rtw_mlme_ext.c:1672:9: note: here
 1672 |         case WIFI_ASSOCREQ:
      |         ^~~~
  CC [M]  /usr/rtl88x2bu-git/core/rtw_mi.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_wlan_util.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_vht.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_pwrctrl.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_rf.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_chplan.o
  CC [M]  /usr/rtl88x2bu-git/core/monitor/rtw_radiotap.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_recv.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_sta_mgt.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_ap.o
  CC [M]  /usr/rtl88x2bu-git/core/wds/rtw_wds.o
  CC [M]  /usr/rtl88x2bu-git/core/mesh/rtw_mesh.o
  CC [M]  /usr/rtl88x2bu-git/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /usr/rtl88x2bu-git/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_xmit.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_p2p.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_rson.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_tdls.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_br_ext.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_iol.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_sreset.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_btcoex_wifionly.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_btcoex.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_beamforming.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_odm.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_rm.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_rm_fsm.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_ft.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_wnm.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_mbo.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_rm_util.o
  CC [M]  /usr/rtl88x2bu-git/core/efuse/rtw_efuse.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_roch.o
  CC [M]  /usr/rtl88x2bu-git/core/crypto/aes-internal.o
  CC [M]  /usr/rtl88x2bu-git/core/crypto/aes-internal-enc.o
  CC [M]  /usr/rtl88x2bu-git/core/crypto/aes-gcm.o
  CC [M]  /usr/rtl88x2bu-git/core/crypto/aes-ccm.o
  CC [M]  /usr/rtl88x2bu-git/core/crypto/aes-omac1.o
  CC [M]  /usr/rtl88x2bu-git/core/crypto/ccmp.o
  CC [M]  /usr/rtl88x2bu-git/core/crypto/gcmp.o
  CC [M]  /usr/rtl88x2bu-git/core/crypto/aes-siv.o
  CC [M]  /usr/rtl88x2bu-git/core/crypto/aes-ctr.o
  CC [M]  /usr/rtl88x2bu-git/core/crypto/sha256-internal.o
  CC [M]  /usr/rtl88x2bu-git/core/crypto/sha256.o
[... 156 lines cut ...]
  CC [M]  /usr/rtl88x2bu-git/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /usr/rtl88x2bu-git/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /usr/rtl88x2bu-git/hal/phydm/halrf/halrf_psd.o
  CC [M]  /usr/rtl88x2bu-git/hal/phydm/rtl8822b/halhwimg8822b_bb.o
  CC [M]  /usr/rtl88x2bu-git/hal/phydm/rtl8822b/halhwimg8822b_mac.o
  CC [M]  /usr/rtl88x2bu-git/hal/phydm/halrf/rtl8822b/halrf_8822b.o
  CC [M]  /usr/rtl88x2bu-git/hal/phydm/rtl8822b/phydm_hal_api8822b.o
  CC [M]  /usr/rtl88x2bu-git/hal/phydm/halrf/rtl8822b/halhwimg8822b_rf.o
  CC [M]  /usr/rtl88x2bu-git/hal/phydm/halrf/rtl8822b/halrf_iqk_8822b.o
  CC [M]  /usr/rtl88x2bu-git/hal/phydm/halrf/rtl8822b/halrf_rfk_init_8822b.o
  CC [M]  /usr/rtl88x2bu-git/hal/phydm/rtl8822b/phydm_regconfig8822b.o
  CC [M]  /usr/rtl88x2bu-git/hal/phydm/rtl8822b/phydm_rtl8822b.o
  CC [M]  /usr/rtl88x2bu-git/hal/phydm/txbf/haltxbf8822b.o
  CC [M]  /usr/rtl88x2bu-git/hal/btc/halbtc8822bwifionly.o
  CC [M]  /usr/rtl88x2bu-git/hal/btc/halbtc8822b1ant.o
  CC [M]  /usr/rtl88x2bu-git/hal/btc/halbtc8822b2ant.o
  CC [M]  /usr/rtl88x2bu-git/platform/platform_ops.o
  CC [M]  /usr/rtl88x2bu-git/core/rtw_mp.o
  LD [M]  /usr/rtl88x2bu-git/88x2bu.o
  MODPOST /usr/rtl88x2bu-git/Module.symvers
  CC [M]  /usr/rtl88x2bu-git/88x2bu.mod.o
  LD [M]  /usr/rtl88x2bu-git/88x2bu.ko
  BTF [M] /usr/rtl88x2bu-git/88x2bu.ko
Skipping BTF generation for /usr/rtl88x2bu-git/88x2bu.ko due to unavailability of vmlinux
make[1]: 离开目录“/usr/src/linux-headers-5.15.0-kali2-amd64”

RinCat, 2022-01-13:
Short answer, you can't. 5.14 kernel headers no longer exists in kali linux repository. Unless you can find it somewhere else.

RinCat, 2022-01-13:
> > 
> 
> I'm still wondering why I can't find the header file.So what now? Reinstall the system?

Update your kernel, `apt update && apt upgrade`, then reboot.

nextyang, 2022-01-13:
> > > 
> > 
> > 
> > 我仍然想知道为什么我找不到头文件。那现在呢？  重装系统？
> 
> 更新你的内核， `apt update && apt upgrade`，然后重新启动。

我竟然成功了！非常感谢您！！！