RinCat/RTL88x2BU-Linux-Driver #103: 0b05:19aa not recognized

Elekkz, 2021-08-28:
I have an Asus adapter RTL8812AU and tried to install it on kernel 5.10, after install it does not appear in if/iwconfig, only in lsusb:
Bus 001 Device 002: ID 0b05:19aa ASUSTek Computer, Inc. ASUS USB-AC58 USB Wireless adapter

dmesg of modprobe:
[ 2856.355735] platform regulatory.0: firmware: direct-loading firmware regulatory.db
[ 2856.356776] platform regulatory.0: firmware: direct-loading firmware regulatory.db.p7s
[ 2856.361794] 88x2bu: loading out-of-tree module taints kernel.
[ 2856.389339] RTW: module init start
[ 2856.389342] RTW: rtl88x2bu v5.8.7.4_37264.20200922_COEX20191120-7777
[ 2856.389343] RTW: build time: Aug 28 2021 06:27:00
[ 2856.389344] RTW: rtl88x2bu BT-Coex version = COEX20191120-7777
[ 2856.389389] usbcore: registered new interface driver rtl88x2bu
[ 2856.389390] RTW: module init ret=0

Is this Device in Whitelist ?

Thanks!

RinCat, 2021-08-28:
Hi, do you have any link where you got the adapter? I cannot find any information about that ID `0b05:19aa`. Thanks.

Elekkz, 2021-08-28:
https://www.asus.com/Networking-IoT-Servers/Adapters/All-series/USB-AC58/

Hello, it is this adapter, but I was having the same issues when installing the official drivers from the Asus site.

RinCat, 2021-08-28:
Fixed in https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/3663366473b5109ec4aca6ada82705d7df575d29, you can give a try.

RinCat, 2021-08-28:
And your adapter is `RTL8822BU`, not `RTL8812AU`

Elekkz, 2021-08-28:
It works now, thanks a lot!

closed 2021-08-28 by Elekkz