RinCat/RTL88x2BU-Linux-Driver #104: Unable to Buil on Kernel 5.11.5 Ubuntu 18.04

octobotics, 2021-08-31:
Hi, I am getting this error, while trying to build the driver:-

My kernel version is : 5.11.5, kernel headers are installed.

ld:scripts/module.lds:6: ignoring invalid character `#' in expression
ld:scripts/module.lds:6: syntax error
scripts/Makefile.modfinal:58: recipe for target '/home/octo-ws/rtl8812au-5.9.3.2/8812au.ko' failed
make[3]: *** [/home/octo-ws/rtl8812au-5.9.3.2/8812au.ko] Error 1
scripts/Makefile.modpost:117: recipe for target '__modpost' failed
make[2]: *** [__modpost] Error 2
Makefile:1710: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.11.5-051105-generic'
Makefile:2362: recipe for target 'modules' failed
make: *** [modules] Error 2

RinCat, 2021-08-31:
Hi, this is a bug in Ubuntu https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1906131
Check the comments to see if any fix works in your system.

PyBonnetainNesterenko, 2021-11-28:
Hello,
Not « only » an Ubuntu bug but a >5.10 kernel change. 
Got this same error (module.lds related) on a RaspberryPi running Gentoo.

Solved it by somehow following hints given in https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1906131.

What we did :

1. get the proper kernel sources
2. in the kernel source directory, copy the .config file from running kernel (modprobe configs ; zcat /proc/config.gz > .config)
3. make olddefconfig && make modules_prepare
4. go to the RTL driver directory, compile and install.

Everything seems to be okay.

AlexeiScherbakov, 2022-05-04:
@PyBonnetainNesterenko this is not helping for Odroid Armbian (5.10.102)
```
scripts/Makefile.build:415: warning: overriding recipe for target 'modules.order'
Makefile:1422: warning: ignoring old recipe for target 'modules.order'
make[1]: *** No rule to make target 'arch/arm64/kernel/vdso/vdso.lds', needed by 'arch/arm64/kernel/vdso/vdso.so.dbg'.  Stop.
make: *** [arch/arm64/Makefile:186: vdso_prepare] Error 2
```

AlexeiScherbakov, 2022-05-05:
Workaround - switching to unstable kernel  5.17.5