RinCat/RTL88x2BU-Linux-Driver #107: Module panics on load (aarch64)

peperunas, 2021-09-23:
The module panics when loaded automatically or through `modprobe 88x2bu`.

Log:

```
[   61.202994] RTW: module init start
[   61.206409] RTW: rtl88x2bu v5.8.7.4_37264.20200922_COEX20191120-7777
[   61.212776] RTW: rtl88x2bu BT-Coex version = COEX20191120-7777
[   61.218754] RTW: [HALMAC]11692M
[   61.218754] HALMAC_MAJOR_VER = 1
[   61.218754] HALMAC_PROTOTYPE_VER = 6
[   61.218754] HALMAC_MINOR_VER = 5
[   61.218754] HALMAC_PATCH_VER = 6*
[   61.236150] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   61.499574] RTW: HW EFUSE
[   61.502243] RTW: 0x000: 29 81 00 3C  09 00 A1 00  B6 04 64 10  00 00 A3 00  
[   61.509369] RTW: 0x010: 33 33 33 33  33 33 33 33  33 33 33 00  00 00 FF FF  
[   61.516468] RTW: 0x020: FF FF 28 28  28 28 28 28  28 28 28 28  28 28 28 28  
[   61.523594] RTW: 0x030: 00 00 FF FF  00 FF D0 00  FF FF 32 32  32 32 32 32  
[   61.530673] RTW: 0x040: 32 32 32 32  32 00 00 00  FF FF FF FF  27 27 27 27  
[   61.537778] RTW: 0x050: 27 27 27 27  27 27 27 27  27 27 00 00  FF FF 00 FF  
[   61.544886] RTW: 0x060: D0 00 FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.552010] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.559127] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.566211] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.573336] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.580414] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 1E 1B 00  FF FF FF FF  
[   61.587497] RTW: 0x0C0: FF 09 00 11  00 00 00 00  00 FF 03 FF  FF FF FF FF  
[   61.594627] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.601712] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.608830] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.615914] RTW: 0x100: DA 0B 12 B8  C1 67 02 BC  EC 23 C3 70  6A 09 03 52  
[   61.623041] RTW: 0x110: 65 61 6C 74  65 6B 1F 03  55 53 42 33  2E 30 20 38  
[   61.630118] RTW: 0x120: 30 32 2E 31  31 61 63 20  31 32 30 30  4D 20 41 64  
[   61.637201] RTW: 0x130: 61 70 74 65  72 08 03 31  32 33 34 35  36 FF FF FF  
[   61.644328] RTW: 0x140: 33 0A 1F 01  00 00 21 0F  FF FF FF FF  FF FF FF FF  
[   61.651413] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.658531] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.665614] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.672762] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.679839] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.686923] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.694051] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.701128] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.708211] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.715343] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.722430] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.729550] RTW: 0x200: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.736633] RTW: 0x210: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.743762] RTW: 0x220: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.750840] RTW: 0x230: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.757923] RTW: 0x240: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.765052] RTW: 0x250: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.772137] RTW: 0x260: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.779258] RTW: 0x270: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.786341] RTW: 0x280: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.793471] RTW: 0x290: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.800548] RTW: 0x2A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.807632] RTW: 0x2B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.814762] RTW: 0x2C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.821847] RTW: 0x2D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.828969] RTW: 0x2E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.836077] RTW: 0x2F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   61.843216] RTW: hal_com_config_channel_plan chplan:0x7F
[   61.848526] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   61.862786] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[   61.958100] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   61.963679] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   61.969223] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   61.974810] RTW: [RF_PATH] ver_id.RF_TYPE:RF_2T2R, rf_reg_path_num:2, max_tx_cnt:2
[... 191 lines cut ...]
[   63.258831]  0xffffffc010502884
[   63.261969]  0xffffffc010477ed8
[   63.265107]  0xffffffc01047861c
[   63.268245]  0xffffffc010478994
[   63.271384]  0xffffffc010478a24
[   63.274522]  0xffffffc0104757f0
[   63.277660]  0xffffffc010477728
[   63.280798]  0xffffffc010477008
[   63.283936]  0xffffffc010479458
[   63.287074]  0xffffffc0105011f0
[   63.290214]  0xffffffc0087ef0e4 [88x2bu@0000000060c924be+0x2a6000]
[   63.296393]  0xffffffc010002e28
[   63.299531]  0xffffffc0100b7294
[   63.302669]  0xffffffc0100b8c94
[   63.305807]  0xffffffc0100b9200
[   63.308946]  0xffffffc0100b941c
[   63.312084]  0xffffffc010012b7c
[   63.315222]  0xffffffc0107af7a4
[   63.318360]  0xffffffc0107afbd4
[   63.321498]  0xffffffc0100024e4
[   63.324637] ---[ end trace c36688be98f48188 ]---
[   63.329719] RTW: WARN free_recv_skb_queue not empty, 8
[   63.334985] usbcore: registered new interface driver rtl88x2bu
[   63.340812] RTW: module init ret=0
```

RinCat, 2021-09-23:
Hi, please enable debug info for linux kernel, only has memory address is not useful.

Sargastic, 2022-02-08:
> The module panics when loaded automatically or through `modprobe 88x2bu`.

Almost same thing here. Module versions 5.8.7.4 and 5.13.1 (at least) panic on kernel 5.10.63, on a Raspberry Pi 3B. No problem at all with kernel 5.10.11 for both versions.