RinCat/RTL88x2BU-Linux-Driver #110: Unable to connect reliably to WPA2/PEAP university network

ZakSN, 2021-11-06:
I have a generic RLT88x2BU based USB Wi-Fi adapter [Amazon link](https://www.amazon.ca/gp/product/B08BYP4RRV/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1), and I can't get it to reliably connect to my university's WPA2/PEAP network.

some context:

- other devices (e.g. android phone), and other Wi-Fi adapters (Dlink dwa-121 150-pico, not sure which chipset) are able to reliably connect to the network from roughly the physical same location (i.e. not a signal strength issue)
- The connection is configured correctly in NetworkManager, I know this since every once in a while I can get the RTL88x2BU adapter to connect with the connection configuration I'm providing. (i.e. certificate and login credentials are correct)
- The device and driver association is correct, I can see networks and can connect to unsecured networks no problem
- I'm attempting to connect with NetworkManager (via `nm-applet`, but `nmcli` produces the same behaviour), with a wpa_supplicant backend (I've tried the `iwd` backend, but wasn't able to get it configured properly)

System information:
```
# output of uname -a
Linux zaks-laptop 5.14.16-arch1-1 #1 SMP PREEMPT Tue, 02 Nov 2021 22:22:59 +0000 x86_64 GNU/Linux
```
attached are the tail of the output of the commands:
```
sudo dmesg --human -P
```
and
```
journalctl -x -n100 NM_CONNECTION=ab7ac2d7-8d07-4b30-be5e-fd6c0c7a5bf9 + NM_DEVICE=wlan1
```
respectively

[dmesg.log](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/7491209/dmesg.log)
[journal.log](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/7491210/journal.log)

Note that the above logs were collected after unplugging a working Wi-Fi adapter, and then plugging in the RTL88x2BU based adapter. As such there may be some garbage at the beginning of these files. (hopefully all of the relevant context is included)

What should I try as a next debugging step?

RinCat, 2021-11-07:
Looks like you are trying to connect restricted WiFi channels, and got timeout.
You can try use 2.4GHz WiFi or use common available channel like 36 for 5GHz.

ZakSN, 2021-11-07:
changing the network manager connection configuration to specify a frequency and a band doesn't seem to allow me to connect.

I ran `nmcli -f ALL dev wifi`, and looked for a 5GHz channel with with a high signal strength, and then modified my network manager connection to use that frequency and channel. This produced the same result as before. the Journal still has the messages about wifi association taking too long

RinCat, 2021-11-07:
Which 5G channel you are using?

ZakSN, 2021-11-07:
I saw that channel 11 had a high signal strength so I tried to connect to that. does the chipset only support some 5GHz channels?

RinCat, 2021-11-07:
No, it support both 2.4G and 5G.

ZakSN, 2021-11-07:
that's what I thought. scanning with `nmcli` seems to pick up both 5GHz and 2.4GHz connections

RinCat, 2021-11-07:
But some 5G channels are restricted by regulations, which you may not able to use them if you are in some countries.

ZakSN, 2021-11-07:
here's a screen cap of the output of `nmcli -f ALL dev wifi`, showing some of the channels that the RTL88xBU2 dongle can see
[image]

There are some 5GHz and some 2.4GHz channels. 

attempting to connect to e.g. 5GHz channel 104 doesn't work.

(In my previous comment I mentioned connecting to 5GHz channel 11, this was a mistake, I meant 2.4GHz channel 11. I've tried again with 5GHz channel 104, and got the same result)

RinCat, 2021-11-07:
You should try connect channel 11. If that failed, there may be an issue in PEAP. But I current not have eduroam nearby, so I cannot test that.

ZakSN, 2021-11-07:
I tried again with 2.4GHz channel 11, and was not able to connect, despite being able to see the connection in nmcli.

I'm sure that the PEAP credentials are correct since I can connect to the network using those credential and a different wifi adapter.

the other wifi adapter does not see 2.4GHz channel 11, so I can't connect to it with the other adapter.

ZakSN, 2021-11-07:
it seems to always fail in the activation phase with the message ` [1636261340.5003] device (wlan1): Activation: (wifi) association took too long`, I'm not sure which end controls that timeout. i.e. can I change the timeout in network manager, or is the access point (which I don't control) deciding to reset the timer

RinCat, 2021-11-07:
That timeout usually means something incompatible, like try to use a different channel or use unsupported encryption or handshake. You may try get `wpa_supplicant` logs.