RinCat/RTL88x2BU-Linux-Driver #111: Monitor mode and "packet injection" compatibility

wedrano99, 2021-11-30:
hey! this is not a issue, but a need some help :) 
how can i activate the monitor mode on this drive? and you guys knows if this type of adapters is compatible with "packet injection"?
thanks for u attention :)

morrownr, 2021-12-01:
@RinCat 

I recently updated my repo to a new version of this driver. I wasn't planning on supporting monitor mode but it works well. Here is the repo:

https://github.com/morrownr/88x2bu-20210702

Be glad to drop a copy of clean source here if you prefer.

Regards,

Nick

RinCat, 2021-12-01:
@wedrano99 This one support monitor mode and packet injection. You need to follow airmon-ng docs, there is nothing you need to set in the driver.

https://github.com/RinCat/RTL88x2BU-Linux-Driver/pull/51

RinCat, 2021-12-01:
@morrownr If you can tell me where you found the new driver would be better.

morrownr, 2021-12-01:
Edimax

RinCat, 2021-12-01:
That seems a huge version bump.

wedrano99, 2021-12-02:
Thanks guys, you are amazing! :)

closed 2021-12-02 by wedrano99