RinCat/RTL88x2BU-Linux-Driver #112: Archer T3U link up fail

trosh, 2021-12-04:
Hello, I've been trouble getting a dongle to work.

* Dongle: TP-Link Archer T3U
* Linux: 5.15.5.arch1-1
* Motherboard: X470 Aorus ultra gaming

The dongle fails to show up in nmtui (+ ip a shows the device as down) when connecting to a USB 2 port (I didn't provide the USB3 option to modprobe).
Here's my dmesg log when I do `sudo ip link set dev wlp2s0f0u9 up`: https://pastebin.com/raw/Z5LCvFRD

Furthermore, `ip a` looks like this:
```
4: wlp2s0f0u9: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 3e:51:20:6e:d9:a7 brd ff:ff:ff:ff:ff:ff permaddr d0:37:45:10:9d:f9
```

I then tried on a USB 3 port with rtw_switch_usb_mode=1 ; also without success. The dmesg output is shown in the same pastebin.

I've never been able to see the device in NetworkManager.

Thanks for any help!!

RinCat, 2021-12-04:
It should shows DOWN if you are not connected to any WiFi, that is not an issue.
The driver is successfully initialized and you have `wlp2s0f0u7`.
You should check NetworkManager logs.

trosh, 2021-12-04:
Thank you for your quick response!

Now that I'm home from some christmas shopping, here's what I got from `sudo journalctl -f` after having set NetworkManager logs to trace with `nmcli general logging level TRACE domains ALL`:

```
Dec 04 21:17:06 archer dbus-daemon[543]: [system] Activating via systemd: service name='org.freedesktop.home1' unit='dbus-org.freedesktop.home1.service' requested by ':1.40' (uid=0 pid=3792 comm="sudo ip link set dev wlp2s0f0u7 up ")
Dec 04 21:17:06 archer dbus-daemon[543]: [system] Activation via systemd failed for unit 'dbus-org.freedesktop.home1.service': Unit dbus-org.freedesktop.home1.service not found.
Dec 04 21:17:06 archer sudo[3792]: pam_systemd_home(sudo:account): systemd-homed is not available: Unit dbus-org.freedesktop.home1.service not found.
Dec 04 21:17:06 archer kernel: audit: type=1105 audit(1638649026.084:94): pid=3792 uid=1000 auid=1000 ses=1 msg='op=PAM:session_open grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/4 res=success'
Dec 04 21:17:06 archer audit[3792]: USER_START pid=3792 uid=1000 auid=1000 ses=1 msg='op=PAM:session_open grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/4 res=success'
Dec 04 21:17:06 archer sudo[3792]:    trosh : TTY=pts/4 ; PWD=/home/trosh ; USER=root ; COMMAND=/usr/bin/ip link set dev wlp2s0f0u7 up
Dec 04 21:17:06 archer sudo[3792]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Dec 04 21:17:06 archer kernel: RTW: [HALMAC][ALWAYS]shall R reg twice!!
Dec 04 21:17:10 archer kernel: RTW: txpath=0x3, rxpath=0x3
Dec 04 21:17:10 archer kernel: RTW: txpath_1ss:0x1, num:1
Dec 04 21:17:10 archer kernel: RTW: txpath_2ss:0x3, num:2
Dec 04 21:17:10 archer kernel: start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
Dec 04 21:17:10 archer kernel: RTW: wlp2s0f0u7- hw port(0) mac_addr =e6:b6:1d:5a:87:19
Dec 04 21:17:10 archer NetworkManager[544]: <trace> [1638649030.8836] platform-linux: event-notification: RTM_NEWLINK, flags 0, seq 0: 3: wlp2s0f0u7 <UP;broadcast,multicast,up> mtu 1500 arp 1 wifi? not-init addrgenmode none addr E6:B6:1D:5A:87:19 brd FF:FF:FF:FF:FF:FF rx:0,0 tx:0,0
Dec 04 21:17:10 archer audit[3792]: USER_END pid=3792 uid=1000 auid=1000 ses=1 msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/4 res=success'
Dec 04 21:17:10 archer audit[3792]: CRED_DISP pid=3792 uid=1000 auid=1000 ses=1 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/4 res=success'
Dec 04 21:17:10 archer NetworkManager[544]: <debug> [1638649030.8836] platform: (wlp2s0f0u7) signal: link changed: 3: wlp2s0f0u7 <UP;broadcast,multicast,up> mtu 1500 arp 1 wifi? init addrgenmode none addr E6:B6:1D:5A:87:19 brd FF:FF:FF:FF:FF:FF driver rtl88x2bu rx:0,0 tx:0,0
Dec 04 21:17:10 archer sudo[3792]: pam_unix(sudo:session): session closed for user root
Dec 04 21:17:10 archer NetworkManager[544]: <debug> [1638649030.8837] device[c7830cd96c002c4c] (wlp2s0f0u7): queued link change for ifindex 3
Dec 04 21:17:10 archer kernel: audit: type=1106 audit(1638649030.880:95): pid=3792 uid=1000 auid=1000 ses=1 msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/4 res=success'
Dec 04 21:17:10 archer kernel: audit: type=1104 audit(1638649030.880:96): pid=3792 uid=1000 auid=1000 ses=1 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/4 res=success'
Dec 04 21:17:13 archer kernel: RTW: nolinked power save enter
```

(I figure this log can fit here but otherwise I can upload it to pastebin)

I'm wondering whether it's something to do with mtu setup?

Again, thank you for your help.

RinCat, 2021-12-04:
I don't see any suspicious, and this one has nothing to do with MTU.
NetworkManager did found the interface, I don't know why it did not shows.