RinCat/RTL88x2BU-Linux-Driver #113: ERROR: modpost: "mcount" [RTL88x2BU-Linux-Driver/88x2bu.ko] undefined!

kisira, 2021-12-08:
I cant build the driver, on Ubuntu 21.10. I has always built before on the same machine. Please help, thanks

RinCat, 2021-12-08:
Hi, please post full logs.

kisira, 2021-12-09:
> Hi, please post full logs.

The code didn't build, as such I could not install it. When I try to install it with "sudo make install" the error I get is:

install -p -m 644 88x2bu.ko  /lib/modules/5.13.0-22-generic/kernel/drivers/net/wireless/
install: cannot stat '88x2bu.ko': No such file or directory
make: *** [Makefile:2460: install] Error 1

I cannot run "modprobe 88x2bu rtw_drv_log_level=5"  because there is no driver installed, since it didn't build. I am also not sure how to generate build logs, if possible. I had already uninstalled te old driver when I was building this driver. I am running Ubuntu 21.10, with Kernel version 5.13.0-22-generic.

RinCat, 2021-12-09:
I mean the build logs. 
The output when running `make`.

kisira, 2021-12-09:
> I mean the build logs. The output when running `make`.

Thanks for the clarification. Please point out which file or files those are. Are they in the same folder as the code? Do I have to run make with some switches?

RinCat, 2021-12-09:
There is no such files, just copy the output after run `make`.

kisira, 2021-12-13:
Please see the attached file.
Thanks

> I mean the build logs. The output when running `make`.

Thanks for the clarification. Please point out which file or files those are. Are they in the same folder as the code? Do I have to run make with some switches
[results_make.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/7705476/results_make.txt)
?

RinCat, 2021-12-13:
There is no log file when you manual `make`.
So that is the correct logs.

From the logs I think there is something wrong with your compilation environment.
You should try run `make V=1` to show full commands, and upload the log.

kisira, 2021-12-13:
> 
Please see the attache file
Thanks
[results_make2.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/7705706/results_make2.txt)

RinCat, 2021-12-13:
`-g -gdwarf-4 -pg -mrecord-mcount`
Your system gcc enabled debug and profile information, which usually not enabled.
Try edit file `Makefile` and remove line `EXTRA_LDFLAGS += --strip-debug`

kisira, 2021-12-13:
> `-g -gdwarf-4 -pg -mrecord-mcount` Your system gcc enabled debug and profile information, which usually not enabled. Try edit file `Makefile` and remove line `EXTRA_LDFLAGS += --strip-debug`

I tried it, but I don't think it helped. 
Here is the resulting log file
Thank
[results_make3.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/7706573/results_make3.txt)
s

RinCat, 2021-12-13:
Try add 
`EXTRA_CFLAGS += -mfentry`
at the top of `Makefile`

kisira, 2021-12-13:
> Try add `EXTRA_CFLAGS += -mfentry` at the top of `Makefile`

Wow! That worked! Thank you so much

kisira, 2021-12-13:
> I do recommended you to fix your local environment, as it could cause the same issue if you compile other software.

Thank you for the advise. How do I do that?

RinCat, 2021-12-13:
> > I do recommended you to fix your local environment, as it could cause the same issue if you compile other software.
> 
> Thank you for the advise. How do I do that?

I don't know, too many things could change that. You should check your linux distribution docs.

kisira, 2021-12-13:
> > > I do recommended you to fix your local environment, as it could cause the same issue if you compile other software.
> > 
> > 
> > Thank you for the advise. How do I do that?
> 
> I don't know, too many things could change that. You should check your linux distribution docs.

Ok, I will look into it. If you don't mind me asking, what does the EXTRA_CFLAGS += -mfentry do?

RinCat, 2021-12-13:
`-mfentry` use a new mechanism to do profiling counter, which override `mcount`.
So your gcc will not enable `-mrecord-mcount`. It not a fix but a workaround.