RinCat/RTL88x2BU-Linux-Driver #114: modprobe doesn't work

ba1989, 2021-12-08:
Hi

Ubuntu 20.04.3LTS, Kernel 5.11.0-41-generic
TPlink Archer T3U Plus --> ID 2357:0138 (TP-Link 802.11ac NIC)

I can't get the "sudo modprobe 88x2bu" to work. It always displays the following Error:
_modprobe: ERROR: could not insert '88x2bu': Operation not permitted_

I also tried to sign the driver, but still the same Error.

What i've done so far:
- cloned the repository
- make
- sudo make install
- sudo modprobe 88x2bu --> here I got the Error

Could someone help me? What am I doing wrong?

Thx ba1989

RinCat, 2021-12-08:
Hi, run `dmesg` should give you the reason why `Operation not permitted`.

ba1989, 2021-12-08:
[  398.381052] Lockdown: modprobe: unsigned module loading is restricted; see man kernel_lockdown.7

It's not working because of the lacking sign.

Did I do something wrong in building this driver? Or do I have to change someting with secureboot?

RinCat, 2021-12-08:
Nothing wrong, but I do not think you can signing your own kernel module when using official kernel.
You can either disable lockdown / mandatory signature (needs disable secureboot), or build your own kernel from sources.

ba1989, 2021-12-08:
Now I got it working.
I first disabled the driver validation with `sudo mokutil --disable-validation`
--> Secure Boot is still enabled, but the driver verification is disabled:
`sudo mokutil --sb-state`
_SecureBoot enabled
SecureBoot validation is disabled in shim_

ba1989, 2021-12-08:
Great, now everything works fine.
Thank you very much for this fast help!

closed 2021-12-10 by RinCat