RinCat/RTL88x2BU-Linux-Driver #115: pop_os 21.04 kernel 5.

rogerguti, 2021-12-10:


RinCat, 2021-12-10:
`rtl88x2bu-git/include/hal_com_reg.h:1886:2: error: invalid preprocessing directive #deHA8191EUW_G_IO_OPW`

This one does not make sense. You sure this is a single line?

rogerguti, 2021-12-10:
Yes, at least this is how it gets formatted in the terminal and the log file.
For what it's worth, I was able to make/install a driver from another repository, so it works for me now. However, I just tried to make again with the files from here, and the error persists. It might be an error from my side, so you can close the issue if you believe so, or I can continue investigating if you want. I won't have access to the PC from now to next Monday though.
Thanks for checking!

RinCat, 2021-12-10:
If you want, I can continue investigating. However the error lines are like mixed/fragmented code, and does not exists in the source file. I do not know what happened when it preprocessing. Can you provide versions about your compilation environment?

rogerguti, 2021-12-14:
I assume you mean the build-essential package, as it is the only one you mention in the readme?
If so:
build-essential/hirsute,now 12.8ubuntu3 amd64 [installed]
build-essential/hirsute 12.8ubuntu3 i386

If you mean something else, can you specify? I know my way around linux, but don't really build and compile much stuff, so I am a bit lost here.

RinCat, 2021-12-14:
I just tested both 5.13.0-7620-generic and 5.15.5-76051505-generic on pop os，and did not able to reproduce this issue.
Maybe try full system upgrade? `sudo apt update && sudo apt upgrade`

rogerguti, 2021-12-16:
I had already updated and upgraded before trying. I did again and it still failed.
I really think it is a problem with my system. As I tried to upgrade to pop os 21.10, the installation failed and I had to refresh the OS. Now with a 'clean' OS, I tried both in 21.04 and 21.10 once I managed to get there, and still fails.
At this point, I have some other drivers working, and since it really looks like it is a problem in my end, I think it is better to close the issue.
Thanks a lot for helping though, I appreciate it a lot. Up until this update, your driver really worked perfectly for me.