RinCat/RTL88x2BU-Linux-Driver #118: How to rtl88x2bu support VIF to evil twin attack

Appleuy, 2021-12-30:
How to rtl88x2bu support VIF to evil twin attack

RinCat, 2021-12-30:
Sorry, I do not understand what you mean.
If you wanna test evil twin attack, you should check Kali linux.

Appleuy, 2021-12-31:
Support multiple Virtual interface @
[image]

Appleuy, 2021-12-31:
The interface wlan1 you have already selected is not supporting VIF (Virtual Interface). This attack needs it to virtually unfold itself to create the fake access point while also performing denial of service (DoS)

RinCat, 2021-12-31:
`make USER_EXTRA_CFLAGS="-DCONFIG_CONCURRENT_MODE"`
will enable VIF, but I did not test if it usable in the attack.

Appleuy, 2022-01-01:
After running the above command, airgeddon still says VIF is not supported
[image]

RinCat, 2022-01-01:
Can you check how many interface you have? After enable that I have 2 same interfaces.