RinCat/RTL88x2BU-Linux-Driver #121: Netgear A6150 - Does not work

Lurux, 2022-01-03:
Tested on Linux Mint 20.2 (kernel 5.4) and Manjaro via AUR (kernel 5.10)

lsusb reports the device as "Netgear, Inc. A6150"

I installed this driver but the wifi interface doesn't show up. https://github.com/cilynx/rtl88x2bu seems to work properly.

Hope this helps, please let me know if I should add anything else !

RinCat, 2022-01-03:
What you got after run `modprobe 88x2bu`?

Lurux, 2022-01-03:
`FATAL: Module 88x2bu not found in directory /lib/modules/5.10.89-1-MANJARO`

Seems like the module didn't get installed properly...

Also it seems like the driver that was working isn't Cilynx's one (I'm currently having trouble trying to build it) - I know I had found a working driver, but I can't seem to find it again unfortunately... :disappointed:

RinCat, 2022-01-03:
Is the running kernel version the same as the kernel version you installed?
Did you run `sudo make install` after compiled the driver?

Lurux, 2022-01-03:
Hold on - I just realized the kernel headers didn't get installed with the package - I didn't realize it didn't have the dependency. Retrying now...

Duh if this is it I'll feel so stupid

Lurux, 2022-01-03:
Installed headers then reinstalled - works now. Maybe it *is* actually this driver that was working actually :smile: 

(Yeah, I managed to get this dongle working on the other machine but I don't have access to it rn and I didn't have the driver reference. That's a stupid situation i just found myself in, I know...)

closed 2022-01-03 by Lurux