RinCat/RTL88x2BU-Linux-Driver #122: unable to build on kernel 5.15

truexfan81, 2022-01-04:
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_br_ext.c:20:18: fatal error: net/ipx.h: No such file or directory
   20 |         #include <net/ipx.h>
      |                  ^~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:277: /var/lib/dkms/rtl88x2bu/git/build/core/rtw_br_ext.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1868: /var/lib/dkms/rtl88x2bu/git/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.15.11_tkg_pds'
make: *** [Makefile:2378: modules] Error 2

truexfan81, 2022-01-04:
did some more experimenting, it's not in the includes in the code thats cloned from the repository, but when you use the command "dkms autoinstall" and it generates the files in /var/lib/dkms/rtl88x2bu/git/build/core/  the includes for net/ipx.h are there. i tried removing the includes from that file, but after running the command again it puts it back

RinCat, 2022-01-04:
remove `/usr/src/rtl88x2bu-git` and redo it again.

RinCat, 2022-01-04:
DKMS do not update the driver, so you have to it every time this repo updated.

truexfan81, 2022-01-04:
> remove `/usr/src/rtl88x2bu-git` and redo it again.

after deleting that i get new errors
sed -i 's/PACKAGE_VERSION="@PKGVER@"/PACKAGE_VERSION="git"/g' /usr/src/rtl88x2bu-git/dkms.conf
sed: can't read /usr/src/rtl88x2bu-git/dkms.conf: No such file or directory

dkms add -m rtl88x2bu -v git
Error! Could not find module source directory.
Directory: /usr/src/rtl88x2bu-git does not exist.

truexfan81, 2022-01-04:
still not working and i can't seem to find out why because dkms is not cooperating
# dkms remove --all rtl88x2bu -v git
# dkms status
rtl88x2bu/git: added

i've even tried with --force it just won't remove it, and this time there is no log to tell me why it failed or isn't loaded, but i don't see it when i lsmod

truexfan81, 2022-01-05:
i gave up on it and installed windows 10 so i guess we can close this one

RinCat, 2022-01-05:
You can try `dkms uninstall rtl88x2bu --all`, but if kernel and header mismatched I cannot do much.

clook, 2022-01-19:
Hello @truexfan81, ipx support has been dropped 2 months ago on `master` branch (to prevent the error you see since kernel 5.15 dropped support for ipx), see https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/master/core/rtw_br_ext.c.

So I would first double check the version that is failing is not an old one.