RinCat/RTL88x2BU-Linux-Driver #123: modprobe error

pypaut, 2022-01-09:
The build and installation process is silky smooth, but running `modprobe 88x2bu` or `modprobe --force 88x2bu` gives me error `modprobe: ERROR: could not insert '88x2bu': Exec format error`.

Tried both methods from the readme file and reboot. I use a TP LINK Archer adapter which is recognized under Windows 10.

Thanks for your work, and thanks by advance if you read this.

RinCat, 2022-01-10:
Hi, can you check if your running kernel version is the same as kernel header version?
Also run `dmesg` may give you some information.

pypaut, 2022-01-10:
Hello, I have since yesterday made a full reinstall of my OS, and cannot check the said error. I have another bug however trying to compile it, maybe I should open another issue instead of keeping this one?

RinCat, 2022-01-10:
You can keep use this one.

pypaut, 2022-01-10:
Thank you!

Here is what I get after running `make -j4`:

```
Skipping BTF generation for /home/pypaut/Downloads/RTL88x2BU-Linux-Driver/88x2bu.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.8-76051508-generic'
```

I read [here](https://www.reddit.com/r/openSUSE/comments/pwxs3i/btf_file_not_generated_because_vmlinux_could_not/) that it might be caused by the `vmlinux` file not being where it used to?

RinCat, 2022-01-10:
You can ignore "Skipping BTF generation" since that is for debug.

pypaut, 2022-01-10:
Okay my bad.

All errors are gone now and my adapter works <3

Thanks again for your work and your reactivity!

closed 2022-01-10 by pypaut