RinCat/RTL88x2BU-Linux-Driver #125: [tp-link Archer T3U AC 1300 RTl8812bu] USB 3.0 slower than USB 2.0

by-speece, 2022-01-28:
Hello,
Today I have already received my second TP-LINK Archer T3U AC1300 wifi module (the first one has been used in my local server for 9 months).
In general, I was tempted by the possibility of obtaining a speed of about 150 Mb / s {Wifi 5ghz) on USB 2.0 (the server has only 2.0 ports) Thus, I found that on USB 3.0 they will be much larger. So I sit down tonight, with my memory, it installs the driver, restarts, and it works.
After a few tests (I carried out each test 3 times), I came to an unpleasant conclusion. USB 2.0 gives you faster speeds than USB 3.0!

All tests were carried out on the website speedtest.pl
---------------------------------------------------------------------------------------------------
#USB 2.0
##[Download]
- 150Mb / s
- 170Mb / s
- 155Mb / s
##[Upload]
- 20Mb / s
- 20Mb / s
- 21Mb / s
#USB 3.0
##[Download]
- 50Mb / s
- 45Mb / s
- 26Mb / s
##[Upload]
- 18Mb / s
- 15Mb / s
- 17Mb / s
---------------------------------------------------------------------------------------------------------
Initially, I thought it was a measurement error, but initially the download speed was 1/3 slower than usb 2.0, but the more tests I performed, the worse it was. I'm counting on your help

RinCat, 2022-01-29:
I would say it could easily run over 300Mb/s down/up if your 5GHz WiFi is good. The only time you want to use USB3.0 is to achieve more than 480Mbps, which is the speed limit of USB 2.0.
This speed difference may be due to poor compatibility with the USB3.0 control chip. Some people find it doesn't even work at all.

by-speece, 2022-01-29:
Anyway, it's weird then I'll test it on my laptops as well and share the results here

RinCat, 2022-03-07:
The old no band name one I have just broken, so I get a new cheap T3U plus.
In my area,

USB 2.0, default
[image]

USB 3.0, `rtw_switch_usb_mode=1`
[image]

Not sure what caused it.