RinCat/RTL88x2BU-Linux-Driver #131: no wifi adapter shows up

9W5iw, 2022-04-04:
Bus 005 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]

COMFAST CF-812AC

http://en.comfast.com.cn/index.php?m=content&c=index&a=lists&catid=30#orientate

RinCat, 2022-04-04:
Hi,

`0bda:b812` is the default ID, it had already in the driver.
What is the result after run `modprobe 88x2bu`?

9W5iw, 2022-04-05:
modprobe: ERROR: could not insert '88x2bu': Operation not permitted
(both in normal mode and sudo, Im using ubuntu 22.04 beta)
Here are make and make install logs, there are some warnings in them:

[sodomake.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/8416194/sodomake.txt)
[sudomakeinstall.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/8416195/sudomakeinstall.txt)

RinCat, 2022-04-05:
Check `dmesg` output, your kernel may not allow unsigned modules to be loaded.

9W5iw, 2022-04-05:
I disabled secure boot and now it works. However it has problems connecting to some networks (I persistently cant connect to my thetering phone).
Is there a way to use it with secure boot turned on?

RinCat, 2022-04-05:
You have to compile and sign your own kernel in order to use it with secure boot. The distribution's private key is not publicly available, so you cannot sign self-compiled modules.