RinCat/RTL88x2BU-Linux-Driver #132: Would it work on an arm based device?

EDLLT, 2022-04-13:
Hey, I've tried installing your driver on my phone which as nethunter but whenever I check the driver being used for my wireless adapter
Archer T4U Plus, it appears to be using 88XXAU, however i'm 100% sure it's an 88XXBU and I double checked by installing the driver on my computer which worked flawlessly

How'd I get it to work on my phone which is an x64 arm based device? it keeps on getting that driver instead of this one and I have no idea how to get rid of it

RinCat, 2022-04-13:
Archer T4U Plus is 8812BU, so this driver works.
To use it on your phone that is highly depends on your phone OS. You need to find the corresponding kernel / kernel-headers and cross-compile the driver. Try to find tutorials on XDA. I only know this is possible, but I've never done it.

EDLLT, 2022-04-15:
> Archer T4U Plus is 8812BU, so this driver works. To use it on your phone that is highly depends on your phone OS. You need to find the corresponding kernel / kernel-headers and cross-compile the driver. Try to find tutorials on XDA. I only know this is possible, but I've never done it.

actually, the archer T4U Plus appears as 8822BU on my pc, but it works fine

however it doesn't on my phone
I installed the kernel from here
https://forum.xda-developers.com/t/kernel-nethunter-for-galaxy-s20-fe-5g-r8q-snapdragon.4205881/

Source code
https://github.com/Svirusx/Nethunter-Kernel-S20_FE_5G

I did some research in google and found this, is this what you meant by cross compiling?
https://forum.xda-developers.com/t/dev-guide-linux-comprehensive-guide-to-cross-compiling.2723240/

RinCat, 2022-04-15:
Yes. But for the kernel, maybe you ask the dev to do the job is much faster. As you need the same compilation environment as the kernel. I see they already added rtl88xxau and rtl8188eus.

closed 2022-04-22 by EDLLT