RinCat/RTL88x2BU-Linux-Driver #133: Output of Compiling code with linux-headers-5.15.0-27-generic UBUNTU 22.04 LTS

nsrshishir, 2022-04-27:
sudo make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.0-27-generic/build M=/home/niaj/rtl88x2bu  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-27-generic'
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_cmd.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_security.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_debug.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_io.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_ioctl_query.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_ioctl_set.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_ieee80211.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_mlme.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_mlme_ext.o
/home/niaj/rtl88x2bu/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/home/niaj/rtl88x2bu/core/rtw_mlme_ext.c:299:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  299 |                 if (rfctl->regd_name)
      |                    ^
/home/niaj/rtl88x2bu/core/rtw_mlme_ext.c:302:9: note: here
  302 |         default:
      |         ^~~~~~~
/home/niaj/rtl88x2bu/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/home/niaj/rtl88x2bu/core/rtw_mlme_ext.c:1670:38: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1670 |                         ptable->func = &OnAuthClient;
      |                         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/home/niaj/rtl88x2bu/core/rtw_mlme_ext.c:1672:9: note: here
 1672 |         case WIFI_ASSOCREQ:
      |         ^~~~
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_mi.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_wlan_util.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_vht.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_pwrctrl.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_rf.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_chplan.o
  CC [M]  /home/niaj/rtl88x2bu/core/monitor/rtw_radiotap.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_recv.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_sta_mgt.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_ap.o
  CC [M]  /home/niaj/rtl88x2bu/core/wds/rtw_wds.o
  CC [M]  /home/niaj/rtl88x2bu/core/mesh/rtw_mesh.o
  CC [M]  /home/niaj/rtl88x2bu/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/niaj/rtl88x2bu/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_xmit.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_p2p.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_rson.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_tdls.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_br_ext.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_iol.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_sreset.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_btcoex_wifionly.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_btcoex.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_beamforming.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_odm.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_rm.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_rm_fsm.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_ft.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_wnm.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_mbo.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_rm_util.o
  CC [M]  /home/niaj/rtl88x2bu/core/efuse/rtw_efuse.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_roch.o
  CC [M]  /home/niaj/rtl88x2bu/core/crypto/aes-internal.o
  CC [M]  /home/niaj/rtl88x2bu/core/crypto/aes-internal-enc.o
  CC [M]  /home/niaj/rtl88x2bu/core/crypto/aes-gcm.o
  CC [M]  /home/niaj/rtl88x2bu/core/crypto/aes-ccm.o
  CC [M]  /home/niaj/rtl88x2bu/core/crypto/aes-omac1.o
  CC [M]  /home/niaj/rtl88x2bu/core/crypto/ccmp.o
  CC [M]  /home/niaj/rtl88x2bu/core/crypto/gcmp.o
  CC [M]  /home/niaj/rtl88x2bu/core/crypto/aes-siv.o
  CC [M]  /home/niaj/rtl88x2bu/core/crypto/aes-ctr.o
  CC [M]  /home/niaj/rtl88x2bu/core/crypto/sha256-internal.o
  CC [M]  /home/niaj/rtl88x2bu/core/crypto/sha256.o
[... 156 lines cut ...]
  CC [M]  /home/niaj/rtl88x2bu/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /home/niaj/rtl88x2bu/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /home/niaj/rtl88x2bu/hal/phydm/halrf/halrf_psd.o
  CC [M]  /home/niaj/rtl88x2bu/hal/phydm/rtl8822b/halhwimg8822b_bb.o
  CC [M]  /home/niaj/rtl88x2bu/hal/phydm/rtl8822b/halhwimg8822b_mac.o
  CC [M]  /home/niaj/rtl88x2bu/hal/phydm/halrf/rtl8822b/halrf_8822b.o
  CC [M]  /home/niaj/rtl88x2bu/hal/phydm/rtl8822b/phydm_hal_api8822b.o
  CC [M]  /home/niaj/rtl88x2bu/hal/phydm/halrf/rtl8822b/halhwimg8822b_rf.o
  CC [M]  /home/niaj/rtl88x2bu/hal/phydm/halrf/rtl8822b/halrf_iqk_8822b.o
  CC [M]  /home/niaj/rtl88x2bu/hal/phydm/halrf/rtl8822b/halrf_rfk_init_8822b.o
  CC [M]  /home/niaj/rtl88x2bu/hal/phydm/rtl8822b/phydm_regconfig8822b.o
  CC [M]  /home/niaj/rtl88x2bu/hal/phydm/rtl8822b/phydm_rtl8822b.o
  CC [M]  /home/niaj/rtl88x2bu/hal/phydm/txbf/haltxbf8822b.o
  CC [M]  /home/niaj/rtl88x2bu/hal/btc/halbtc8822bwifionly.o
  CC [M]  /home/niaj/rtl88x2bu/hal/btc/halbtc8822b1ant.o
  CC [M]  /home/niaj/rtl88x2bu/hal/btc/halbtc8822b2ant.o
  CC [M]  /home/niaj/rtl88x2bu/platform/platform_ops.o
  CC [M]  /home/niaj/rtl88x2bu/core/rtw_mp.o
  LD [M]  /home/niaj/rtl88x2bu/88x2bu.o
  MODPOST /home/niaj/rtl88x2bu/Module.symvers
  CC [M]  /home/niaj/rtl88x2bu/88x2bu.mod.o
  LD [M]  /home/niaj/rtl88x2bu/88x2bu.ko
  BTF [M] /home/niaj/rtl88x2bu/88x2bu.ko
Skipping BTF generation for /home/niaj/rtl88x2bu/88x2bu.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-27-generic'

RinCat, 2022-04-27:
Hi, I do not see any error here. `BTF generation` can be safe skipped.

nsrshishir, 2022-04-28:
Yes there are no errors. but several warnings are there, which I thought should share with you.

RinCat, 2022-04-28:
Oh, thanks. Those warnings are expected, since some compiler versions do not recognize the fall through comments in the code.