RinCat/RTL88x2BU-Linux-Driver #134: Intial Connection to a new IP extremly slow even tough download speed is fast

ZagButNoZig, 2022-04-28:
With the latest git driver and an Archer T3U Plus I get pretty fast internet speed (almost as fast as my phone over WiFi).
[image]

But loading a new web page is extremely slow. By extremely slow I mean easily 5 to 10 seconds or more. 
I also tried to ping google.com and found that after executing the ping command the first messages takes easily 8 seconds to appear even tough it says that the packet only took a few ms to send.
[image]

As you can see the total time of the packets isn't even close to the total time the `ping` command shows near the bottom. So I think there is something wrong with the driver sending packets. 

I have also checked the `Network` tab in Firefox when loading a web page and each get request takes multiple seconds. 
[image]

But I think this time comes from the driver not sending the message in time, as my tested ping is very low and over ethernet or my phone I get much faster load times.

I am running : 

[image]

Let me know if you need any further information and thanks for the great work on this driver!

ZagButNoZig, 2022-04-28:
This is when pinging my local router. I get some pretty fishy times. And again only the first few packets are really slow and the thing gradually speed up.
```
icmp_seq=1 ttl=64 time=1547 ms
icmp_seq=2 ttl=64 time=544 ms
icmp_seq=3 ttl=64 time=2312 ms
icmp_seq=4 ttl=64 time=1287 ms
icmp_seq=5 ttl=64 time=273 ms
icmp_seq=6 ttl=64 time=1794 ms
icmp_seq=7 ttl=64 time=768 ms
icmp_seq=8 ttl=64 time=82.0 ms
icmp_seq=9 ttl=64 time=2.30 ms
icmp_seq=10 ttl=64 time=9.61 ms
icmp_seq=11 ttl=64 time=5.07 ms
icmp_seq=12 ttl=64 time=2.78 ms
icmp_seq=13 ttl=64 time=4.79 ms
icmp_seq=14 ttl=64 time=9.33 ms
```

[image]
Similar results when pinging `1.1.1.1`(pinging from Europe).

I let `ping ROUTER_IP` run for a bit longer and it seams like roughly 20% of packets are really slow (500-2000ms) and the rest is below 20ms.

RinCat, 2022-04-29:
Maybe you can try https://www.wireshark.org/ to see if there is any issue on the connections. Also check `dmesg`.

ZagButNoZig, 2022-04-29:
Ok so... I don't know what changed because I don't remember changing anything but now for some random reason the stability issue is fixed. 
[image]

But for same random reason the overall connection speed is much lower :
[image]
Tested on my phone to make sure its not the routers fault and indeed it seems to only have slowed down with the Archer T3U Plus.
Anyway, the situation is much better now, as my internet is now slower but still reasonably fast but much much more consistent.
I don't know if there is a way to fix the slower speed though.

RinCat, 2022-04-29:
I guess the wifi configuration may be a little incompatible. You can try to change the channel.
The unknown element is caused by having an unsupported extension protocol, which can usually be safely ignored.

ZagButNoZig, 2022-05-07:
Thanks!

I will have to test a bit more, it seems to be stable at max speed currently but I will have to do further testing.
I find it pretty funny that this driver is faster and more stable than the official windows driver, wonder if that is due to the custom patches or linux networking just being better. I will close the issue once I'm done.