RinCat/RTL88x2BU-Linux-Driver #135: OpenWrt Kernel(5.15),build failed with "88x2bu.ko is missing"

xspeed1989, 2022-05-02:
ERROR: module '/home/xx/Desktop/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-20-a543f246/88x2bu.ko' is missing.

RinCat, 2022-05-02:
Hi, any extra information? I can only say you may have failed compile or install this driver, so something cannot find it.

xspeed1989, 2022-05-02:
> Hi, any extra information? I can only say you may have failed compile or install this driver, so something cannot find it.
I found the Makefile of OpenWrt package is wrong, So it didn't compile the kernel module,
Unfortunately, I can't write the Makefile for OpenWrt package

RinCat, 2022-05-02:
You may try cross compile you self and put `88x2bu.ko` to your OpenWRT system. The current Makefile doesn't take OpenWRT into account at all.

xspeed1989, 2022-05-02:
> You may try cross compile you self and put `88x2bu.ko` to your OpenWRT system. The current Makefile doesn't take OpenWRT into account at all.

I will spend time studying how to write an openwrt package with a exists repo, then compile it.

closed 2022-05-02 by xspeed1989

valBarajas, 2022-12-05:
someone already managed to make it work in OpenWrt?

RinCat, 2022-12-05:
@valBarajas you need build it yourself.

lukenvn, 2024-03-21:
@RinCat  can you please show me some instruction how to build and make it work on openwrt? Thank you very much

RinCat, 2024-03-22:
> @RinCat can you please show me some instruction how to build and make it work on openwrt? Thank you very much

You will need to build openwrt yourself and add something like [this](https://github.com/hkucs-makerlab/openwrt-rtl8812bu-package/tree/openwrt-22.03) . I never use it with openwrt, but I know some people was doing it.

mirobiala, 2024-03-23:
> @RinCat  can you please show me some instruction how to build and make it work on openwrt? Thank you very much

Hi @lukenvn,
I created a patch for the @RinCat 's driver for OpenWRT. Here is the repository: [https://github.com/mirobiala/rtl88x2bu-cl](https://github.com/mirobiala/rtl88x2bu-cl). You can give it a try 😉.

p.s : @RinCat thank you for the best driver for rtl88x2bu that actually works!

lukenvn, 2024-03-26:
Thank you guys for your hard work and answers @RinCat @mirobiala . I managed to rebuild the openwrt with the driver as the repo of @mirobiala  said, after 3 hours the job done, I can boot up the openwrt on my rashperry pi, but when I plugged  the wifi adapter( T3U Plus) to my rashpberry, and type `iw dev` or `iwinfo` there are no new wifi interface appear there.
Is there anyway that i can check that the building is failed or i need to do something more after install the openwrt?

RinCat, 2024-03-26:
@lukenvn run `modprobe 88x2bu` and see result, also check `dmesg`

lukenvn, 2024-03-26:
> dmesg

here the result, should i rebuild the image again ?
```
--------------------------------------------------
root@OpenWrt:~# modprobe 88x2bu
failed to find a module named 88x2bu
root@OpenWrt:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.137 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23630-842932a63d) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Tue Nov 14 13:38:11 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 17 pages/cpu s31832 r8192 d29608 u69632
[    0.000000] pcpu-alloc: s31832 r8192 d29608 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028848
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=E4:5F:01:0A:8C:BA vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 7941576K/8245248K available (8320K kernel code, 912K rwdata, 1296K rodata, 448K init, 554K bss, 238136K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: 0xffffffc0083dea60
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000223] Console: colour dummy device 80x25
[    0.000517] printk: console [tty1] enabled
[    0.000563] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000588] pid_max: default: 32768 minimum: 301
[    0.000918] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.001072] Mountpoint-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.001715] cgroup: Disabling memory control group subsystem
[    0.002789] rcu: Hierarchical SRCU implementation.
[    0.003504] smp: Bringing up secondary CPUs ...
[    0.003929] Detected PIPT I-cache on CPU1
[    0.003999] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.004494] Detected PIPT I-cache on CPU2
[    0.004535] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.004982] Detected PIPT I-cache on CPU3
[    0.005024] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.005099] smp: Brought up 1 node, 4 CPUs
[    0.005157] SMP: Total of 4 processors activated.
[    0.005170] CPU features: detected: 32-bit EL0 Support
[    0.005182] CPU features: detected: CRC32 instructions
[    0.005251] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[... 183 lines cut ...]
[    5.221014] NET: Registered PF_PPPOX protocol family
[    5.237310] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.246152] usbcore: registered new interface driver brcmfmac
[    5.251977] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin failed with error -2
[    5.263634] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin
[    5.278243] usbcore: registered new interface driver rt2800usb
[    5.284245] kmodloader: done loading kernel modules from /etc/modules.d/*
[... cut ...]
[    6.725346] brcmfmac mmc1:0001:1 phy0-sta0: renamed from wlan0
[    7.527837] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[    8.117578] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-sta0: link becomes ready
[  192.762182] bcmgenet fd580000.ethernet eth0: Link is Down
[  192.767804] br-lan: port 1(eth0) entered disabled state
[11089.984111] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[11089.992358] br-lan: port 1(eth0) entered blocking state
[11089.997613] br-lan: port 1(eth0) entered forwarding state

```

mirobiala, 2024-03-26:
@lukenvn, try `modprobe -v rtl88x2bu`, also did you enable `Kernel modules->Wireless Drivers->kmod-rtl88x2bu-cl` in menuconfig?

lukenvn, 2024-03-26:
hi @mirobiala  yeah i missed this step `Kernel modules->Wireless Drivers->kmod-rtl88x2bu-cl`.  Now my dyi wifi repeater worked like a charm. Thank you guys very much I learn a lot with this project and you guy really awesome @RinCat @mirobiala