RinCat/RTL88x2BU-Linux-Driver #139: installation

azerool, 2022-05-21:
hi is there any installation quide . . ??
almost 3 day i try . . still not working an got manny error

(MY OS)
RETTY_NAME="Kali GNU/Linux Rolling"
NAME="Kali GNU/Linux"
ID=kali
VERSION="2022.2"
VERSION_ID="2022.2"
VERSION_CODENAME="kali-rolling"
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="https://www.kali.org/"
SUPPORT_URL="https://forums.kali.org/"
BUG_REPORT_URL="https://bugs.kali.org/"

RinCat, 2022-05-21:
Hi, post error logs are more useful.

azerool, 2022-05-21:
Can u send me link how to install ill get back to you

Iam trying difrent step aight now it need (VER)

iam just newbie

RinCat, 2022-05-21:
The README included base install information https://github.com/RinCat/RTL88x2BU-Linux-Driver#how-to-use-this-kernel-module= .

azerool, 2022-05-21:
iam doing this aight now iam using Archer T4U v3

1.git clone https://github.com/cilynx/rtl88x2bu.git
<https://www.youtube.com/redirect?event=video_description&redir_token=QUFFLUhqbG5GX2xHanlFZk9lVlFfYjVmTTU3VXJkUXcxUXxBQ3Jtc0tsNm1rMm5Jem05bXNraTFZa3ZlRlo5RnFSdnlSTDltZW41R1AzQi1tQV8xWHhTY2ZzbXkzU1ZyTGwtMlNDeE5Kd3Z1UklINFJaZzItUGo2cnJpR1VWMGtLVEJPUFc2QXBCVHVxdTMwWmw3LXU5blE1OA&q=https%3A%2F%2Fgithub.com%2Fcilynx%2Frtl88x2bu.git&v=x5ccyhrK9NU>
2. cd rtl88x2bu 3. VER=$(sed -n 's/\PACKAGE_VERSION="\(.*\)"/\1/p'
dkms.conf) 4. sudo rsync -rvhP ./ /usr/src/rtl88x2bu-${VER} 5 sudo dkms add
-m rtl88x2bu -v ${VER} 6 sudo dkms build -m rtl88x2bu -v ${VER} 7 sudo dkms
install -m rtl88x2bu -v ${VER} 8 sudo modprobe 88x2bu

step no 6 got this error
Error! Arguments <module> and <module-version> are not specified.
Usage: add <module>/<module-version> or
       add -m <module>/<module-version> or
       add -m <module> -v <module-version>

RinCat, 2022-05-21:
I guess you are not run them in the same terminal, so you need to run `VER=$(sed -n 's/\PACKAGE_VERSION="\(.*\)"/\1/p' dkms.conf)` again before `sudo dkms XXXXXXXX`

azerool, 2022-05-21:
new error
Error! Your kernel headers for kernel 5.10.102.1-generic+ cannot be found
at /lib/modules/5.10.102.1-generic+/build or
/lib/modules/5.10.102.1-generic+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.

RinCat, 2022-05-21:
run 
```
sudo apt-get update
sudo apt-get install –y linux-headers-$(uname -r)
```

azerool, 2022-05-21:
sudo apt-get install –y linux-headers-$(uname -r)

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package –y
E: Unable to locate package linux-headers-5.10.102.1-generic
E: Couldn't find any package by glob 'linux-headers-5.10.102.1-generic'
E: Couldn't find any package by regex 'linux-headers-5.10.102.1-generic'
😞😞😞😞😞😞😞

azerool, 2022-05-21:
i already  try wsl2 kernel stll this thing happen

RinCat, 2022-05-21:
Oh, kali is rolling dist, so you need to do full system update before doing that. 
```
sudo apt update
sudo apt upgrade
```

azerool, 2022-05-21:
the old one

Linux version 5.10.102.1-microsoft-standar-WSL2 (x86_64-msft-linux-gcc
(GCC) 9.3.0, GNU ld (GNU Binutils) 2.34.0.20200220) #1

update kernel
inux version 5.10.102.1-generic+ ***@***.***) (gcc (Debian
11.3.0-1) 11.3.0, GNU ld (GNU Binutils for De>

RinCat, 2022-05-21:
emmm, why you try to use WSL kernel? The system is in Windows?

azerool, 2022-05-21:
ya coz this is the 3rd time iam reset my whole system
what you recommend me to do?

i will try

RinCat, 2022-05-21:
First, is this one in a VM or WSL?

azerool, 2022-05-21:
before i forget iam using kali-win-kex aight now

azerool, 2022-05-21:
WSL2. .  RUNNING ON KALI-WIN-KEX
log in - kex --win -s

azerool, 2022-05-21:
when i try to install according ur step, step 1 already have problem

"make"

"make clean"

RinCat, 2022-05-21:
You do not have linux kernel headers, so you cannot build any kernel modules. 
From WSL discussion, you have to build your own kernel in order to build modules.
https://github.com/MicrosoftDocs/WSL/issues/1253#issuecomment-931719333

azerool, 2022-05-21:
> MicrosoftDocs/WSL#1253 (comment)

when come to this iam not sure what ive to do
https://docs.microsoft.com/en-us/windows/wsl/wsl-config#configure-global-options-with-wslconfig

RinCat, 2022-05-21:
It completely unrelated to the link you post.

azerool, 2022-05-21:
i already dont it as i send u after i  configure wsl2. . .

azerool, 2022-05-21:
this thiing happen when i do WSL 2nd step
make KCONFIG_CONFIG=Microsoft/config-wsl
mkdir: cannot create directory ‘.tmp_30533’: Permission denied
mkdir: cannot create directory ‘.tmp_30535’: Permission denied
mkdir: cannot create directory ‘.tmp_30537’: Permission denied
mkdir: cannot create directory ‘.tmp_30539’: Permission denied
mkdir: cannot create directory ‘.tmp_30541’: Permission denied
mkdir: cannot create directory ‘.tmp_30543’: Permission denied
mkdir: cannot create directory ‘.tmp_30546’: Permission denied
mkdir: cannot create directory ‘.tmp_30548’: Permission denied
mkdir: cannot create directory ‘.tmp_30550’: Permission denied
mkdir: cannot create directory ‘.tmp_30552’: Permission denied
mkdir: cannot create directory ‘.tmp_30554’: Permission denied
mkdir: cannot create directory ‘.tmp_30556’: Permission denied
mkdir: cannot create directory ‘.tmp_30558’: Permission denied
mkdir: cannot create directory ‘.tmp_30560’: Permission denied
mkdir: cannot create directory ‘.tmp_30562’: Permission denied
mkdir: cannot create directory ‘.tmp_30564’: Permission denied
mkdir: cannot create directory ‘.tmp_30566’: Permission denied
mkdir: cannot create directory ‘.tmp_30568’: Permission denied
mkdir: cannot create directory ‘.tmp_30570’: Permission denied
mkdir: cannot create directory ‘.tmp_30572’: Permission denied
mkdir: cannot create directory ‘.tmp_30574’: Permission denied
mkdir: cannot create directory ‘.tmp_30576’: Permission denied
mkdir: cannot create directory ‘.tmp_30578’: Permission denied
arch/x86/Makefile:147: CONFIG_X86_X32 enabled but no binutils support
mkdir: cannot create directory ‘.tmp_30580’: Permission denied
mkdir: cannot create directory ‘.tmp_30582’: Permission denied
mkdir: cannot create directory ‘.tmp_30587’: Permission denied
mkdir: cannot create directory ‘.tmp_30589’: Permission denied
mkdir: cannot create directory ‘.tmp_30591’: Permission denied
mkdir: cannot create directory ‘.tmp_30593’: Permission denied
mkdir: cannot create directory ‘.tmp_30595’: Permission denied
mkdir: cannot create directory ‘.tmp_30597’: Permission denied
mkdir: cannot create directory ‘.tmp_30599’: Permission denied
mkdir: cannot create directory ‘.tmp_30601’: Permission denied
mkdir: cannot create directory ‘.tmp_30603’: Permission denied
mkdir: cannot create directory ‘.tmp_30605’: Permission denied
mkdir: cannot create directory ‘.tmp_30607’: Permission denied
mkdir: cannot create directory ‘.tmp_30610’: Permission denied
mkdir: cannot create directory ‘.tmp_30612’: Permission denied
mkdir: cannot create directory ‘.tmp_30615’: Permission denied
mkdir: cannot create directory ‘.tmp_30617’: Permission denied
mkdir: cannot create directory ‘.tmp_30619’: Permission denied
mkdir: cannot create directory ‘.tmp_30621’: Permission denied
mkdir: cannot create directory ‘.tmp_30623’: Permission denied
mkdir: cannot create directory ‘.tmp_30625’: Permission denied
mkdir: cannot create directory ‘.tmp_30627’: Permission denied
mkdir: cannot create directory ‘.tmp_30629’: Permission denied
mkdir: cannot create directory ‘.tmp_30631’: Permission denied
mkdir: cannot create directory ‘.tmp_30633’: Permission denied
mkdir: cannot create directory ‘.tmp_30635’: Permission denied
mkdir: cannot create directory ‘.tmp_30638’: Permission denied
mkdir: cannot create directory ‘.tmp_30641’: Permission denied
/bin/sh: 1: cannot create include/config/.kernel.release.tmp: Permission
denied
  UPD     include/config/kernel.release
mv: cannot stat 'include/config/.kernel.release.tmp': No such file or
directory
make: *** [Makefile:1192: include/config/kernel.release] Error 1

RinCat, 2022-05-21:
You may check https://github.com/microsoft/WSL/issues/4836

azerool, 2022-05-21:
> You may check microsoft/WSL#4836

thakz ya, keep replying my msg
ill try to figure it out,

azerool, 2022-05-26:
hi Rin may i ask you .. the driver has been installed
but i could  not get wireless connection

i attech some detail .

hope u have time to reply, tq

RinCat, 2022-05-26:
Hi, you cannot attached files in github issues. Post it to https://gist.github.com/ .

azerool, 2022-05-29:
hi
just wanna inform you, my wlan is enable.. .  thankz you

azerool, 2022-10-11:
i did that already
not sure what i mist

azerool, 2022-10-11:
this are the error when
sudo apt upgrade

Hit:1 http://kali.cs.nctu.edu.tw/kali kali-rolling InRelease
Hit:2 http://kali.cs.nctu.edu.tw/kali kali-last-snapshot InRelease
Hit:3 http://kali.cs.nctu.edu.tw/kali kali-experimental InRelease
Hit:4 http://kali.cs.nctu.edu.tw/kali kali-bleeding-edge InRelease
Reading package lists... Done
W: Target Packages (main/binary-amd64/Packages) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Packages (main/binary-all/Packages) is configured multiple times
in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Translations (main/i18n/Translation-en_US) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Translations (main/i18n/Translation-en) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Contents-deb (main/Contents-amd64) is configured multiple times
in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Contents-deb (main/Contents-all) is configured multiple times in
/etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Packages (contrib/binary-amd64/Packages) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Packages (contrib/binary-all/Packages) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Translations (contrib/i18n/Translation-en_US) is configured
multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Translations (contrib/i18n/Translation-en) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Contents-deb (contrib/Contents-amd64) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Contents-deb (contrib/Contents-all) is configured multiple times
in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Packages (non-free/binary-amd64/Packages) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Packages (non-free/binary-all/Packages) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Translations (non-free/i18n/Translation-en_US) is configured
multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Translations (non-free/i18n/Translation-en) is configured
multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Contents-deb (non-free/Contents-amd64) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Contents-deb (non-free/Contents-all) is configured multiple times
in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Packages (main/binary-amd64/Packages) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Packages (main/binary-all/Packages) is configured multiple times
in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Translations (main/i18n/Translation-en_US) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Translations (main/i18n/Translation-en) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Contents-deb (main/Contents-amd64) is configured multiple times
in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Contents-deb (main/Contents-all) is configured multiple times in
/etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Packages (contrib/binary-amd64/Packages) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Packages (contrib/binary-all/Packages) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Translations (contrib/i18n/Translation-en_US) is configured
multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Translations (contrib/i18n/Translation-en) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Contents-deb (contrib/Contents-amd64) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Contents-deb (contrib/Contents-all) is configured multiple times
in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Packages (non-free/binary-amd64/Packages) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Packages (non-free/binary-all/Packages) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Translations (non-free/i18n/Translation-en_US) is configured
multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Translations (non-free/i18n/Translation-en) is configured
multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Contents-deb (non-free/Contents-amd64) is configured multiple
times in /etc/apt/sources.list:2 and /etc/apt/sources.list:4
W: Target Contents-deb (non-free/Contents-all) is configured multiple times
in /etc/apt/sources.list:2 and /etc/apt/sources.list:4

RinCat, 2022-10-12:
You had same repo in `/etc/apt/sources.list` line 2 and 4, remove one of them.