RinCat/RTL88x2BU-Linux-Driver #140: THANK YOU!!!!!!  YOU ARE THE BEST!!!

TCROC, 2022-05-21:
Just wanted to let you know you are a true hero!!  I've been struggling all morning to get my TP LINK AC1300 working since the latest Ubuntu update.  This driver did the trick!  Feel free to close when you've read this! :)

brisback, 2022-08-11:
Can you post your terminal commands that you used? I am trying to get this driver running on the latest version of POP OS 22.04.

charlesfoster, 2022-10-24:
@brisback did you have any luck? I'm in the same situation now...

RinCat, 2022-10-25:
> @brisback did you have any luck? I'm in the same situation now...

Have you checked the README yet?

charlesfoster, 2022-10-25:
Thanks for the message. Yes, I checked the README, and the solutions weren't working. However, since posting my question, I found other issues stopping me from installing this driver (including missing cfg80211.ko module). In my case, I needed to reinstall the Linux 6.02 kernel, then I was able to install the rtl88x2bu driver successfully using your instructions. Cheers

TCROC, 2022-12-13:
> 

Just noticed this.  Sorry for late reply.  I believe all I did was follow along in the README