RinCat/RTL88x2BU-Linux-Driver #142: Install Troubles

MrSlipicus, 2022-06-29:
I was successfully (or so it reported) able to run the dkms auto install. Afterwards lsusb reports. 

pi@raspberrypi:/usr/src/rtl88x2bu-git $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 1a2c:2124 China Resource Semico Co., Ltd Keyboard
Bus 001 Device 004: ID 413c:301a Dell Computer Corp. Dell MS116 Optical Mouse
Bus 001 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

This is my currently installed and running Kernel. This is on an RPI 4 Model B. 

I have been having real trouble getting any driver for this device to install. It is a Techkey AC 1300Mbps usb 3.0 adapter. 

MrEngmans script reports it is RTL8822BU however all of the software install options report RTL8812BU. 

MrEngmans also says there are no drivers for my current Kernel which is 

Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux

Am I barking up the wrong tree?

RinCat, 2022-07-02:
RTL8812BU and RTL8822BU are the same thing in term of WiFi, one of them has Bluetooth support, but not in this driver.
Try use `modprobe 88x2bu` to load the driver. If it not works, check `dmesg`kernel logs.

RinCat, 2022-07-05:
I just tested and success dump data using `tcpdump` in monitor mode. So I had no idea why it shows not support in your system.