RinCat/RTL88x2BU-Linux-Driver #143: Linux kernel 5.19-RC4: build failed

Programmist11180, 2022-07-04:
Hello.
I just installed Linux kernel 5.19-RC4 (Debian experimental, linux-image-5.19.0-rc4-amd64-unsigned), and headers.
But module build failed.

> DKMS make.log for rtl88x2bu-git for kernel 5.19.0-rc4-amd64 (x86_64)
Пн 04 июл 2022 22:11:57 MSK
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.0-rc4-amd64/build M=/var/lib/dkms/rtl88x2bu/git/build  modules
make[1]: вход в каталог «/usr/src/linux-headers-5.19.0-rc4-amd64»
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_security.o
/bin/sh: 1: /usr/src/linux-headers-5.19.0-rc4-common/scripts/check-local-export: not found
make[2]: *** [/usr/src/linux-headers-5.19.0-rc4-common/scripts/Makefile.build:254: /var/lib/dkms/rtl88x2bu/git/build/core/rtw_security.o] Ошибка 127
make[2]: *** Удаляется файл «/var/lib/dkms/rtl88x2bu/git/build/core/rtw_security.o»
make[2]: *** Ожидание завершения заданий…
/bin/sh: 1: /usr/src/linux-headers-5.19.0-rc4-common/scripts/check-local-export: not found
make[2]: *** [/usr/src/linux-headers-5.19.0-rc4-common/scripts/Makefile.build:254: /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.o] Ошибка 127
make[2]: *** Удаляется файл «/var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.o»
make[1]: *** [/usr/src/linux-headers-5.19.0-rc4-common/Makefile:1867: /var/lib/dkms/rtl88x2bu/git/build] Ошибка 2
make[1]: выход из каталога «/usr/src/linux-headers-5.19.0-rc4-amd64»
make: *** [Makefile:2454: modules] Ошибка 2

File /usr/src/linux-headers-5.19.0-rc4-common/scripts/check-local-export does not exist.

RinCat, 2022-07-04:
Maybe a bug in Debian? It works for my vanilla kernel. `scripts/check-local-export` exists in my kernel sources.

Programmist11180, 2022-07-05:
Debian don't have file `check-local-export` at all
[search result on packages.debian.org](https://packages.debian.org/search?suite=default&section=all&arch=any&searchon=contents&keywords=check-local-export)

RinCat, 2022-07-05:
I think you have to report this to Debian, as the build scripts are from the kernel, not in this driver.
And it works for my vanilla 5.19 RC4, so maybe Debian missing something.

Programmist11180, 2022-07-11:
I have created a report [https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1014633](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1014633)