RinCat/RTL88x2BU-Linux-Driver #144: excessive status logging?

mgchristensen, 2022-07-09:
Built and loaded module per the README.md. Approximately four or five times per second, what appears to be a status report for the device is written to the log, as shown by dmesg.

[309021.137326] RTW: ============ STA [8a:2a:xx:xx:xx:xx]  ===================
[309021.137341] RTW: mac_id : 0
[309021.137344] RTW: wireless_mode : 0x44
[309021.137347] RTW: mimo_type : 2
[309021.137350] RTW: static smps : N
[309021.137352] RTW: bw_mode : 40MHz, ra_bw_mode : 40MHz
[309021.137356] RTW: rate_id : 9
[309021.137358] RTW: rssi : 49 (%), rssi_level : 5
[309021.137362] RTW: is_support_sgi : Y, is_vht_enable : Y
[309021.137365] RTW: disable_ra : N, disable_pt : N
[309021.137368] RTW: is_noisy : N
[309021.137370] RTW: txrx_state : 2
[309021.137373] RTW: curr_tx_rate : VHT2SMCS9 (L)
[309021.137376] RTW: curr_tx_bw : 40MHz
[309021.137378] RTW: curr_retry_ratio : 0
[309021.137381] RTW: ra_mask : 0x00000000fffffff0

Likely something simple I missed, but I am unable to see it.  Please tell me what I miossed
Fedora 35, kernel 5.18.9-100.fc35.x86_64

lsmod | grep 88x2 shows:
88x2bu               3698688  0
cfg80211             1060864  2 wl,88x2bu

Thanks.

mgchristensen, 2022-07-09:
Additional info:
after:
sudo modprobe -r 88x2bu
dmesg shows:
[354319.837631] usbcore: deregistering interface driver rtl88x2bu
then, after:
sudo modprobe 88x2bu rtw_switch_usb_mode=1 rtw_drv_log_level=0
dmesg shows:
[354348.168161] usbcore: registered new interface driver rtl88x2bu
[354348.177728] rtl88x2bu 3-2:1.0 wlp0s20u2: renamed from wlan0
[354348.806667] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[354353.905528] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[354357.107618] IPv6: ADDRCONF(NETDEV_CHANGE): wlp0s20u2: link becomes ready

So, the default build of the driver enables a higher (debug) log level?

RinCat, 2022-07-10:
It has already been reduced to 3 (WARNING) from Realtek default 4 (INFO).
You can check/change log level in `/proc/net/rtl88x2bu/log_level `