RinCat/RTL88x2BU-Linux-Driver #146: D-Link DWA-T185 cannot use

YW0330, 2022-07-30:
Hello, I following the commands in the README.md on Ubuntu 16.04, but it's not successful, DWA_T185 doesn't work.  Following information of my system:
- `uname -r`
```
4.15.0-142-generic
```
-  `lsusb`
```
Bus 002 Device 008: ID 2001:3322 D-Link Corp. 
```
- `lsmod`
```
Module                  Size  Used by
88x2bu               2441216  0
cfg80211              634880  1 88x2bu
```
- `dmesg | grep 88x2bu`
```
[   65.017652] 88x2bu: loading out-of-tree module taints kernel.
[   65.018111] 88x2bu: module verification failed: signature and/or required key missing - tainting kernel
[   65.021614] usbcore: registered new interface driver rtl88x2bu
```
Could you give some suggestion to me, thank you!

RinCat, 2022-07-30:
You need disable Secure Boot in order to load out-of-tree module. See https://wiki.ubuntu.com/UEFI/SecureBoot/DKMS

YW0330, 2022-07-30:
Secure Boot is still disable. But boot mode is Legacy, is this influence the result?
[image]
Thanks

RinCat, 2022-07-30:
OK, I just found out your device USB ID is unknown. It is not the same as DWA-185.
I just pushed a update, try again.

YW0330, 2022-07-30:
The same problem when I install the module :cry: .
```
[   60.979857] 88x2bu: loading out-of-tree module taints kernel.
[   60.980455] 88x2bu: module verification failed: signature and/or required key missing - tainting kernel
[   60.983976] usbcore: registered new interface driver rtl88x2bu
```

RinCat, 2022-07-30:
That is not an error, do you able to see the interface?

YW0330, 2022-07-30:
- `lsusb`
```
Bus 002 Device 005: ID 2001:3322 D-Link Corp.
```
This command?

RinCat, 2022-07-30:
`ip link` and see if there is something like `wlan0`.

YW0330, 2022-07-30:
This is all result of `ip link`
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: enp1s0f0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether 94:c6:91:58:5e:cb brd ff:ff:ff:ff:ff:ff
```

RinCat, 2022-07-30:
Did you unloaded the old module first or reboot?

YW0330, 2022-07-30:
I follow the following steps:
1. `sudo modprobe -r 88x2bu`
2. `sudo make install`
3. `reboot`
4. `sudo modprobe 88x2bu rtw_switch_usb_mode=1`

RinCat, 2022-07-30:
Try without `rtw_switch_usb_mode=1`? Is there anything else in the dmesg log?

YW0330, 2022-07-30:
This is directly log without reboot
```
dmesg | grep 88x
[   46.202612] 88x2bu: loading out-of-tree module taints kernel.
[   46.203449] 88x2bu: module verification failed: signature and/or required key missing - tainting kernel
[   46.207434] usbcore: registered new interface driver rtl88x2bu
[ 1211.205326] usbcore: deregistering interface driver rtl88x2bu
[ 1216.568612] usbcore: registered new interface driver rtl88x2bu
```

RinCat, 2022-07-30:
The module log starts with `RTW:`

YW0330, 2022-07-30:
Sorry, I don't know the `RTW` meaning, so I paste the whole information of `dmesg`
```
[    0.000000] microcode: microcode updated early to revision 0xde, date = 2020-05-25
[    0.000000] Linux version 4.15.0-142-generic (buildd@lgw01-amd64-039) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.12)) #146~16.04.1-Ubuntu SMP Tue Apr 13 09:27:15 UTC 2021 (Ubuntu 4.15.0-142.146~16.04.1-generic 4.15.18)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-142-generic root=UUID=43ddbb42-831d-4e2c-9151-7f6a2fb9f6cc ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x00000000000997ff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000099800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000008aacdfff] usable
[    0.000000] BIOS-e820: [mem 0x000000008aace000-0x000000008be20fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008be21000-0x000000008be9dfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000008be9e000-0x000000008c3c8fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008c3c9000-0x000000008cc0dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008cc0e000-0x000000008cc0efff] usable
[    0.000000] BIOS-e820: [mem 0x000000008cc0f000-0x000000008fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000046dffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 3.1.1 present.
[    0.000000] DMI: Acer Veriton M4660G/Veriton M4660G, BIOS R01-A2 09/13/2018
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x46e000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 00FF000000 mask 7FFF000000 write-protect
[    0.000000]   1 base 0000000000 mask 7F80000000 write-back
[    0.000000]   2 base 0080000000 mask 7FF0000000 write-back
[    0.000000]   3 base 0100000000 mask 7F00000000 write-back
[    0.000000]   4 base 0200000000 mask 7E00000000 write-back
[    0.000000]   5 base 0400000000 mask 7C00000000 write-back
[    0.000000]   6 base 008D800000 mask 7FFF800000 uncachable
[    0.000000]   7 base 008E000000 mask 7FFE000000 uncachable
[    0.000000]   8 base 0090000000 mask 7FF0000000 write-combining
[    0.000000]   9 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000000] e820: last_pfn = 0x8cc0f max_arch_pfn = 0x400000000
[    0.000000] found SMP MP-table at [mem 0x000fcdb0-0x000fcdbf]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Using GB pages for direct mapping
[    0.000000] RAMDISK: [mem 0x31438000-0x34a13fff]
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000F05B0 000024 (v02 ACRSYS)
[    0.000000] ACPI: XSDT 0x000000008BE3A0B0 0000DC (v01 ACRSYS ACRPRDCT 01072009 AMI  00010013)
[    0.000000] ACPI: FACP 0x000000008BE7B4C8 000114 (v06 ACRSYS ACRPRDCT 01072009 AMI  00010013)
[    0.000000] ACPI: DSDT 0x000000008BE3A218 0412AF (v02 ACRSYS ACRPRDCT 01072009 INTL 20160527)
[    0.000000] ACPI: FACS 0x000000008C376080 000040
[... 850 lines cut ...]
[    3.791377] input: HDA Intel PCH Front Mic as /devices/pci0000:00/0000:00:1f.3/sound/card0/input14
[    3.791422] input: HDA Intel PCH Line as /devices/pci0000:00/0000:00:1f.3/sound/card0/input15
[    3.791465] input: HDA Intel PCH Line Out as /devices/pci0000:00/0000:00:1f.3/sound/card0/input16
[    3.791509] input: HDA Intel PCH Front Headphone as /devices/pci0000:00/0000:00:1f.3/sound/card0/input17
[    3.791561] input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input18
[    3.791626] input: HDA Intel PCH HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input19
[... cut ...]
[   16.962630] r8169 0000:01:00.0 enp1s0f0: link up
[   16.962636] IPv6: ADDRCONF(NETDEV_CHANGE): enp1s0f0: link becomes ready
[   39.469386] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   39.471213] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   39.484455] 88x2bu: loading out-of-tree module taints kernel.
[   39.484926] 88x2bu: module verification failed: signature and/or required key missing - tainting kernel
[   39.488424] usbcore: registered new interface driver rtl88x2bu
```

RinCat, 2022-07-30:
There should be something like
```
[188178.605109] RTW: module init start
[188178.605112] RTW: rtl88x2bu v5.13.1-20-gbd7c7eb9d.20210702_COEX20210316-18317b7b
[188178.605113] RTW: rtl88x2bu BT-Coex version = COEX20210316-18317b7b
[188178.605181] usbcore: registered new interface driver rtl88x2bu
[188178.605182] RTW: module init ret=0
```

`sudo modprobe 88x2bu` did not given any error?

YW0330, 2022-07-30:
My command is like this:
`~/RTL88x2BU-Linux-Driver$ sudo modprobe 88x2bu`
but the system did not show any feedback.

RinCat, 2022-07-30:
And `lsmod|grep 88` shows `88x2bu` exists?

YW0330, 2022-07-30:
Yes
```
88x2bu               2441216  0
cfg80211              634880  1 88x2bu
kvm_intel             217088  0
snd_usb_audio         208896  1
videodev              188416  3 videobuf2_core,videobuf2_v4l2,uvcvideo
hid                   122880  2 usbhid,hid_generic
```

RinCat, 2022-07-30:
Can you try remove and plug in again your DWA-T185, and check what shows in `dmesg`

YW0330, 2022-07-30:
I remove DWA-T185, and insert again. This is information of `dmesg`
```
[   39.469386] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   39.471213] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   39.484455] 88x2bu: loading out-of-tree module taints kernel.
[   39.484926] 88x2bu: module verification failed: signature and/or required key missing - tainting kernel
[   39.488424] usbcore: registered new interface driver rtl88x2bu
[ 1472.947463] usb 2-10: USB disconnect, device number 5
[ 1486.715401] usbcore: deregistering interface driver rtl88x2bu
[ 1495.978121] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[ 1495.978258] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[ 1495.983762] usbcore: registered new interface driver rtl88x2bu
[ 1534.951444] usb 2-10: new high-speed USB device number 7 using xhci_hcd
[ 1535.100033] usb 2-10: New USB device found, idVendor=2001, idProduct=3322
[ 1535.100038] usb 2-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1535.100041] usb 2-10: Product: 802.11ac NIC
[ 1535.100044] usb 2-10: Manufacturer: Realtek
[ 1535.100047] usb 2-10: SerialNumber: 123456
```

RinCat, 2022-07-30:
Let me download 16.04 and see what happened. BTW 16.04 is end of life, you are recommended upgrade to new one.

RinCat, 2022-07-30:
I had no problem with it, can you try do a new clone and install again?

```
[  275.766257] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  275.767558] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[  275.768946] 88x2bu: loading out-of-tree module taints kernel.
[  275.769495] 88x2bu: module verification failed: signature and/or required key missing - tainting kernel
[  275.772752] RTW: module init start
[  275.772755] RTW: rtl88x2bu v5.13.1-20-gbd7c7eb9d.20210702_COEX20210316-18317b7b
[  275.772756] RTW: rtl88x2bu BT-Coex version = COEX20210316-18317b7b
[  275.772792] usbcore: registered new interface driver rtl88x2bu
[  275.772793] RTW: module init ret=0
```

YW0330, 2022-07-30:
Hello, I just test this progress on another computer, it was feasible. They both are Ubuntu 16.04, but original one kernel is `4.15.0-142-generic`, and the new one is `4.15.0-041500rc8-generic`. I'm not sure if it is reason that. I choose the Ubuntu 16.04, because my devices need run ROS kinetic version.

YW0330, 2022-07-30:
> I had no problem with it, can you try do a new clone and install again?
>
> ```
> [  275.766257] cfg80211: Loading compiled-in X.509 certificates for regulatory database
> [  275.767558] cfg80211: Loaded [...]

I  think I need reinstall OS on my original one :laughing: . I learn more things about linux, thank you very much. 謝謝

RinCat, 2022-07-30:
`4.15.0-142-generic` is newer than `4.15.0-041500rc8-generic`, but I had no idea if it is the kernel issue or not.

YW0330, 2022-07-30:
OK, I reinstall my computer first. Thank you very much.

closed 2022-07-30 by YW0330