RinCat/RTL88x2BU-Linux-Driver #147: Does this driver supports roaming features?

seanxu112, 2022-08-02:
Hello, I am pretty uneducated with networking topics. Based on my knowledge Linux (or Ubuntu in my case) handles roaming by switching access points once the current access point is out performed by another access point. I have a mobile platform, so this is pretty important.

The only way I know of to achieve this is to use iwconfig to set sens to a certain value, so that Ubuntu searches for other access point once the current one is lower than the set sens value. 

It seems like iwconfig cannot set sens (set failed with device ****, Operation not permitted. Is this problem related to this driver or the hardware.

Sincerely,
Sean

RinCat, 2022-08-02:
It does not support manual set sensitivity. However you network manager should able roaming as they wish.

seanxu112, 2022-08-02:
I see, thank you for the quick reply. It pointed me to looking into network manager, and I solved the issue with 
sudo wpa_cli -i my_interface set_network 0 bgscan '"simple:5:-60:10"'
to check the signal strength for access points

closed 2022-08-02 by seanxu112