RinCat/RTL88x2BU-Linux-Driver #149: DKMS not updating driver when kernel updated

do-dat-thang, 2022-08-17:
DKMS does not seem to update this driver when I run `sudo apt full-upgrade` which gets a kernel update.

Is this problem due to the single quotes around `make` in the one line of dkms.conf?

> PACKAGE_NAME="rtl88x2bu"
> PACKAGE_VERSION=git-RinCat
> MAKE="'make' -j$(nproc) KVER=${kernelver} KSRC=/lib/modules/${kernelver}/build"
> CLEAN="make clean"
> BUILT_MODULE_NAME[0]="88x2bu"
> DEST_MODULE_LOCATION[0]="/kernel/drivers/net/wireless"
> AUTOINSTALL="yes"

After updating my system when a new kernel is in the update, the /var/lib/dkms/rtl88x2bu directory no longer exists.

On the prior level, i ran `sudo dkms install rtl88x2bu/git-RinCat` and then the `sudo dkms status` shows similar to `rtl88x2bu, git-RinCat, 5.15.56-v7l+, armv7l: installed` (except with the prior kernel version which was 5.15.32-v7l+).  After the update it is gone and I run the same `dkms install` comand again and all is well again until the next time (which the whole purpose of the dkms is to avoid having to rebuild/install each time). I ran into this on the last kernel update also.

I realize this is probably not the right place for dkms issues, but I'm wondering if you have verified this and if it is something simple with this setup.

This is on a Raspberry Pi 4 running the latest released Raspberry OS (Debian 11 bullseye).

RinCat, 2022-08-17:
Hi, `/var/lib/dkms/rtl88x2bu` should exist after an update, so I don't know why it is missing.

RinCat, 2022-08-17:
The source itself still in `/usr/src/rtl88x2bu-git`?

do-dat-thang, 2022-08-17:
Yes it is still in `/usr/src/rtl88x2bu-git-RinCat` ... all I have to do after it goes away in /var/lib is `sudo dkms install rtl88x2bu/git-RinCat` and it builds and installs the driver.

I went back and in the system logs, I could see that dkms has a few entries that appear to be from the time frame of the `apt upgrade`, but they don't really indicate what dkms was up to or results.

RinCat, 2022-08-17:
I did some searching but found nothing like this. So it could be something broken in you system.
Is your `/var` in a separated mount point?

do-dat-thang, 2022-08-18:
No, entire filesystem on SD.

As I mentioned earlier,  this isn't really the place for dkms issues.  I see that you are quite knowledgeable with linux and just wanted to check if you had seen/heard of this problem. I will close this issue and debug from a dkms perspective. If it turns out to be something with this (very doubtful) I'll come back.

Thank you for taking the time to respond. What you are providing with this repo is greatly appreciated by many!

closed 2022-08-18 by do-dat-thang

do-dat-thang, 2022-08-18:
Quick comment in case this comes up again here....   The problem is with Debian:  [https://github.com/dell/dkms/issues/37](url)