RinCat/RTL88x2BU-Linux-Driver #151: "Realtek Semiconductor Co., Ltd. Device b852" driver problem

jixiedaima, 2022-08-21:
Installed this driver on ubuntu22.04, but the computer still can't recognize wifi.
Could you tell me how to install the driver?

RinCat, 2022-08-21:
Hi, please run `modprobe 88x2bu` and see if any error exists. Also check the `dmesg` result.

jixiedaima, 2022-08-21:
hinkbook@thinkbook:~$ sudo  modprobe 88x2bu
[sudo] thinkbook 的密码： 
thinkbook@thinkbook:~$ dmesg
dmesg: 读取内核缓冲区失败: 不允许的操作
thinkbook@thinkbook:~$ sudo dmesg
[    0.000000] Linux version 5.15.0-47-generic (buildd@lcy02-amd64-060) (gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #51-Ubuntu SMP Thu Aug 11 07:51:15 UTC 2022 (Ubuntu 5.15.0-47.51-generic 5.15.46)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-47-generic root=UUID=601b3349-9268-45fc-baa8-024f8be08273 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x200: 'Protection Keys User registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[9]:  832, xstate_sizes[9]:    8
[    0.000000] x86/fpu: Enabled xstate features 0x207, context size is 840 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 3376
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000bffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009afffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009b00000-0x0000000009dfffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000009e00000-0x0000000009efffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009f00000-0x0000000009f27fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000009f28000-0x000000005fdfefff] usable
[    0.000000] BIOS-e820: [mem 0x000000005fdff000-0x000000005fdfffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000005fe00000-0x0000000062363fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000062364000-0x0000000064563fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000064564000-0x000000006456cfff] usable
[    0.000000] BIOS-e820: [mem 0x000000006456d000-0x000000006456ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000064570000-0x000000007077efff] usable
[    0.000000] BIOS-e820: [mem 0x000000007077f000-0x0000000072f7efff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000072f7f000-0x000000007af7efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007af7f000-0x000000007affefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007afff000-0x000000007affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x000000007bffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007d800000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000a0200000-0x00000000a02fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed80fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000003fe2fffff] usable
[    0.000000] BIOS-e820: [mem 0x00000003ff340000-0x00000004a01fffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x5fc43018-0x5fc52e57] usable ==> usable
[    0.000000] e820: update [mem 0x5fc43018-0x5fc52e57] usable ==> usable
[    0.000000] e820: update [mem 0x5fc38018-0x5fc42e57] usable ==> usable
[    0.000000] e820: update [mem 0x5fc38018-0x5fc42e57] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009f000-0x00000000000bffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000009afffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000009b00000-0x0000000009dfffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000009e00000-0x0000000009efffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000009f00000-0x0000000009f27fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000009f28000-0x000000005fc38017] usable
[    0.000000] reserve setup_data: [mem 0x000000005fc38018-0x000000005fc42e57] usable
[    0.000000] reserve setup_data: [mem 0x000000005fc42e58-0x000000005fc43017] usable
[    0.000000] reserve setup_data: [mem 0x000000005fc43018-0x000000005fc52e57] usable
[    0.000000] reserve setup_data: [mem 0x000000005fc52e58-0x000000005fdfefff] usable
[    0.000000] reserve setup_data: [mem 0x000000005fdff000-0x000000005fdfffff] reserved
[    0.000000] reserve setup_data: [mem 0x000000005fe00000-0x0000000062363fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000062364000-0x0000000064563fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000064564000-0x000000006456cfff] usable
[    0.000000] reserve setup_data: [mem 0x000000006456d000-0x000000006456ffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000064570000-0x000000007077efff] usable
[    0.000000] reserve setup_data: [mem 0x000000007077f000-0x0000000072f7efff] reserved
[... 1408 lines cut ...]
[ 4448.439583] amdgpu 0000:05:00.0: amdgpu: ring comp_1.1.1 uses VM inv eng 8 on hub 0
[ 4448.439584] amdgpu 0000:05:00.0: amdgpu: ring comp_1.2.1 uses VM inv eng 9 on hub 0
[... cut ...]
[ 4448.594237] audit: type=1420 audit(1661059904.412:63): subj_apparmor=unconfined
[ 4448.732696] IPv6: ADDRCONF(NETDEV_CHANGE): enx00e04c8042bf: link becomes ready
[ 4448.732873] r8152 4-2.4:1.0 enx00e04c8042bf: carrier on
[ 5151.916195] RTW: module init start
[ 5151.916197] RTW: rtl88x2bu v5.13.1-20-gbd7c7eb9d.20210702_COEX20210316-18317b7b
[ 5151.916198] RTW: rtl88x2bu BT-Coex version = COEX20210316-18317b7b
[ 5151.916254] usbcore: registered new interface driver rtl88x2bu
[ 5151.916255] RTW: module init ret=0

RinCat, 2022-08-21:
Your module loaded successful. What USB WiFi hardware do you have? check `lsusb`.

jixiedaima, 2022-08-21:
thinkbook@thinkbook:~/Downloads/test/RTL88x2BU-Linux-Driver$ lspci | grep Network
02:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852
thinkbook@thinkbook:~$ lsusb
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 04f2:b758 Chicony Electronics Co., Ltd Integrated Camera
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 005: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 004 Device 004: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 0bda:4853 Realtek Semiconductor Corp. Bluetooth Radio
Bus 003 Device 005: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 003 Device 002: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 04f3:0c4b Elan Microelectronics Corp. ELAN:Fingerprint
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

RinCat, 2022-08-21:
None of them is a USB WiFi device, Are you sure the hardware is working?

jixiedaima, 2022-08-21:
My computer is Thinkbook 16P NX. At win11, it can connect wifi. But at ubuntu, it can't. So I think it maybe the problem of wifi driver.

RinCat, 2022-08-21:
Well, that is the wifi driver issue, but not this driver.  Your laptop has a PCI-E WiFi hardware and most likely the ID is 10ec:b852. 

Go check this one https://github.com/lwfinger/rtw8852be

Notice that it ends with E, which is for PCI-E. My driver ends with U, it is only for USB WiFi devices.

jixiedaima, 2022-08-21:
I see, thank you!

closed 2022-08-21 by RinCat