RinCat/RTL88x2BU-Linux-Driver #156: TP-Link Archer T3U Plus not working after the recent update in Ubuntu 22.04

DarshitSolanki04, 2022-09-02:
Previously it was working after using the commands mentioned here - [https://community.tp-link.com/en/home/forum/topic/547276](https://community.tp-link.com/en/home/forum/topic/547276). But after the recent ubuntu update it has stopped working completely.

RinCat, 2022-09-02:
Hi, what is your kernel version?

DarshitSolanki04, 2022-09-02:
Hello,
It is 5.15.0-47-generic.

DarshitSolanki04, 2022-09-02:
Even if I try to connect to another Wi-Fi source or forget the existing one, I am not getting any response to connect or forget, even after clicking multiple times. I can just turn on and off the Wi-Fi, nothing else.

RinCat, 2022-09-02:
Can you try follow this repo README and install again? You may used outdated version.

DarshitSolanki04, 2022-09-03:
I did and it still doesn't work. My "Leoxsys LEO-HG150N 150Mbps Wireless USB Wi-Fi Adapter LAN Card" works fine.
For me -
> $ sudo dkms status

shows:
> bcmwl/6.30.223.271+bdcom, 5.15.0-43-generic, x86_64: installed
bcmwl/6.30.223.271+bdcom, 5.15.0-47-generic, x86_64: installed
rtl88x2bu/git, 5.15.0-47-generic, x86_64: installed

I uninstalled rtl88x2ub and installed it again but still nothing.

DarshitSolanki04, 2022-09-03:
My */usr/src* contains these files:
> bcmwl-6.30.223.271+bdcom
linux-headers-5.15.0-43-generic
linux-headers-5.15.0-47-generic
linux-headers-5.15.0-43
linux-headers-5.15.0-47
rtl88x2bu-git

DarshitSolanki04, 2022-09-03:
**Update:** I found out **'TP-Link Archer T3U Plus'** can connect to my mobile hotspot but not connecting to my institute's Wi-Fi after the Ubuntu update same case for my laptop's in-built Wi-Fi card but **"Leoxsys LEO-HG150N 150Mbps Wireless USB Wi-Fi Adapter LAN Card"** can connect to my institute's Wi-Fi.

> Even if I try to connect to another Wi-Fi source or forget the existing one, I am not getting any response to connect or forget, even after clicking multiple times. I can just turn on and off the Wi-Fi, nothing else.

This is happening with my institue's Wi-Fi. Any idea or solution?? @RinCat

RinCat, 2022-09-03:
So your laptop WiFi card also cannot connect to your institute's WiFi?
It sounds like a bug somewhere else. 
Can you try change the WiFi control system to iwd? https://wiki.debian.org/NetworkManager/iwd

The Leoxsys LEO-HG150N 150Mbps can connect is because it is very old, only support 802.11n at 2.4GHz.

DarshitSolanki04, 2022-09-03:
After this

> $ iwctl

I am getting:

> Rejected send message, 2 matched rules; type="method_call", sender=":1.125" (uid=1000 pid=3126 comm="iwctl " label="unconfined") interface="org.freedesktop.DBus.ObjectManager" member="GetManagedObjects" error name="(unset)" requested_reply="0" destination="net.connman.iwd" (uid=0 pid=840 comm="/usr/libexec/iwd " label="unconfined")
Failed to retrieve IWD dbus objects, quitting...

Now, **"Leoxsys LEO-HG150N 150Mbps Wireless USB Wi-Fi Adapter LAN Card"** stopped working.

> Even if I try to connect to another Wi-Fi source or forget the existing one, I am not getting any response to connect or forget, even after clicking multiple times. I can just turn on and off the Wi-Fi, nothing else.

Same thing now with **Leoxsys**.

How to revert back??

RinCat, 2022-09-03:
Your iwd seems not started, did you reboot your system?

RinCat, 2022-09-03:
Also you will still using NetworkManager, not iwctl.

DarshitSolanki04, 2022-09-03:
Yes, I did reboot but still the same. Not able to use any Wi-Fi card.

RinCat, 2022-09-03:
Try restart NM `sudo systemctl restart NetworkManager`

DarshitSolanki04, 2022-09-03:
I already did it.

Same issue now:

> Even if I try to connect to another Wi-Fi source or forget the existing one, I am not getting any response to connect or forget, even after clicking multiple times. I can just turn on and off the Wi-Fi, nothing else.

RinCat, 2022-09-03:
Can you check the `dmesg` output?

DarshitSolanki04, 2022-09-03:
Strange. Now, I can connect to my mobile hotspot but not to institute's wifi with the **Leoxsys** which was happening previously with **TP-Link Archer T3U Plus**.

Sure, let me check.

DarshitSolanki04, 2022-09-03:
`dmesg`

Output:

> dmesg: read kernel buffer failed: Operation not permitted

RinCat, 2022-09-03:
`sudo dmesg`

DarshitSolanki04, 2022-09-03:
Output:

> [    0.000000] microcode: microcode updated early to revision 0x21, date = 2019-02-13
[    0.000000] Linux version 5.15.0-47-generic (buildd@lcy02-amd64-060) (gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #51-Ubuntu SMP Thu Aug 11 07:51:15 UTC 2022 (Ubuntu 5.15.0-47.51-generic 5.15.46)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-47-generic root=UUID=b45ba98c-e2a5-4c70-84c4-40b1cbaf1aa6 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] signal: max sigframe size: 1776
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009d3ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009d400-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x0000000040003fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040004000-0x0000000040004fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040005000-0x0000000090baffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000090bb0000-0x00000000913affff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000913b0000-0x000000009a3befff] usable
[    0.000000] BIOS-e820: [mem 0x000000009a3bf000-0x000000009aebefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000009aebf000-0x000000009afbefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000009afbf000-0x000000009affefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000009afff000-0x000000009affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000009b000000-0x000000009f9fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb00000-0x00000000feb03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffb80000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000025f5fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: Dell Inc. Inspiron 3521/06X37M, BIOS A16 05/24/2018
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 1795.765 MHz processor
[    0.001490] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001497] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001515] last_pfn = 0x25f600 max_arch_pfn = 0x400000000
[    0.001683] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.002499] last_pfn = 0x9b000 max_arch_pfn = 0x400000000
[    0.020966] found SMP MP-table at [mem 0x000fe1b0-0x000fe1bf]
[    0.022731] RAMDISK: [mem 0x30729000-0x3438bfff]
[    0.022742] ACPI: Early table checksum verification disabled
[    0.022747] ACPI: RSDP 0x00000000000FE020 000024 (v02 DELL  )
[    0.022755] ACPI: XSDT 0x000000009AFFE210 0000B4 (v01 DELL   CL09     00000001      01000013)
[    0.022765] ACPI: FACP 0x000000009AFFA000 00010C (v05 DELL   CL09     00000001 ASL  00040000)
[    0.022777] ACPI: DSDT 0x000000009AFED000 009332 (v01 DELL   CL09     00000000 ASL  00040000)
[    0.022784] ACPI: FACS 0x000000009AFB9000 000040
[    0.022789] ACPI: SLIC 0x000000009AFFD000 000176 (v01 DELL   CL09     00000001 ASL  00040000)
[    0.022796] ACPI: UEFI 0x000000009AFFC000 000236 (v01 DELL   CL09     00000001 ASL  00040000)
[    0.022802] ACPI: ASF! 0x000000009AFFB000 0000A5 (v32 DELL   CL09     00000001 ASL  00040000)
[    0.022808] ACPI: HPET 0x000000009AFF9000 000038 (v01 DELL   CL09     00000001 ASL  00040000)
[    0.022815] ACPI: APIC 0x000000009AFF8000 00008C (v03 DELL   CL09     00000001 ASL  00040000)
[    0.022822] ACPI: MCFG 0x000000009AFF7000 00003C (v01 DELL   CL09     00000001 ASL  00040000)
[    0.022828] ACPI: SLIC 0x000000009AFEC000 000176 (v01 DELL   CL09     00000001 ASL  00040000)
[    0.022835] ACPI: SSDT 0x000000009AFEB000 0006FE (v01 COMPAL CRV ORB  00001000 ACPI 00040000)
[    0.022841] ACPI: BOOT 0x000000009AFE9000 000028 (v01 DELL   CL09     00000001 ASL  00040000)
[    0.022847] ACPI: ASPT 0x000000009AFE7000 000034 (v07 DELL   CL09     00000001 ASL  00040000)
[    0.022854] ACPI: DBGP 0x000000009AFE6000 000034 (v01 DELL   CL09     00000001 ASL  00040000)
[... 1102 lines cut ...]
[ 1007.336051] usb 3-2: Product: 802.11 n WLAN
[ 1007.336056] usb 3-2: Manufacturer: Ralink
[ 1007.336060] usb 3-2: SerialNumber: 1.0
[ 1007.470123] usb 3-2: reset high-speed USB device number 8 using xhci_hcd
[... cut ...]
[ 1567.321062] wlan5: deauthenticating from 52:02:91:b3:79:15 by local choice (Reason: 3=DEAUTH_LEAVING)
[ 1618.894634] wlan5: authenticate with 02:94:66:8d:05:76
[ 1618.906574] wlan5: send auth to 02:94:66:8d:05:76 (try 1/3)
[ 1618.910053] wlan5: authenticated
[ 1618.915271] wlan5: associate with 02:94:66:8d:05:76 (try 1/3)
[ 1618.921309] wlan5: RX AssocResp from 02:94:66:8d:05:76 (capab=0x411 status=0 aid=1)
[ 1618.926130] wlan5: associated
[ 1619.197593] IPv6: ADDRCONF(NETDEV_CHANGE): wlan5: link becomes ready

RinCat, 2022-09-03:
The log shows you did not use T3U at all. Try change the configuration in NM?

DarshitSolanki04, 2022-09-03:
Rightnow I am using *'Leoxsys'* to connect with my mobile hotspot. I have just recently (a month) started using Ubuntu, so I don't know much about it. I would appreciate if you can just guide me to the relevant link. I have no idea how to do that.

DarshitSolanki04, 2022-09-03:
Or shall I go ahead with reinstalling Ubuntu? It is going to be a lot of work. 🥲

RinCat, 2022-09-03:
Try unplug Leoxsys and use T3U to connect the WiFi you cannot connected, and save `dmesg` output.

DarshitSolanki04, 2022-09-03:
Output:

[    0.000000] microcode: microcode updated early to revision 0x21, date = 2019-02-13
[    0.000000] Linux version 5.15.0-47-generic (buildd@lcy02-amd64-060) (gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #51-Ubuntu SMP Thu Aug 11 07:51:15 UTC 2022 (Ubuntu 5.15.0-47.51-generic 5.15.46)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-47-generic root=UUID=b45ba98c-e2a5-4c70-84c4-40b1cbaf1aa6 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] signal: max sigframe size: 1776
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009d3ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009d400-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x0000000040003fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040004000-0x0000000040004fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040005000-0x0000000090baffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000090bb0000-0x00000000913affff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000913b0000-0x000000009a3befff] usable
[    0.000000] BIOS-e820: [mem 0x000000009a3bf000-0x000000009aebefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000009aebf000-0x000000009afbefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000009afbf000-0x000000009affefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000009afff000-0x000000009affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000009b000000-0x000000009f9fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb00000-0x00000000feb03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffb80000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000025f5fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: Dell Inc. Inspiron 3521/06X37M, BIOS A16 05/24/2018
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 1795.837 MHz processor
[    0.001490] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001496] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001515] last_pfn = 0x25f600 max_arch_pfn = 0x400000000
[    0.001682] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.002497] last_pfn = 0x9b000 max_arch_pfn = 0x400000000
[    0.020962] found SMP MP-table at [mem 0x000fe1b0-0x000fe1bf]
[    0.021849] RAMDISK: [mem 0x30729000-0x3438bfff]
[    0.021859] ACPI: Early table checksum verification disabled
[    0.021863] ACPI: RSDP 0x00000000000FE020 000024 (v02 DELL  )
[    0.021871] ACPI: XSDT 0x000000009AFFE210 0000B4 (v01 DELL   CL09     00000001      01000013)
[    0.021882] ACPI: FACP 0x000000009AFFA000 00010C (v05 DELL   CL09     00000001 ASL  00040000)
[    0.021893] ACPI: DSDT 0x000000009AFED000 009332 (v01 DELL   CL09     00000000 ASL  00040000)
[    0.021900] ACPI: FACS 0x000000009AFB9000 000040
[    0.021906] ACPI: SLIC 0x000000009AFFD000 000176 (v01 DELL   CL09     00000001 ASL  00040000)
[    0.021912] ACPI: UEFI 0x000000009AFFC000 000236 (v01 DELL   CL09     00000001 ASL  00040000)
[    0.021918] ACPI: ASF! 0x000000009AFFB000 0000A5 (v32 DELL   CL09     00000001 ASL  00040000)
[    0.021925] ACPI: HPET 0x000000009AFF9000 000038 (v01 DELL   CL09     00000001 ASL  00040000)
[    0.021931] ACPI: APIC 0x000000009AFF8000 00008C (v03 DELL   CL09     00000001 ASL  00040000)
[    0.021937] ACPI: MCFG 0x000000009AFF7000 00003C (v01 DELL   CL09     00000001 ASL  00040000)
[    0.021943] ACPI: SLIC 0x000000009AFEC000 000176 (v01 DELL   CL09     00000001 ASL  00040000)
[    0.021949] ACPI: SSDT 0x000000009AFEB000 0006FE (v01 COMPAL CRV ORB  00001000 ACPI 00040000)
[    0.021956] ACPI: BOOT 0x000000009AFE9000 000028 (v01 DELL   CL09     00000001 ASL  00040000)
[    0.021962] ACPI: ASPT 0x000000009AFE7000 000034 (v07 DELL   CL09     00000001 ASL  00040000)
[    0.021968] ACPI: DBGP 0x000000009AFE6000 000034 (v01 DELL   CL09     00000001 ASL  00040000)
[... 1059 lines cut ...]
[   88.157012] radeon 0000:01:00.0: fence driver on ring 5 use gpu addr 0x0000000000075a18
[   88.157414] debugfs: File 'radeon_ring_gfx' in directory '1' already present!
[... cut ...]
[   88.631472] [drm] ib test on ring 5 succeeded
[   93.671561] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[  107.683120] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[  108.358593] usb 3-1: USB disconnect, device number 2
[  108.432327] wlx1cbfce14891f: deauthenticating from 88:b1:e1:af:de:60 by local choice (Reason: 3=DEAUTH_LEAVING)
[  126.642398] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)

RinCat, 2022-09-03:
[image]

Go to your wifi settings and change the MAC Address to T3U, in your system seems to be `wlxb4b0242bfae9`

DarshitSolanki04, 2022-09-03:
Initially, this connected things was not appearing. So, I was able to do nothing.

[image]

But then I went to **'Known Wi-Fi Networks'** and removed all of them, including my institute's, after that I was able to reconnect and fill the details and then T3U started working.

DarshitSolanki04, 2022-09-03:
But thanks 'Code Goddess' for all your effort. :+1:

RinCat, 2022-09-03:
It seems caused by you have multiple WiFi adapters and the configurations are conflict each other. Surely you can remove all of them and create new one. You better disable others if you not plan to use them.

DarshitSolanki04, 2022-09-03:
Actually, I was not using other adapters. I started to use them only when T3U stopped working after the Ubuntu update.

closed 2022-09-03 by DarshitSolanki04