RinCat/RTL88x2BU-Linux-Driver #157: After install is working but when reboot it not working

wincipher, 2022-09-22:
After I install via Manual DKMS installation

it working fine untill reboot my pc

`uname -r`
```
5.4.0-126-generic
```

`ls -la /usr/srcr`
```
drwxr-xr-x 25 root root 4096 ก.ย.  21 17:49 linux-headers-4.15.0-193
drwxr-xr-x  8 root root 4096 ก.ย.  21 17:49 linux-headers-4.15.0-193-generic
drwxr-xr-x 24 root root 4096 ก.ย.  21 19:12 linux-headers-5.4.0-126
drwxr-xr-x  7 root root 4096 ก.ย.  21 19:12 linux-headers-5.4.0-126-generic
drwxr-xr-x  8 root root 4096 ก.ย.  22 16:56 rtl88x2bu-git
drwxr-xr-x  2 root root 4096 ก.ค.  22  2017 .tmp_versions
```

`dkms status`
```
rtl88x2bu, git, 5.4.0-126-generic, x86_64: installed
```

after reboot I run `modprobe 88x2bu` then nothing response and `dmesg -w`
[
https://pastebin.com/xR1yJ3VM

RinCat, 2022-09-22:
What system you are using? It seems kernel issue that wireless module crashed.

```
[    3.077524] ------------[ cut here ]------------
[    3.077543] WARNING: CPU: 3 PID: 362 at net/wireless/core.c:867 wiphy_register+0x6d1/0x930 [cfg80211]
[    3.077543] Modules linked in: mei_hdcp snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_intel snd_intel_dspcfg snd_hda_codec snd_hda_core snd_hwdep usbhid hid snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel nouveau(+) kvm snd_seq crct10dif_pclmul ghash_clmulni_intel mxm_wmi snd_seq_device ttm aesni_intel snd_timer crypto_simd drm_kms_helper 8822bu(OE+) cryptd ums_realtek glue_helper i2c_algo_bit fb_sys_fops rapl uas syscopyarea snd sysfillrect intel_cstate usb_storage sysimgblt cfg80211 soundcore mei_me mei wmi_bmof mac_hid sch_fq_codel nf_nat_pptp nf_conntrack_pptp nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c ipmi_devintf ipmi_msghandler msr parport_pc ppdev lp parport drm ip_tables x_tables autofs4 crc32_pclmul i2c_i801 ahci libahci lpc_ich e1000e wmi video
[    3.077563] CPU: 3 PID: 362 Comm: systemd-udevd Tainted: G           OE     5.4.0-126-generic #142-Ubuntu
[    3.077564] Hardware name: LENOVO 3212CTO/MAHOBAY, BIOS 9SKT71AUS 07/02/2013
[    3.077572] RIP: 0010:wiphy_register+0x6d1/0x930 [cfg80211]
[    3.077573] Code: ff e9 78 fa ff ff 0f 0b b8 ea ff ff ff e9 6c fa ff ff 0f 0b b8 ea ff ff ff e9 60 fa ff ff 0f 0b b8 ea ff ff ff e9 54 fa ff ff <0f> 0b b8 ea ff ff ff e9 48 fa ff ff 0f 0b b8 ea ff ff ff e9 3c fa
[    3.077574] RSP: 0000:ffffa9ef4075f8e0 EFLAGS: 00010246
[    3.077574] RAX: ffffffffc082cfb0 RBX: ffff9da430eb7300 RCX: ffffffffc082cfa0
[    3.077575] RDX: ffffffffc082d040 RSI: 0000000000000001 RDI: 000000000000001e
[    3.077575] RBP: ffffa9ef4075f958 R08: 0000000000000004 R09: 0000000000000003
[    3.077576] R10: 0000000000000000 R11: 0000000000000001 R12: ffff9da430eb7300
[    3.077576] R13: ffffa9ef4026d000 R14: ffff9da43406c000 R15: ffffffffc08570e0
[    3.077577] FS:  00007f85a8a77880(0000) GS:ffff9da43db80000(0000) knlGS:0000000000000000
[    3.077578] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    3.077578] CR2: 000055ee3ad28030 CR3: 00000007f46ee003 CR4: 00000000001606e0
[    3.077579] Call Trace:
[    3.077583]  ? printk+0x58/0x6f
[    3.077622]  rtw_wiphy_register+0x26/0x51 [8822bu]
[    3.077648]  rtw_cfg80211_ndev_res_register+0x19/0x5b [8822bu]
[    3.077672]  rtw_os_ndev_register+0x22/0xb1 [8822bu]
[    3.077697]  rtw_os_ndevs_register+0xb7/0x106 [8822bu]
[    3.077721]  rtw_os_ndevs_init+0x29/0x3d [8822bu]
[    3.077744]  rtw_drv_init+0x4e4/0x606 [8822bu]
[    3.077747]  usb_probe_interface+0xed/0x290
[    3.077749]  really_probe+0x2b0/0x3d0
[    3.077751]  driver_probe_device+0xbc/0x100
[    3.077752]  device_driver_attach+0x5d/0x70
[    3.077753]  __driver_attach+0x8f/0x150
[    3.077754]  ? device_driver_attach+0x70/0x70
[    3.077755]  bus_for_each_dev+0x7e/0xc0
[    3.077756]  driver_attach+0x1e/0x20
[    3.077758]  bus_add_driver+0x161/0x200
[    3.077759]  driver_register+0x74/0xd0
[    3.077760]  usb_register_driver+0x89/0x130
[    3.077761]  ? 0xffffffffc08db000
[    3.077778]  rtw_drv_entry+0xc0/0x1000 [8822bu]
[    3.077780]  do_one_initcall+0x4a/0x200
[    3.077781]  ? kfree+0x231/0x250
[    3.077783]  ? _cond_resched+0x19/0x30
[    3.077784]  ? kmem_cache_alloc_trace+0x177/0x240
[    3.077786]  do_init_module+0x52/0x240
[    3.077787]  load_module+0x11e6/0x1320
[    3.077789]  __do_sys_finit_module+0xbe/0x120
[    3.077790]  ? __do_sys_finit_module+0xbe/0x120
[    3.077792]  __x64_sys_finit_module+0x1a/0x20
[    3.077793]  do_syscall_64+0x57/0x190
[    3.077794]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[    3.077795] RIP: 0033:0x7f85a92a073d
[    3.077796] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 23 37 0d 00 f7 d8 64 89 01 48
[    3.077797] RSP: 002b:00007fff92865ff8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[    3.077798] RAX: ffffffffffffffda RBX: 000055ee3ade98e0 RCX: 00007f85a92a073d
[    3.077798] RDX: 0000000000000000 RSI: 00007f85a9180ded RDI: 000000000000000f
[    3.077798] RBP: 0000000000020000 R08: 0000000000000000 R09: 0000000000000000
[    3.077799] R10: 000000000000000f R11: 0000000000000246 R12: 00007f85a9180ded
[    3.077799] R13: 0000000000000000 R14: 000055ee3ab8eab0 R15: 000055ee3ade98e0
[    3.077801] ---[ end trace 2413a783334f659f ]---
[    3.077802] RTW: rtw_cfg80211_ndev_res_register rtw_wiphy_register fail for if1
```

wincipher, 2022-09-23:
Hi,

Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.5 LTS
Release:	20.04
Codename:	focal

now I upgrade kernel 5.4.0-126-generic to 5.18.19-051819-generic and uninstall all WIFI driver

but wifi adapter still not working and can't install your driver again. what kernel version do you suggest ?

thank you

h3itoravila, 2022-09-23:
Hello how are you? I suggest you follow the following step-by-step:

Some commands may be redundant, but I strongly recommend their execution, as it is a "golden path".

Open a terminal (without root) in the folder where the GIT files are located and run:

sudo apt update
sudo apt install git linux-headers-generic dkms
sudo make uninstall
make clean
make
sudo make install
sudo modprobe 88x2bu
sudo reboot

If it's still not working, my suggestion is to roll back your kernel to a previous version.

Me: Fedora 36 Workstation with kernel 5.19.9-200.fc36.x86_64.

wincipher, 2022-09-27:
thank you for your advice, It already working fine.

bck01215, 2022-10-07:
This worked on arch as well. Thanks!

GrgMdmn, 2023-07-25:
> Hello how are you? I suggest you follow the following step-by-step:
>
> Some commands may be redundant, but I strongly recommend their execution, as it is a "golden path".
>
> Open a terminal (without [...]

Hi ! I have the same problem (unable to connect to wifi after rebooting). I tried your workaround but I am encountering an error when using sudo modprobe :
```
sudo modprobe 88x2bu
modprobe: FATAL: Module 88x2bu not found in directory /lib/modules/5.15.0-78-generic
```

Here is my configuration :

RinCat, 2023-07-25:
> Hi ! I have the same problem (unable to connect to wifi after rebooting). I tried your workaround but I am encountering an error when using sudo modprobe :
> 
> ```
> sudo modprobe 88x2bu
> modprobe: FATAL: Module 88x2bu not found in directory /lib/modules/5.15.0-78-generic
> ```
> 
> My distribution is Linux Mint 21.2 Cinnamon (5.8.4)

Your kernel changed, reinstall it.

GrgMdmn, 2023-07-25:
I have not changed my kernel, it is a fresh installation.

By the way, one thing I have noticed is if I ask the network assistant to forgot the wifi network, then, I can connect to it normally, typing the password, and that works again...

RinCat, 2023-07-25:
Then check your `dmesg` output, could be secure boot

GrgMdmn, 2023-07-25:
Can I leave you the result, or it could be compromising for me ?

RinCat, 2023-07-25:
It will have some private information, better remove them before post.

GrgMdmn, 2023-07-25:
I am not skilled enough to know what is compromising. By following the result, the only information given are about my  peripherals and my drivers... Maybe is it okay ? Or else, can you say me where to find in the result if the problem is about secure boot or not ?

RinCat, 2023-07-25:
You will see something like `modprobe: ERROR: could not insert '88x2bu': Key was rejected by service` if  secure boot block you to load it.

GrgMdmn, 2023-07-25:
Thanks ! I do not see any error linked to modprobe in the dsmeg log

RinCat, 2023-07-25:
Then please try reinstall it again. And make sure you current running kernel is the kernel you installed.

GrgMdmn, 2023-07-25:
```
user@user: ~$ uname -r
5.15.0-78-generic
user@user:~$ dpkg --list | grep linux-image
ii  linux-image-5.15.0-76-generic              5.15.0-76.83                               amd64        Signed kernel image generic
ii  linux-image-5.15.0-78-generic              5.15.0-78.85                               amd64        Signed kernel image generic
ii  linux-image-generic                        5.15.0.78.75                               amd64        Generic Linux kernel image
```

RinCat, 2023-07-25:
Then please install this driver again, do you get any error?

GrgMdmn, 2023-07-25:
Ok it seems to work, also after rebooting. I do not completely understand why did it not work, but now, it is good ! Thank you :)

GrgMdmn, 2023-07-26:
Hi again ! Unfortunately, it seems that I have the same issue. When restarting, my device is unable to reconnect automatically (the wifi connection loads, loads, loads and then stops). But if I ask to forgot the network and then reconnect to it by typing the password again, then it works... I cannot explain this :(

RinCat, 2023-07-26:
Try use `iwd` replace `wpa_supplicant`. https://wiki.debian.org/NetworkManager/iwd

GrgMdmn, 2023-07-26:
It does not work better. Moreover, I have lost my "_5GHz" signal, only keeping the _2.4GHz one

GrgMdmn, 2023-07-26:
I think that things are even worse with iwd : after the installation, the key is detected. I have managed to connect to my router, but the connection was very bad (worse than before)

Moreover, after having reboot, the key is not even detected. I am obliged to open a terminal and do some iwd in order to find the networks... and after having connected, the connection is bad. And here is the end of the loop, now we can restart the paragraph again and again ^^'

RinCat, 2023-07-26:
This reconnect loop may not be the driver problem. From some of the previous discussions it looks like it could be the control program trying to connect using some less compatible method. Is it possible for you to change the router channel?

GrgMdmn, 2023-07-26:
I can try but I have never had any problem with my channels when using the key on Windows

GrgMdmn, 2023-08-08:
Hi ! Sorry for the delay ! My router was unable to change channel (the ISP has change the firmware and it's now unable to change wifi channel whereas it should and it could before). So I have ordered an additionnal router that is wired to my ISP router, and everything is fine. I think the problem was not on the driver's side. Moreover, I think that the 5 GHz is kinda difficult to find. I remember now that on windows I had problems finding it. However, after having found it and connected to, I have not encountered any problem anymore. It seems to be kinda different of linux, but the real problem was my box I think.

Thanks a lot for your patience and your advice :)