RinCat/RTL88x2BU-Linux-Driver #159: modprobe: ERROR: could not insert '88x2bu': Operation not permitted

napogf, 2022-10-09:
hi guys, 
after compiled the driver (with some warnings) modprobe result is:
`sudo modprobe 88x2bu rtw_drv_log_level=5
modprobe: ERROR: could not insert '88x2bu': Operation not permitted
`

My laptop config:

`giacomo@~$ sudo lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy
giacomo@~$ uname -a
Linux giacomo-ThinkPad-S5-S540 5.15.0-48-generic #54-Ubuntu SMP Fri Aug 26 13:26:29 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
`
warnings during driver compilation:
`giacomo@~/wifidriver/RTL88x2BU-Linux-Driver(master)$ make > compiling.log 
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:299:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  299 |                 if (rfctl->regd_name)
      |                    ^
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:302:9: note: here
  302 |         default:
      |         ^~~~~~~
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:1670:38: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1670 |                         ptable->func = &OnAuthClient;
      |                         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:1672:9: note: here
 1672 |         case WIFI_ASSOCREQ:
      |         ^~~~
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_get_chbwoff_from_cfg80211_chan_def’:
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/os_dep/linux/ioctl_cfg80211.c:6811:21: warning: this statement may fall through [-Wimplicit-fallthrough=]
 6811 |                 *ht = 0;
      |                 ~~~~^~~
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/os_dep/linux/ioctl_cfg80211.c:6813:9: note: here
 6813 |         case NL80211_CHAN_WIDTH_20:
      |         ^~~~
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/hal/hal_intf.c: In function ‘c2h_handler’:
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/hal/hal_intf.c:1370:24: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1370 |                 sub_id = payload[0];
      |                 ~~~~~~~^~~~~~~~~~~~
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/hal/hal_intf.c:1374:9: note: here
 1374 |         default:
      |         ^~~~~~~
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/hal/rtl8822b/rtl8822b_phy.c: In function ‘_sounding_config_su’:
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/hal/rtl8822b/rtl8822b_phy.c:1303:28: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1303 |                         if (0 == bfee->su_reg_index)
      |                            ^
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/hal/rtl8822b/rtl8822b_phy.c:1308:17: note: here
 1308 |                 case CHANNEL_WIDTH_40:
      |                 ^~~~
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/hal/rtl8822b/rtl8822b_phy.c:1309:28: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1309 |                         if (0 == bfee->su_reg_index)
      |                            ^
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/hal/rtl8822b/rtl8822b_phy.c:1314:17: note: here
 1314 |                 case CHANNEL_WIDTH_20:
      |                 ^~~~
In file included from /home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/include/drv_types.h:60,
                 from /home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/hal/hal_halmac.c:17:
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/hal/hal_halmac.c: In function ‘_send_general_info’:
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/include/rtw_debug.h:159:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  159 |                 if (_DRV_WARNING_ <= rtw_drv_log_level) {\
      |                    ^
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/hal/hal_halmac.c:3109:17: note: in expansion of macro ‘RTW_WARN’
 3109 |                 RTW_WARN("%s: halmac_send_general_info() fail because fw not dl!\n",
      |                 ^~~~~~~~
/home/giacomo/wifidriver/RTL88x2BU-Linux-Driver/hal/hal_halmac.c:3112:9: note: here
[... 104 lines cut ...]
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x0006
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        1023 micro seconds
Device Status:     0x0000
  (Bus Powered)
`

RinCat, 2022-10-09:
Hi please check `dmesg` output, you may have to disable Secure Boot.

napogf, 2022-10-14:
Hi, you was right, I had secure boot enable, but after disabling the command sudo modprobe 88x2bu get this error in dmesg:
`[  301.217512] 88x2bu: disagrees about version of symbol module_layout
[  311.211132] 88x2bu: disagrees about version of symbol module_layout
`

napogf, 2022-10-15:
Hi RinCat, the problem is solved.
I install another time buildessential
`sudo apt-get install make gcc linux-headers-$(uname -r) build-essential`
and after recompiling driver modprobe load it 
Thanks

napogf, 2022-10-15:
SOLVED

closed 2022-10-15 by napogf