RinCat/RTL88x2BU-Linux-Driver #161: Sometimes it fails to connect

TheYoke, 2022-11-03:
Occasionally, when I disconnect the Wi-Fi and connect back again (or restart the OS) I have a problem where It won't connect and keeps giving me to input password even if I already did. And sometimes It works right away.

I checked the dmesg log, and it keeps reporting this when It fails to connect:

> [  359.480566] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[  360.281365] RTW: nolinked power save leave
[  366.695333] RTW: rtw_set_802_11_connect(wlx1cbfce4d328b)  fw_state=0x00000008
[  373.403496] RTW: WARN _beamforming_leave: 16:13:46:aa:99:7a is neither beamforming ee or er!!
[  377.169266] RTW: nolinked power save enter
[  383.691206] RTW: [HALMAC][ALWAYS]shall R reg twice!!

5.15.0-52-generic

RinCat, 2022-11-03:
Is this a public WiFi with multiple APs? It might be a compatibility issue.

TheYoke, 2022-11-04:
> Is this a public WiFi with multiple APs? It might be a compatibility issue.

Nope, It's just my home Wi-Fi, and It used to work fine before I upgraded my kernel from 5.4 to 5.15
Btw, when I complie the driver It has this warning. Don't know if it has anything to do with it.

> warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0

Right now, the only way to access the internet is to just keep connecting until it works. Which is really annoying :(

TheYoke, 2022-11-10:
Ok, I think I found a solution to my problem, that stared at me the whole time. I force my device to use USB 3.0 using `rtw_switch_usb_mode=1`. Not only it connects right away without failing, It also improves my down/up speed from by about 50%. What I forgot to mention was I bought a new computer, and It has USB 3.0, but the old one didn't. That's probably why it had a problem connecting. Hope this helps anyone with the same issue, Bye :)

RinCat, 2023-06-15:
Hi, someone has found a possible cause of the problem, if you are still using this, please update the driver to see if it is fixed. 
https://github.com/RinCat/RTL88x2BU-Linux-Driver/issues/198

labrat97, 2023-06-15:
This could be the wireless mode setting for the device also, depending on your model different wireless modes shouldn't be used. As an example, most people with an Archer T4U Plus experience some extra disconnects if they don't set `rtw_wireless_mode=84` in the module options for their system.

TheYoke, 2023-06-17:
> Hi, someone has found a possible cause of the problem, if you are still using this, please update the driver to see if it is fixed. #198

Updated it, now it fixes (my original problem) without any extra options. Thanks, @RinCat 

> This could be the wireless mode setting for the device also, depending on your model different wireless modes shouldn't be used. As an example, most people with an Archer T4U Plus experience some extra disconnects if they don't set `rtw_wireless_mode=84` in the module options for their system.

Indeed, I've experienced some disconnection occasionally. I'm going to see if this new version fixes. otherwise, will use the option.

closed 2023-06-17 by TheYoke