RinCat/RTL88x2BU-Linux-Driver #163: Blackscreen after reboot.

Chaile-Mendez, 2022-11-25:
After i made reboot, my POPos can't boot.
I get a gray screen for a few seconds (the normal popos gray of the login but only that, without the login) then when i press a key or after a while a lot of words run in the screen and then it stays black with a   _   in the superior left part of the screen

RinCat, 2022-11-27:
That's not this driver issue, external kernel module will not cause fail to boot. Since it is not loaded during boot.
Check your bootloader and GPU driver. Remove `quiet` and `splash` from your kernel parameters to see the boot process and logs.

Chaile-Mendez, 2022-12-02:
I tried it in a Clean new Pop os setup, i works perfectly but when I reboot next time I get :

cryptsetup waiting for encrypted source device 

And I can't boot from it. Not even in secure mode

RinCat, 2022-12-03:
cryptsetup is a disk encryption program, at that boot process the driver isn't even loadable. 
Please check related docs for LUKS full disk encryption.