RinCat/RTL88x2BU-Linux-Driver #164: Slower Speeds with CONCURRENT_MODE?

chrisvollorenew, 2022-11-27:
Hello, please forgive if this question is from ignorance:

I tested TX speeds of the driver on RTL8822BU in USB2.0 mode both with and without concurrent mode enabled (USER_EXTRA_CFLAGS='-DCONFIG_CONCURRENT_MODE).

The results were about 12MB/s without concurrent mode and 6MB/s with concurrent mode given my settings in HOSTAPD.

Is this normal, or should we throretically be able to achieve the same speeds with concurrent mode?

RinCat, 2022-11-27:
In theory you should able achieve same speed (that shared in 2 links), but it depends on many factors.

chrisvollorenew, 2022-11-30:
@RinCat what could these factors be? ... do you think it would help if I keep one of the virtual WAN DOWN while the other is in HOST mode?

RinCat, 2022-11-30:
I don't know how CONCURRENT_MODE implemented. They share antennas so they can't be used at the same time, I don't know if they are divided according to demand or completely divided by time. But if you keep one down it should same as just one IF?