RinCat/RTL88x2BU-Linux-Driver #165: Driver Crashing - Linux 5.17.0-rc8

chrisvollorenew, 2022-11-27:
Hello, 

I am using this driver on Linux 5.17.0-rc8 (arm architecture) with commit at the link below -
https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/5f6ef2ff4f15975fabf17f0fae1e651017f122e9

I am using Yocto to build and my make settings are as follows:
```
	ARCH=${ARCH} \
	KSRC=${STAGING_KERNEL_BUILDDIR} \
	CONFIG_RTL8822BU=m \
	USER_EXTRA_CFLAGS='-DCONFIG_CONCURRENT_MODE' \
```
ARCH=arm

Currently using a linux config at least base on sunxi_defconfig
I tried to use `CONFIG_PLATFORM_ARM_SUNxI` for this driver, but did not have good results. 
Currently using `CONFIG_PLATFORM_I386_PC` (default).

The driver works, but spews out this crash log when it starts.
Any suggestions?
```
[    9.478474] 88x2bu: loading out-of-tree module taints kernel.
[    9.667857] RTW: module init start
[    9.674403] RTW: rtl88x2bu v5.13.1-20-gbd7c7eb9d.20210702_COEX20210316-18317b7b
[    9.685349] RTW: rtl88x2bu BT-Coex version = COEX20210316-18317b7b
[    9.695236] RTW: [HALMAC]55772M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 6
               HALMAC_MINOR_VER = 6
               HALMAC_PATCH_VER = 23
[    9.732221] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    9.802039] systemd-journald[127]: Received client request to flush runtime journal.
[   10.364144] RTW: HW EFUSE
[   10.369983] RTW: 0x000: 29 81 00 3C  09 00 21 00  B6 04 64 04  00 00 A3 00  
[   10.380298] RTW: 0x010: 2A 2A 2A 2A  2A 2A 2A 2A  2A 2A 2A 02  00 00 FF FF  
[   10.390656] RTW: 0x020: FF FF 24 24  24 24 25 25  25 25 25 25  24 24 24 24  
[   10.400987] RTW: 0x030: 02 00 FF FF  00 FF 10 00  FF FF 2B 2B  2B 2B 2B 2B  
[   10.413692] RTW: 0x040: 2A 2A 2A 2A  2A 02 00 00  FF FF FF FF  28 28 28 28  
[   10.424050] RTW: 0x050: 28 28 28 28  28 28 28 28  28 28 02 00  FF FF 00 FF  
[   10.434462] RTW: 0x060: 10 00 FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.444777] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.455164] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.455246] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.455293] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.455339] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 23 1D 00  FF FF FF FF  
[   10.455385] RTW: 0x0C0: FF 21 00 10  00 00 00 00  00 FF 03 FF  FF FF FF FF  
[   10.455430] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.455476] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.455521] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.455567] RTW: 0x100: DA 0B 2C B8  E7 67 82 60  1D 9D DA 7E  AE 09 03 52  
[   10.455613] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
[   10.455659] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF  
[   10.455705] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.455750] RTW: 0x140: 33 0A 1F 01  00 00 21 0F  FF FF FF FF  FF FF FF FF  
[   10.455796] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.455842] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.455888] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.455933] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.455979] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.456025] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.456071] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.456116] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.456162] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.456208] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.456253] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.456298] RTW: 0x200: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.456343] RTW: 0x210: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.456389] RTW: 0x220: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.456435] RTW: 0x230: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.456481] RTW: 0x240: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[... 118 lines cut ...]
[   29.018808] WARNING: CPU: 3 PID: 1288 at net/core/dev_addr_lists.c:517 dev_addr_check+0xfc/0x12c
[   29.031915] netdevice: wlan0: Incorrect netdev->dev_addr
[   29.041503] Modules linked in: 88x2bu(O)
[   29.049743] CPU: 3 PID: 1288 Comm: hostapd Tainted: G        W  O      5.17.0-rc8 #1
[   29.061813] Hardware name: Allwinner sun8i Family
[   29.070762]  unwind_backtrace from show_stack+0x10/0x14
[   29.080189]  show_stack from dump_stack_lvl+0x40/0x4c
[   29.089386]  dump_stack_lvl from __warn+0xb0/0xd8
[   29.098137]  __warn from warn_slowpath_fmt+0x90/0xb8
[   29.107043]  warn_slowpath_fmt from dev_addr_check+0xfc/0x12c
[   29.116591]  dev_addr_check from __dev_open+0x34/0x190
[   29.125596]  __dev_open from __dev_change_flags+0x17c/0x1f4
[   29.134974]  __dev_change_flags from dev_change_flags+0x14/0x44
[   29.144653]  dev_change_flags from devinet_ioctl+0x78c/0x7bc
[   29.153988]  devinet_ioctl from inet_ioctl+0x18c/0x21c
[   29.162814]  inet_ioctl from sock_ioctl+0x47c/0x568
[   29.171388]  sock_ioctl from sys_ioctl+0x53c/0xbe4
[   29.179823]  sys_ioctl from ret_fast_syscall+0x0/0x54
[   29.188491] Exception stack(0xd09adfa8 to 0xd09adff0)
[   29.197108] dfa0:                   be8959ac 00000000 00000008 00008914 be8959ac 00001003
[   29.208926] dfc0: be8959ac 00000000 00556d20 00000036 00000008 00555450 00000000 00000000
[   29.220800] dfe0: 0054fbcc be89599c 004eb4b3 b6c3b408
[   29.229648] ---[ end trace 0000000000000000 ]---
[   29.284386] RTW: assoc success
```

RinCat, 2022-11-27:
You should try `arm` or `arm64`. The error ` netdevice: wlan0: Incorrect netdev->dev_addr` maybe endianness issue?

RinCat, 2022-11-30:
I tried `CONFIG_PLATFORM_ARM_SUNxI=y`, but that related code had not been updated over 5 years. Do not expect it will works.

I just noticed:
```
[   28.549256] rtl88x2bu 1-1:1.2 wlan1: Current addr:  62 1d 9d da 7e ae 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[   28.562856] rtl88x2bu 1-1:1.2 wlan1: Expected addr: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
So it was got a empty MAC address then changed, not sure how that happened.

RinCat, 2022-11-30:
`mach/sys_config.h` had been removed long times ago, I don't know what should to use now.

The MAC issue maybe a bug, I will check it later.

chrisvollorenew, 2022-11-30:
@RinCat thank you. 

For the SUNxI support, it would be nice. 
I did some research - It looks like we might be able to remove mach/sys_config.h and modify the code to replace. 
Basically the output from those functions in sys_config.h doesn't seem to be of critical importance.

chrisvollorenew, 2022-11-30:
I am going to create a new issue for SUNxI support