RinCat/RTL88x2BU-Linux-Driver #166: CONFIG_PLATFORM_SUNxI Support

chrisvollorenew, 2022-11-30:
Hello, this is a side issue from another issue:

https://github.com/RinCat/RTL88x2BU-Linux-Driver/issues/165

To re-cap:
CONFIG_PLATFORM_SUNxI was not compiling because of mach/sys_config.h missing.

I did some research - It looks like we might be able to remove mach/sys_config.h and modify the code to replace.
Basically the output from those functions in sys_config.h doesn't seem to be of critical importance.

Here is the mach/sys_config.h ... it only has a few function definitions that start with script_
https://github.com/allwinner-zh/linux-3.4-sunxi/blob/master/arch/arm/mach-sunxi/include/mach/sys_config.h

Here's the file in this repo that references those functions, it looks like those functuons are only used to detect errors, and may not be critical to the main operations of the wifi driver ...

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/657b7cfde9958e273febdeaeac579902e407f577/platform/platform_ARM_SUNxI_usb.c

I tried this patch and was able to compile, but still need to test:
```
diff --git a/platform/platform_ARM_SUNxI_usb.c b/platform/platform_ARM_SUNxI_usb.c
index 9c2abc4..4ef91cb 100644
--- a/platform/platform_ARM_SUNxI_usb.c
+++ b/platform/platform_ARM_SUNxI_usb.c
@@ -20,7 +20,7 @@
  */
 
 #include <drv_types.h>
-#include <mach/sys_config.h>
+//#include <mach/sys_config.h>
 
 #ifdef CONFIG_PLATFORM_ARM_SUNxI
 extern int sw_usb_disable_hcd(__u32 usbc_no);
@@ -51,12 +51,14 @@ int platform_wifi_power_on(void)
 #ifndef CONFIG_RTL8723A
        {
                /* ----------get usb_wifi_usbc_num------------- */
+               /*
                ret = script_parser_fetch("usb_wifi_para", "usb_wifi_usbc_num", (int *)&usb_wifi_host, 64);
                if (ret != 0) {
                        RTW_INFO("ERR: script_parser_fetch usb_wifi_usbc_num failed\n");
                        ret = -ENOMEM;
                        goto exit;
                }
+               */
                RTW_INFO("sw_usb_enable_hcd: usbc_num = %d\n", usb_wifi_host);
                sw_usb_enable_hcd(usb_wifi_host);
        }
@@ -65,6 +67,7 @@ int platform_wifi_power_on(void)
 
 #if defined(CONFIG_PLATFORM_ARM_SUN6I) || defined(CONFIG_PLATFORM_ARM_SUN7I)
        {
+               /*
                script_item_value_type_e type;
 
                type = script_get_item("wifi_para", "wifi_usbc_id", &item);
@@ -73,7 +76,7 @@ int platform_wifi_power_on(void)
                        ret = -ENOMEM;
                        goto exit;
                }
-
+               */
                printk("sw_usb_enable_hcd: usbc_num = %d\n", item.val);
                wifi_pm_power(1);
                mdelay(10);
```

RinCat, 2022-12-01:
I don't have SUNxI device, so you had to test yourself.

chrisvollorenew, 2022-12-01:
After thinking about the code more, I am wondering if `CONFIG_PLATFORM_SUNxI` makes any difference to the wifi opertation at all. There are only two functions `int platform_wifi_power_on(void)` and `int platform_wifi_power_off(void)`. I think the default behavior of these functions for other platforms is just return 0 and do nothing. So it seems that this is for platforms that need special instruction to turn on/off their wifi. This doesn't seem to be an issue with the SUNxI boards. Am I missing something? Does `CONFIG_PLATFORM_SUNxI` do anything else important?