RinCat/RTL88x2BU-Linux-Driver #168: I got an error in make rtw_proc.c

ohbake, 2022-12-15:
Hello, 

When I performed make on RHEL9, I get the following error.
Are there any workarounds?

[noguchi@kioxia-rhel:RTL88x2BU-Linux-Driver 2022-12-15 15:44:12]$ git pull
Already up to date.
[noguchi@kioxia-rhel:RTL88x2BU-Linux-Driver 2022-12-15 15:44:16]$ make
(...snip...)
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.0-162.6.1.el9_1.x86_64/build M=/home/noguchi/RTL88x2BU-Linux-Driver  modules
make[1]: ディレクトリ '/usr/src/kernels/5.14.0-162.6.1.el9_1.x86_64' に入ります
  CC [M]  /home/noguchi/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_proc.o
/home/noguchi/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_proc.c: 関数 ‘rtw_drv_proc_open’ 内:
/home/noguchi/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_proc.c:41:25: エラー: implicit declaration of function ‘PDE_DATA’; did you mean ‘NODE_DATA’? [-Werror=implicit-function-declaration]
   41 | #define pde_data(inode) PDE_DATA(inode)
      |                         ^~~~~~~~
/home/noguchi/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_proc.c:41:25: 備考: in definition of macro ‘pde_data’
   41 | #define pde_data(inode) PDE_DATA(inode)
      |                         ^~~~~~~~
cc1: いくつかの警告はエラーとして取り扱われます
make[2]: *** [scripts/Makefile.build:295: /home/noguchi/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_proc.o] エラー 1
make[1]: *** [Makefile:1915: /home/noguchi/RTL88x2BU-Linux-Driver] エラー 2
make[1]: ディレクトリ '/usr/src/kernels/5.14.0-162.6.1.el9_1.x86_64' から出ます
make: *** [Makefile:2455: modules] エラー 2

[noguchi@kioxia-rhel:RTL88x2BU-Linux-Driver 2022-12-15 15:44:23]$ uname -a
Linux kioxia-rhel 5.14.0-162.6.1.el9_1.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Sep 30 07:36:03 EDT 2022 x86_64 x86_64 x86_64 GNU/Linux

RinCat, 2022-12-15:
Hi, I do not have access to Red Hat, so I can only guess it backported some new kernel function to the old one.

Try:
https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/a2ac3b220db7ac38fb85d5f24730429ec2e68910/os_dep/linux/rtw_proc.c#L37-L42

Change the `KERNEL_VERSION(5, 17, 0)) ` to `KERNEL_VERSION(5, 14, 0)) `

ohbake, 2022-12-15:
Thanks for your advice.
It worked well!!

closed 2023-02-23 by RinCat