RinCat/RTL88x2BU-Linux-Driver #169: Make: Skipping BTF generation for 88x2bu.ko due to unavailability of vmlinux

bluehangel, 2022-12-18:
Hi,
got this error on make:
` CC [M]  /home/benoit/RTL88x2BU-Linux-Driver/hal/btc/halbtc8822b1ant.o
  CC [M]  /home/benoit/RTL88x2BU-Linux-Driver/hal/btc/halbtc8822b2ant.o
  CC [M]  /home/benoit/RTL88x2BU-Linux-Driver/platform/platform_ops.o
  CC [M]  /home/benoit/RTL88x2BU-Linux-Driver/core/rtw_mp.o
  LD [M]  /home/benoit/RTL88x2BU-Linux-Driver/88x2bu.o
  MODPOST /home/benoit/RTL88x2BU-Linux-Driver/Module.symvers
  CC [M]  /home/benoit/RTL88x2BU-Linux-Driver/88x2bu.mod.o
  LD [M]  /home/benoit/RTL88x2BU-Linux-Driver/88x2bu.ko
  BTF [M] /home/benoit/RTL88x2BU-Linux-Driver/88x2bu.ko
Skipping BTF generation for /home/benoit/RTL88x2BU-Linux-Driver/88x2bu.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-26-generic'`

`benoit@benoit-VirtualBox:~/RTL88x2BU-Linux-Driver$ uname -mr; lsb_release -d
5.19.0-26-generic x86_64
Description:	Ubuntu 22.10`

Is there any references of vmlinux in make or drivers ?

please also have a look of this topic: https://www.reddit.com/r/openSUSE/comments/pwxs3i/btf_file_not_generated_because_vmlinux_could_not/

thanks,

RinCat, 2022-12-18:
If I remember correctly, you can safe ignore that error. It only used for kernel debug.

bluehangel, 2022-12-20:
I ignore that error and install drivers. I can't connect to my wifi access point successfully.

'benoit@benoit-VirtualBox:~/RTL88x2BU-Linux-Driver$ sudo make install'
'sudo] password for benoit: '
'install -p -m 644 88x2bu.ko  /lib/modules/5.19.0-26-generic/kernel/drivers/net/wireless/'
'/sbin/depmod -a 5.19.0-26-generic'
'benoit@benoit-VirtualBox:~/RTL88x2BU-Linux-Driver$ sudo modprobe 88x2bu'
'benoit@benoit-VirtualBox:~/RTL88x2BU-Linux-Driver$ lsmod | grep cfg80211'
'cfg80211             1040384  1 88x2bu'
'benoit@benoit-VirtualBox:~/RTL88x2BU-Linux-Driver$ nmcli dev wifi'
'IN-USE  BSSID              SSID               MODE   CHAN  RATE        SIGNAL  BARS  SECURITY    '
'        F4:CA:E5:C0:7D:E5  FreeWifi_secure    Infra  6     195 Mbit/s  89      ▂▄▆█  WPA2 802.1X '
'        F4:CA:E5:C0:7D:E4  lemurblanc         Infra  6     195 Mbit/s  82      ▂▄▆█  WPA2        '
'        44:AD:B1:69:8C:44  Bbox-1F9A7455      Infra  100   540 Mbit/s  60      ▂▄▆_  WPA2        '
'        34:49:5B:70:27:80  Bbox-7C5DE6A5      Infra  1     130 Mbit/s  49      ▂▄__  WPA1 WPA2   '
'        44:AD:B1:69:8C:40  Bbox-1F9A7455      Infra  11    130 Mbit/s  40      ▂▄__  WPA2        '
'        34:49:5B:70:27:84  Bbox-7C5DE6A5      Infra  48    540 Mbit/s  40      ▂▄__  WPA1 WPA2   '
'        38:35:FB:F3:47:00  Bbox-SAILLER       Infra  1     130 Mbit/s  37      ▂▄__  WPA1 WPA2   '
'        38:35:FB:F3:47:04  Bbox-SAILLER-5GHz  Infra  44    540 Mbit/s  30      ▂___  WPA1 WPA2   '
'benoit@benoit-VirtualBox:~/RTL88x2BU-Linux-Driver$ sudo nmcli --ask dev wifi connect lemurblanc'
'Password: ••••••••••'
'Error: Connection activation failed: (53) The Wi-Fi network could not be found.'
'benoit@benoit-VirtualBox:~/RTL88x2BU-Linux-Driver$ nmcli dev wifi'
'IN-USE  BSSID              SSID               MODE   CHAN  RATE        SIGNAL  BARS  SECURITY    '
'        F4:CA:E5:C0:7D:E5  FreeWifi_secure    Infra  6     195 Mbit/s  89      ▂▄▆█  WPA2 802.1X '
'        F4:CA:E5:C0:7D:E4  lemurblanc         Infra  6     195 Mbit/s  85      ▂▄▆█  WPA2        '
'        34:49:5B:70:27:80  Bbox-7C5DE6A5      Infra  1     130 Mbit/s  50      ▂▄__  WPA1 WPA2   '
'        34:49:5B:70:27:84  Bbox-7C5DE6A5      Infra  48    540 Mbit/s  42      ▂▄__  WPA1 WPA2   '
'        44:AD:B1:69:8C:40  Bbox-1F9A7455      Infra  11    130 Mbit/s  40      ▂▄__  WPA2        '
'        38:35:FB:F3:47:00  Bbox-SAILLER       Infra  1     130 Mbit/s  32      ▂▄__  WPA1 WPA2   '
'        38:35:FB:F3:47:04  Bbox-SAILLER-5GHz  Infra  44    540 Mbit/s  29      ▂___  WPA1 WPA2   '
'        68:3F:7D:24:C0:A0  Livebox-C0A0       Infra  11    195 Mbit/s  15      ▂___  WPA2        '

I use the same setup with Manjaro without issue. Same drivers, installed with AUR (rtl88x2bu-dkms-git)

Thanks for your advices,

RinCat, 2022-12-21:
The secrets issue maybe not the secrets, but some configuration/encryption/channel incompatible, and failed handshake.
Can you try use iwd not wpa_supplicant?

bluehangel, 2022-12-21:
I have no issue with a Ralink RT3072 adapter. Can't success to use iwd, seem's working for Ralink bot not for Realtek, command line related to iwd control fail.

`...Rejected send message, 2 matched rules; type="method_call",...`

RinCat, 2022-12-21:
You seems try use `iwctl`? If you have networkmanager running, you need use it. https://wiki.debian.org/NetworkManager/iwd

bluehangel, 2022-12-29:
Success to use iwd and get a connexion but navigation speed is slow and speed connexion is "unknow" in the connexion information panel of networkmanager. When using Ralink adapter, network speed is well showed and navigation comfortable.

bluehangel, 2023-07-21:
Facing similar issue this past week with Manjaro. An update of OpenPGP fix this few days after.

closed 2023-07-21 by bluehangel