RinCat/RTL88x2BU-Linux-Driver #170: Module has to be re-installed when kernel updates

geajack, 2022-12-31:
I installed this module following the README (`make`/`sudo make install`), and then Ubuntu updated the kernel from `5.15.0-43` to `5.15.0-56` or something like that, and the module stopped working. No big deal, just had to reinstall it again following the same procedure, but is there anything that can be changed in the install script so this doesn't happen?

RinCat, 2022-12-31:
No. To get it auto update with kernel, use DKMS.

Strykar, 2023-07-26:
@RinCat DKMS appears to rebuild it on Arch LInux when a new kernel update is detected.
However, a wifi device does not show up until I manually rebuild the `rtl88x2bu-dkms-git` AUR package and then load the modele.

`lsusb` after it is working - https://gist.github.com/Strykar/3a5e23b493b3c2ac39b7f703f7169191