RinCat/RTL88x2BU-Linux-Driver #173: new device

wp-coin, 2023-01-16:
I have a new device 

usb-ax55 nano, chipset is realtek RTL8832BU

https://www.asus.com/networking-iot-servers/adapters/all-series/usb-ax55-nano/helpdesk_download/?model2Name=USB-AX55-Nano

RinCat, 2023-01-17:
Hi, I checked the RTL8832BU/RTL8852BU. They are WiFi 6 hardware, and their drivers do not support 8812 and 8822.

`831 files changed, 674007 insertions(+), 73843 deletions(-)`
And their code is significantly different, therefore probably nothing I can do about it.

There is a linux driver here you can try: https://www.zyxel.com/no/no/support/download?model=NWD7605
Or try the in-tree RTW88 driver.

wp-coin, 2023-01-17:
thanks for that, thats what I need for wifi 6

i'm getting some errors on Ubuntu 

```
note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:297: /home/mason/Downloads/Linux/RTL8852BU_RTL8832BU_WiFi_linux_v1.15.7-112-g1d36f2f7.20211230/driver/rtl8852BU_WiFi_linux_v1.15.7-112-g1d36f2f7.20211227/os_dep/linux/recv_linux.o] Error 1
make[1]: *** [Makefile:1902: /home/mason/Downloads/Linux/RTL8852BU_RTL8832BU_WiFi_linux_v1.15.7-112-g1d36f2f7.20211230/driver/rtl8852BU_WiFi_linux_v1.15.7-112-g1d36f2f7.20211227] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-58-generic'
make: *** [Makefile:633: modules] Error 2
##################################################
Compile make driver error: 2
Please check error Mesg

```

wp-coin, 2023-01-18:
never mind I fixed the code for kernel 5.15+

it installed in my wireless directory 

```
/lib/modules/5.15.0-58-generic/kernel/drivers/net/wireless/

```
my usb mini is still not recognized, no wifi icons show up in menu

should it be moved the file to "realtek/rtl8xxxu" directory?

RinCat, 2023-01-18:
You may need to add the USB ID to the driver, check `lsusb` and `os_dep/linux/usb_intf.c` file. And change to something like [this](https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/cd2a6b3f651ed3dee741618cec07dc1ce58e4eaa)

RinCat, 2023-01-18:
Apple this patch and rebuild:
```
diff --git a/os_dep/linux/usb_intf.c b/os_dep/linux/usb_intf.c
index 566ceda..63719b7 100755
--- a/os_dep/linux/usb_intf.c
+++ b/os_dep/linux/usb_intf.c
@@ -156,6 +156,7 @@ static struct usb_device_id rtw_usb_id_tbl[] = {
 	{USB_DEVICE_AND_INTERFACE_INFO(USB_VENDER_ID_REALTEK, 0xB85A, 0xff, 0xff, 0xff), .driver_info = RTL8852B},
 	{USB_DEVICE_AND_INTERFACE_INFO(USB_VENDER_ID_REALTEK, 0xA85B, 0xff, 0xff, 0xff), .driver_info = RTL8852B},
 	{USB_DEVICE_AND_INTERFACE_INFO(0x0586, 0x3428, 0xff, 0xff, 0xff), .driver_info = RTL8852B},
+	{USB_DEVICE_AND_INTERFACE_INFO(0x0b05, 0x1a62, 0xff, 0xff, 0xff), .driver_info = RTL8852B},
 #endif /* CONFIG_RTL8852B */
 
 	{}	/* Terminating entry */
```

eblu3, 2023-05-25:
Hi! I stumbled across this issue trying to get the same device to work under Linux and as much as I tried to follow the steps in this issue, I couldn't get anything to compile; when I applied the patch to the driver in the download link above I got an error related to missing headers. As much as I'd like to fix this myself, I'm nowhere near that experienced in C development to try and fix it. :P

Is there a more straightforward solution available? I'm running Arch if that helps.

RinCat, 2023-05-25:
> Hi! I stumbled across this issue trying to get the same device to work under Linux and as much as I tried to follow the steps in this issue, I couldn't get anything to compile; when I applied the [...]

Did you installed `linux-headers`?

eblu3, 2023-05-25:
Linux headers are installed. The issue seems to be within the driver itself; the build fails with an implicit declaration function on `complete_and_exit(comp, 0);` in line 491 of include/osdep_service_linux.h.

RinCat, 2023-05-26:
> Linux headers are installed. The issue seems to be within the driver itself; the build fails with an implicit declaration error on `complete_and_exit(comp, 0);` in line 491 of include/osdep_service_linux.h.

Which kernel? LTS?

eblu3, 2023-05-26:
> > Linux headers are installed. The issue seems to be within the driver itself; the build fails with an implicit declaration error on `complete_and_exit(comp, 0);` in line 491 of include/osdep_service_linux.h.
> 
> Which kernel? LTS?

It should be the latest, I'm running Arch

RinCat, 2023-05-26:
> > > Linux headers are installed. The issue seems to be within the driver itself; the build fails with an implicit declaration error on `complete_and_exit(comp, 0);` in line 491 of include/osdep_service_linux.h.
> > 
> > 
> > Which kernel? LTS?
> 
> It should be the latest, I'm running Arch

I don't have issue with arch `linux-6.3.4-arch1-1`. 
How did you installed this driver?

eblu3, 2023-05-26:
I tried building it with both the install.sh script in the driver's root directory and the makefile; both returned the same error. It was never installed because of this.

RinCat, 2023-05-26:
> I tried building it with both the install.sh script in the driver's root directory and the makefile; both returned the same error. It was never installed because of this.

There are no `install.sh` in this driver. Did you mixed with other driver or something.

If you are building for the current running kernel, you only need to run `make` (and `make clean` to remove old compiled files)

eblu3, 2023-05-26:
Okay, it looks like we're not on the same page here. The patch that added the AX55 was for the driver that you linked earlier in the thread; I tried applying that and got multiple errors. I'm focused on building *that* driver because it's the only one with an actual proposed solution to the problem of the hardware not being recognized by any driver that I'm aware of but I understand if you can't help; I'm just trying to figure out what wp-coin meant by "fixing the code".

RinCat, 2023-05-26:
Oh, the one that fixed that is in this commit https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/ac5758c404dc0585a0454182b137990c50f815c1.

However you need to patch multiple things in order to use it on 6.3 kernel.
This driver commit history should able to help you to fix most of the issues.

kzajac83, 2023-09-12:
Dear All.

I got USB 3.0 device with chip RTL8832BU Wifi6 but after download drivers from Zyxel got the same error as @wp-coin however he fix somehow but not tell how.

How to fix the below issue:

`make[1]: Opuszczenie katalogu '/usr/src/linux-headers-6.1.0-11-amd64'
make: *** [Makefile:729: modules] Błąd 2
##################################################
Compile make driver error: 2
Please check error Mesg
##################################################
` 

PS. There are also suggest use rtw88 but not sure what to use:

`rtw88_8723d   rtw88_8723de  rtw88_8821c   rtw88_8821ce  rtw88_8822b   rtw88_8822be  rtw88_8822c   rtw88_8822ce  rtw88_pci`

lsusb give this output:

`Bus 001 Device 007: ID 0bda:b832 Realtek Semiconductor Corp. 802.11ac WLAN Adapter
`