RinCat/RTL88x2BU-Linux-Driver #174: Failed to install in kernel 6.1.7-1-default

cudla100, 2023-01-22:
Driver failed to install in 6.1.7-1-default in openSUSE.
Dkms output:
```
cudla100@cudla100-hp-pc:~> sudo dkms install rtl88x2bu/git

Building module:
cleaning build area...
'make' -j16 KVER=6.1.7-1-default KSRC=/lib/modules/6.1.7-1-default/build........
cleaning build area...

88x2bu.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.7-1-default/updates/
Adding any weak-modules
getopt: unrecognized option '--no-initramfs'
Usage: ${0##*/} [options] {--add-modules|--remove-modules}
       ${0##*/} [options] {--add-kernel|--remove-kernel} {kernel-release}

--add-modules
        Add a list of modules read from standard input. Create
        symlinks in compatible kernel's weak-updates/ directory.
        The list of modules is read from standard input.

--remove-modules
        Remove compatibility symlinks from weak-updates/ directories
        for a list of modules.  The list of modules is read from
        standard input.

--add-kernel
        Add compatibility symlinks for all compatible modules to the
        specified or running kernel.

--remove-kernel
        Remove all compatibility symlinks for the specified or current
        kernel.

--verbose
        Print the commands executed.

-dry-run
        Do not create/remove any files.
depmod...
```
sudo inxi -Fazy:
```
System:
  Kernel: 6.1.6-1-default arch: x86_64 bits: 64 compiler: gcc v: 12.2.1
    parameters: BOOT_IMAGE=/boot/vmlinuz-6.1.6-1-default
    root=UUID=08ef971c-169f-4f6f-908b-7183bd1a26ee splash=silent
    resume=/dev/disk/by-uuid/5fd2b7e6-e363-4fb9-921b-adfe19b622b5
    mitigations=auto quiet security=apparmor nosimplefb=1
  Console: pty pts/1 wm: kwin_x11 DM: SDDM Distro: openSUSE Tumbleweed
    20230118
Machine:
  Type: Desktop System: HP product: HP Pavilion Gaming Desktop TG01-0xxx v: N/A
    serial: <filter> Chassis: type: 3 serial: <filter>
  Mobo: HP model: 8643 v: SMVB serial: <filter> UEFI: AMI v: F.23
    date: 07/07/2020
CPU:
  Info: model: AMD Ryzen 7 3700X socket: AM4 bits: 64 type: MT MCP arch: Zen 2
    gen: 3 level: v3 note: check built: 2020-22 process: TSMC n7 (7nm)
    family: 0x17 (23) model-id: 0x71 (113) stepping: 0 microcode: 0x8701021
  Topology: cpus: 1x cores: 8 tpc: 2 threads: 16 smt: enabled cache:
    L1: 512 KiB desc: d-8x32 KiB; i-8x32 KiB L2: 4 MiB desc: 8x512 KiB L3: 32 MiB
    desc: 2x16 MiB
  Speed (MHz): avg: 2537 high: 3600 min/max: 2200/4426 boost: enabled
    base/boost: 3600/4400 scaling: driver: acpi-cpufreq governor: schedutil
    volts: 1.1 V ext-clock: 100 MHz cores: 1: 2200 2: 2200 3: 2200 4: 2200
    5: 2200 6: 2200 7: 2200 8: 2200 9: 3600 10: 2800 11: 2800 12: 3600 13: 2200
    14: 3600 15: 2200 16: 2200 bogomips: 114986
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
  Vulnerabilities:
  Type: itlb_multihit status: Not affected
  Type: l1tf status: Not affected
  Type: mds status: Not affected
  Type: meltdown status: Not affected
  Type: mmio_stale_data status: Not affected
  Type: retbleed mitigation: untrained return thunk; SMT enabled with STIBP
    protection
  Type: spec_store_bypass mitigation: Speculative Store Bypass disabled via
    prctl
  Type: spectre_v1 mitigation: usercopy/swapgs barriers and __user pointer
    sanitization
  Type: spectre_v2 mitigation: Retpolines, IBPB: conditional, STIBP:
    always-on, RSB filling, PBRSB-eIBRS: Not affected
  Type: srbds status: Not affected
  Type: tsx_async_abort status: Not affected
Graphics:
  Device-1: NVIDIA TU106 [GeForce RTX 2060 Rev. A] vendor: Hewlett-Packard
    driver: nvidia v: 525.78.01 alternate: nouveau,nvidia_drm non-free: 520.xx+
    status: current (as of 2022-10) arch: Turing code: TUxxx process: TSMC 12nm
    built: 2018-22 pcie: gen: 1 speed: 2.5 GT/s lanes: 16 link-max: gen: 3
    speed: 8 GT/s ports: active: none off: HDMI-A-1 empty: DP-1,DVI-D-1
    bus-ID: 07:00.0 chip-ID: 10de:1f08 class-ID: 0300
  Display: x11 server: X.Org v: 21.1.6 with: Xwayland v: 22.1.7
    compositor: kwin_x11 driver: X: loaded: nvidia gpu: nvidia,nvidia-nvswitch
    display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1920x1080 s-dpi: 92 s-size: 530x301mm (20.87x11.85")
    s-diag: 610mm (24")
  Monitor-1: HDMI-A-1 mapped: HDMI-0 note: disabled model: HP X24ih
    serial: <filter> built: 2021 res: 1920x1080 dpi: 93 gamma: 1.2
    size: 527x297mm (20.75x11.69") diag: 605mm (23.8") ratio: 16:9 modes:
    max: 1920x1080 min: 640x480
  API: OpenGL v: 4.6.0 NVIDIA 525.78.01 renderer: NVIDIA GeForce RTX
    2060/PCIe/SSE2 direct render: Yes
Audio:
  Device-1: NVIDIA TU106 High Definition Audio vendor: Hewlett-Packard
    driver: snd_hda_intel v: kernel bus-ID: 1-3:2 pcie: chip-ID: 0d8c:013c gen: 3
    class-ID: 0300 speed: 8 GT/s lanes: 16 bus-ID: 07:00.1 chip-ID: 10de:10f9
    class-ID: 0403
  Device-2: AMD Starship/Matisse HD Audio vendor: Hewlett-Packard
    driver: snd_hda_intel v: kernel pcie: gen: 4 speed: 16 GT/s lanes: 16
    bus-ID: 09:00.4 chip-ID: 1022:1487 class-ID: 0403
  Device-3: C-Media CM108 Audio Controller type: USB
    driver: hid-generic,snd-usb-audio,usbhid
  Sound API: ALSA v: k6.1.6-1-default running: yes
  Sound Server-1: PulseAudio v: 16.1 running: no
  Sound Server-2: PipeWire v: 0.3.64 running: yes
Network:
  Device-1: Realtek RTL8821CE 802.11ac PCIe Wireless Network Adapter
    vendor: Hewlett-Packard driver: N/A modules: rtw88_8821ce pcie: gen: 1
    speed: 2.5 GT/s lanes: 1 port: e000 bus-ID: 05:00.0 chip-ID: 10ec:c821
    class-ID: 0280
  Device-2: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet
    vendor: Hewlett-Packard driver: r8169 v: kernel pcie: gen: 1 speed: 2.5 GT/s
    lanes: 1 port: d000 bus-ID: 06:00.0 chip-ID: 10ec:8168 class-ID: 0200
  IF: enp6s0 state: down mac: <filter>
  Device-3: TP-Link Archer T4U ver.3 type: USB driver: rtl88x2bu
    bus-ID: 1-11:3 chip-ID: 2357:0115 class-ID: 0000 serial: <filter>
  IF: wlp2s0f0u11 state: up mac: <filter>
Bluetooth:
  Device-1: Realtek Bluetooth 4.2 Adapter type: USB driver: btusb v: 0.8
    bus-ID: 1-12:4 chip-ID: 0bda:b00a class-ID: e001 serial: <filter>
  Report: rfkill ID: hci0 rfk-id: 1 state: down bt-service: enabled,running
    rfk-block: hardware: no software: yes address: see --recommends
Drives:
  Local Storage: total: 953.87 GiB used: 287.33 GiB (30.1%)
  ID-1: /dev/nvme0n1 maj-min: 259:0 vendor: Intel model: SSDPEKNW010T8H
    size: 953.87 GiB block-size: physical: 512 B logical: 512 B speed: 31.6 Gb/s
    lanes: 4 type: SSD serial: <filter> rev: HPS0 temp: 34.9 C scheme: GPT
  SMART: yes health: PASSED on: 106d 8h cycles: 497
    read-units: 17,978,136 [9.20 TB] written-units: 15,419,730 [7.89 TB]
Partition:
  ID-1: / raw-size: 936.41 GiB size: 936.41 GiB (100.00%)
    used: 287.32 GiB (30.7%) fs: btrfs block-size: 4096 B dev: /dev/nvme0n1p2
    maj-min: 259:2
  ID-2: /boot/efi raw-size: 300 MiB size: 299.4 MiB (99.80%)
    used: 3.5 MiB (1.2%) fs: vfat block-size: 512 B dev: /dev/nvme0n1p1
    maj-min: 259:1
  ID-3: /home raw-size: 936.41 GiB size: 936.41 GiB (100.00%)
    used: 287.32 GiB (30.7%) fs: btrfs block-size: 4096 B dev: /dev/nvme0n1p2
    maj-min: 259:2
  ID-4: /opt raw-size: 936.41 GiB size: 936.41 GiB (100.00%)
    used: 287.32 GiB (30.7%) fs: btrfs block-size: 4096 B dev: /dev/nvme0n1p2
    maj-min: 259:2
  ID-5: /var raw-size: 936.41 GiB size: 936.41 GiB (100.00%)
    used: 287.32 GiB (30.7%) fs: btrfs block-size: 4096 B dev: /dev/nvme0n1p2
    maj-min: 259:2
Swap:
  Kernel: swappiness: 60 (default) cache-pressure: 100 (default)
  ID-1: swap-1 type: partition size: 17.16 GiB used: 0 KiB (0.0%)
    priority: -2 dev: /dev/nvme0n1p3 maj-min: 259:3
Sensors:
  System Temperatures: cpu: 48.6 C mobo: N/A gpu: nvidia temp: 30 C
  Fan Speeds (RPM): N/A gpu: nvidia fan: 45%
Info:
  Processes: 359 Uptime: 0h 32m wakeups: 0 Memory: 15.53 GiB
  used: 2.88 GiB (18.6%) Init: systemd v: 252 default: graphical
  tool: systemctl Compilers: gcc: 12.2.1 alt: 12 Packages: pm: rpm pkgs: N/A
  note: see --rpm tools: yast,zypper pm: flatpak pkgs: 5 Shell: Bash (sudo)
  v: 5.2.15 running-in: konsole inxi: 3.3.23
```
I forgot to boot with latest kernel. I hope there is everything.
I am using this driver on TP-LINK Archer T4U AC1300 V3.2.  Another problem I usually have is that this driver takes a while to connect to the wifi. It is true that I am further than I should be, but even so, this driver fails to connect for the first couple tries.

RinCat, 2023-01-26:
Hi, I do not see any error, except `getopt: unrecognized option '--no-initramfs'`, which seems a bug in your system.
Can you try to manually install and see if it works?

cudla100, 2023-01-27:
I loaded the snapshot with broken driver, removed the driver with DKMS and rebooted. Then I made manual install and later tried with DKMS again. The driver seems to work again. I have no clue why it was not working in the first place. It did spit out  some warnings during manual install and I am going to put them here, but otherwise I have no large issues except the driver taking its time to connect. It is also working on 6.1.8-1-default openSUSE tumbleweed.

[makeclean.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/10522290/makeclean.txt)
[make.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/10522297/make.txt)
[sudomakeinstall.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/10522307/sudomakeinstall.txt)

closed 2023-05-06 by cudla100