RinCat/RTL88x2BU-Linux-Driver #175: wifi keeps disconnecting in periods

BendabizAdam, 2023-02-03:
Hi, i'm using a TP-Link Archer T3U Plus on ubuntu 22.04 LTS
before i used your driver the os didn't detect the card at all, but when i installed it it started detecting, after i connect to wifi, it starts disconnecting each 2mn, hope you can help and thank you

RinCat, 2023-02-04:
Check this https://wiki.archlinux.org/title/NetworkManager#Regular_network_disconnects,_latency_and_lost_packets_(WiFi)

BendabizAdam, 2023-02-04:
@RinCat 
well locking BSSID didn't help here's some logs

b 04 23:19:26 benjamina-pc NetworkManager[714]: <info>  [1675549166.3523] device (p2p-dev-wlp0s20f3): supplicant management interface state: scanning -> disconnected
Feb 04 23:19:35 benjamina-pc systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
Feb 04 23:20:13 benjamina-pc kernel: Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
Feb 04 23:20:28 benjamina-pc audit[6321]: SECCOMP auid=1000 uid=1000 gid=1000 ses=3 subj=snap.brave.brave pid=6321 comm="brave" exe="/snap/brave/197/opt/brave.com/brave/brave" sig=0 arch=c000003e syscall=330 compat=0 ip=0x7fae4af354e7 code=0x50000
Feb 04 23:20:28 benjamina-pc kernel: audit: type=1326 audit(1675549228.731:137): auid=1000 uid=1000 gid=1000 ses=3 subj=snap.brave.brave pid=6321 comm="brave" exe="/snap/brave/197/opt/brave.com/brave/brave" sig=0 arch=c000003e syscall=330 compat=0 ip=0x7fae4af354e7 code=0x50000
Feb 04 23:20:30 benjamina-pc audit[6360]: SECCOMP auid=1000 uid=1000 gid=1000 ses=3 subj=snap.brave.brave pid=6360 comm="brave" exe="/snap/brave/197/opt/brave.com/brave/brave" sig=0 arch=c000003e syscall=330 compat=0 ip=0x7fae4af354e7 code=0x50000
Feb 04 23:20:30 benjamina-pc kernel: audit: type=1326 audit(1675549230.335:138): auid=1000 uid=1000 gid=1000 ses=3 subj=snap.brave.brave pid=6360 comm="brave" exe="/snap/brave/197/opt/brave.com/brave/brave" sig=0 arch=c000003e syscall=330 compat=0 ip=0x7fae4af354e7 code=0x50000
Feb 04 23:20:49 benjamina-pc kernel: Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
Feb 04 23:20:55 benjamina-pc kernel: RTW: OnDisassoc(wlx3460f954564b) reason=34, ta=76:a7:41:6d:ee:98
Feb 04 23:20:55 benjamina-pc wpa_supplicant[747]: wlx3460f954564b: CTRL-EVENT-DISCONNECTED bssid=76:a7:41:6d:ee:98 reason=34
Feb 04 23:20:55 benjamina-pc wpa_supplicant[747]: BSSID 76:a7:41:6d:ee:98 ignore list count incremented to 2, ignoring for 10 seconds
Feb 04 23:20:55 benjamina-pc wpa_supplicant[747]: wlp0s20f3: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Feb 04 23:20:55 benjamina-pc kernel: RTW: WARN _beamforming_leave: 76:a7:41:6d:ee:98 is neither beamforming ee or er!!
Feb 04 23:20:55 benjamina-pc NetworkManager[714]: <info>  [1675549255.3569] device (wlx3460f954564b): supplicant interface state: completed -> disconnected
Feb 04 23:20:55 benjamina-pc NetworkManager[714]: <info>  [1675549255.3569] device (p2p-dev-wlx3460f954564b): supplicant management interface state: completed -> disconnected
Feb 04 23:20:55 benjamina-pc gnome-shell[2065]: An active wireless connection, in infrastructure mode, involves no access point?
Feb 04 23:20:55 benjamina-pc NetworkManager[714]: <info>  [1675549255.4576] device (wlx3460f954564b): supplicant interface state: disconnected -> scanning
Feb 04 23:20:55 benjamina-pc NetworkManager[714]: <info>  [1675549255.4577] device (p2p-dev-wlx3460f954564b): supplicant management interface state: disconnected -> scanning
Feb 04 23:20:56 benjamina-pc kernel: RTW: nolinked power save enter
Feb 04 23:20:59 benjamina-pc kernel: RTW: [HALMAC][ALWAYS]shall R reg twice!!
Feb 04 23:21:00 benjamina-pc kernel: start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
Feb 04 23:21:00 benjamina-pc kernel: RTW: nolinked power save leave
Feb 04 23:21:03 benjamina-pc NetworkManager[714]: <info>  [1675549263.7013] device (wlx3460f954564b): supplicant interface state: scanning -> disconnected
Feb 04 23:21:03 benjamina-pc NetworkManager[714]: <info>  [1675549263.7014] device (p2p-dev-wlx3460f954564b): supplicant management interface state: scanning -> disconnected
Feb 04 23:21:03 benjamina-pc kernel: RTW: nolinked power save enter
Feb 04 23:21:04 benjamina-pc kernel: RTW: [HALMAC][ALWAYS]shall R reg twice!!
Feb 04 23:21:04 benjamina-pc kernel: start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
Feb 04 23:21:04 benjamina-pc NetworkManager[714]: <info>  [1675549264.9703] device (wlx3460f954564b): supplicant interface state: disconnected -> scanning
Feb 04 23:21:04 benjamina-pc NetworkManager[714]: <info>  [1675549264.9704] device (p2p-dev-wlx3460f954564b): supplicant management interface state: disconnected -> scanning
Feb 04 23:21:04 benjamina-pc kernel: RTW: nolinked power save leave
Feb 04 23:21:08 benjamina-pc wpa_supplicant[747]: wlx3460f954564b: Trying to associate with 76:a7:41:6d:ee:98 (SSID='mikeetechguest' freq=2437 MHz)
Feb 04 23:21:08 benjamina-pc kernel: RTW: rtw_set_802_11_connect(wlx3460f954564b)  fw_state=0x00000008
Feb 04 23:21:08 benjamina-pc NetworkManager[714]: <info>  [1675549268.7668] device (wlx3460f954564b): supplicant interface state: scanning -> associating
Feb 04 23:21:08 benjamina-pc NetworkManager[714]: <info>  [1675549268.7669] device (p2p-dev-wlx3460f954564b): supplicant management interface state: scanning -> associating
Feb 04 23:21:08 benjamina-pc wpa_supplicant[747]: wlx3460f954564b: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=2437 ht_enabled=1 ch_offset=0 ch_width=20 MHz cf1=2437 cf2=0
Feb 04 23:21:08 benjamina-pc kernel: RTW: start auth
Feb 04 23:21:08 benjamina-pc kernel: RTW: auth success, start assoc
Feb 04 23:21:08 benjamina-pc wpa_supplicant[747]: wlx3460f954564b: CTRL-EVENT-ASSOC-REJECT status_code=30
Feb 04 23:21:08 benjamina-pc wpa_supplicant[747]: BSSID 76:a7:41:6d:ee:98 ignore list count incremented to 2, ignoring for 10 seconds
[... cut ...]
Feb 04 23:22:57 benjamina-pc dbus-daemon[713]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.9' (uid=0 pid=714 comm="/usr/sbin/NetworkManager --no-daemon " label="unconfined")
Feb 04 23:22:57 benjamina-pc systemd[1]: Starting Network Manager Script Dispatcher Service...
Feb 04 23:22:57 benjamina-pc dbus-daemon[713]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Feb 04 23:22:57 benjamina-pc systemd[1]: Started Network Manager Script Dispatcher Service.
Feb 04 23:23:07 benjamina-pc systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.

RinCat, 2023-02-05:
From the logs, it still scanning every 2 mins. Maybe you can try use iwd https://wiki.debian.org/NetworkManager/iwd

RinCat, 2023-06-15:
Hi, someone has found a possible cause of the problem, if you are still using this, please update the driver to see if it is fixed. 
https://github.com/RinCat/RTL88x2BU-Linux-Driver/issues/198