RinCat/RTL88x2BU-Linux-Driver #177: TP-Link Archer T4U AC1300 MU-MIMO USB adapter failed to connect

Apisteftos, 2023-02-04:
The authentication keeps failing. I installed properly the files except from the `sudo dkms autoinstall` an error occurs.

`Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j12 KVER=5.15.0-58-generic KSRC=/lib/modules/5.15.0-58-generic/build............
Signing module:
Generating a new Secure Boot signing key:
Can't load /var/lib/shim-signed/mok/.rnd into RNG

 - /var/lib/dkms/rtl88x2bu/git/5.15.0-58-generic/x86_64/module/88x2bu.ko
Secure Boot not enabled on this system.
cleaning build area...

88x2bu.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/5.15.0-58-generic/updates/dkms/

depmod...`

What I am doing wrong here? The secure boot is disabled because I had issues on windows.

RinCat, 2023-02-04:
Hi, you need give some logs.

Fisrt check if the driver is loadable by run `sudo modprobe 88x2bu`.
then check the output of `dmesg`.

Apisteftos, 2023-02-04:
Yes the driver is loaded under `sudo modprobe 88x2bu`, here is the list with `lsmod` 

>
Module                  Size  Used by
ntfs3                 274432  0
rfcomm                 81920  4
cmac                   16384  3
algif_hash             16384  1
algif_skcipher         16384  1
af_alg                 32768  6 algif_hash,algif_skcipher
intel_rapl_msr         20480  0
intel_rapl_common      40960  1 intel_rapl_msr
snd_hda_codec_realtek   159744  1
bnep                   28672  2
snd_hda_codec_generic   102400  1 snd_hda_codec_realtek
edac_mce_amd           36864  0
ledtrig_audio          16384  1 snd_hda_codec_generic
snd_hda_codec_hdmi     77824  1
kvm_amd               155648  0
snd_hda_intel          53248  3
snd_intel_dspcfg       28672  1 snd_hda_intel
snd_intel_sdw_acpi     20480  1 snd_intel_dspcfg
amdgpu               9867264  22
kvm                  1028096  1 kvm_amd
snd_hda_codec         163840  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
snd_usb_audio         356352  1
binfmt_misc            24576  1
snd_usbmidi_lib        45056  1 snd_usb_audio
snd_hda_core          110592  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
snd_hwdep              16384  2 snd_usb_audio,snd_hda_codec
crct10dif_pclmul       16384  1
iommu_v2               24576  1 amdgpu
ghash_clmulni_intel    16384  0
snd_pcm               143360  5 snd_hda_codec_hdmi,snd_hda_intel,snd_usb_audio,snd_hda_codec,snd_hda_core
gpu_sched              45056  1 amdgpu
drm_ttm_helper         16384  1 amdgpu
aesni_intel           376832  4
ttm                    86016  2 amdgpu,drm_ttm_helper
snd_seq_midi           20480  0
snd_seq_midi_event     16384  1 snd_seq_midi
crypto_simd            16384  1 aesni_intel
drm_kms_helper        311296  1 amdgpu
cryptd                 24576  3 crypto_simd,ghash_clmulni_intel
snd_rawmidi            49152  2 snd_seq_midi,snd_usbmidi_lib
88x2bu               3657728  0
uvcvideo              106496  0
cec                    61440  1 drm_kms_helper
snd_seq                77824  2 snd_seq_midi,snd_seq_midi_event
videobuf2_vmalloc      20480  1 uvcvideo
rc_core                65536  1 cec
btusb                  61440  0
videobuf2_memops       20480  1 videobuf2_vmalloc
btrtl                  24576  1 btusb
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
videobuf2_v4l2         32768  1 uvcvideo
btbcm                  24576  1 btusb
i2c_algo_bit           16384  1 amdgpu
nls_iso8859_1          16384  1
rapl                   20480  0
wmi_bmof               16384  0
snd_timer              40960  2 snd_seq,snd_pcm
videobuf2_common       77824  4 videobuf2_vmalloc,videobuf2_v4l2,uvcvideo,videobuf2_memops
btintel                40960  1 btusb
fb_sys_fops            16384  1 drm_kms_helper
syscopyarea            16384  1 drm_kms_helper
videodev              258048  3 videobuf2_v4l2,uvcvideo,videobuf2_common
bluetooth             704512  33 btrtl,btintel,btbcm,bnep,btusb,rfcomm
snd                   106496  21 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_usb_audio,snd_usbmidi_lib,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_pcm,snd_rawmidi
sysfillrect            20480  1 drm_kms_helper
ecdh_generic           16384  2 bluetooth
[... 177 lines cut ...]
[  916.751773] usb 3-2: reset high-speed USB device number 2 using xhci_hcd
[  921.151792] usb 3-2: reset high-speed USB device number 2 using xhci_hcd
[  933.567798] usb 3-2: reset high-speed USB device number 2 using xhci_hcd
[  985.002665] RTW: WARN cfg80211_rtw_scan (wlxd03745a447ab) : scan abort!! BusyTraffic
[ 1036.472263] usb 1-3: USB disconnect, device number 2
[ 1039.936593] usb 1-3: new full-speed USB device number 6 using xhci_hcd
[ 1040.184180] usb 1-3: New USB device found, idVendor=0bda, idProduct=8771, bcdDevice= 2.00
[ 1040.184189] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1040.184217] usb 1-3: Product: Bluetooth 5.1 Radio
[ 1040.184220] usb 1-3: Manufacturer: Realtek
[ 1040.184223] usb 1-3: SerialNumber: 00E04C239987
[ 1040.202056] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[ 1040.204919] Bluetooth: hci0: RTL: rom_version status=0 version=1
[ 1040.204927] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_fw.bin
[ 1040.205013] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_config.bin
[ 1040.205051] Bluetooth: hci0: RTL: cfg_sz 6, total sz 27814
[ 1040.898097] Bluetooth: hci0: RTL: fw version 0x09a98a6b
[ 1127.343944] RTW: recv eapol packet - WPA Group Key 1/2
[ 1127.344553] RTW: set group key camid:2, addr:3c:a6:2f:45:c8:59, kid:2, type:AES
[ 1127.344560] RTW: send eapol packet - WPA Group Key 2/2
[ 1284.946916] RTW: WARN cfg80211_rtw_scan (wlxd03745a447ab) : scan abort!! BusyTraffic
[ 1303.650766] RTW: WARN cfg80211_rtw_scan (wlxd03745a447ab) : scan abort!! BusyTraffic
[ 1307.264441] Bluetooth: hci0: command 0x0c3a tx timeout
[ 1309.280384] Bluetooth: hci0: command 0x0c1a tx timeout
[ 1425.230103] input: GHS-700 BT (AVRCP) as /devices/virtual/input/input26

RinCat, 2023-02-04:
No, you do not need to `sudo modprobe 88x2bu` every time, it should be loaded automatically.

Are you trying to connect to the SSID that had multiple APs?

Apisteftos, 2023-02-04:
I have a WLAN, one works in 2.4GHz and one in 5GHz with different SSIDs, because I had the same problem in Windows and I couldn't connect with 5GHz channel. The same here I am able to connect only with 2.4GHz. The SSIDs are different for both channels.

RinCat, 2023-02-04:
Which country do you live in? Sounds like channel incompatible.

Apisteftos, 2023-02-04:
I am living in Germany. What do you mean incompatible? My router channel is incompatible? I suppose not, because router works in both frequencies, 2.4GHz and 5GHz.

RinCat, 2023-02-04:
The WiFi channels allowed to be used in different countries are very different, see https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_(802.11a/h/j/n/ac/ax)

The device usually refuses to connect if the country setting does not match.

Apisteftos, 2023-02-04:
So, I have to check the WiFi channel and in which frequency is installed. And how do I know which channel and frequency works the usb adapter?

RinCat, 2023-02-04:
The usb adapter should works for most of the channels. However, the default configuration is the worldwide configuration AKA most restrictive configuration. You may wanna use the channel 32 and 149 for the safest.

Apisteftos, 2023-02-07:
No it's not matter of channel automatically choice. I change it manual on channel 9 for 2.4GHz and on channel 44 for 5GHz.

RinCat, 2023-02-08:
You can just remove the driver file, usually in `/lib/modules/XXXXXX-x86_64/kernel/drivers/net/wireless/`. And the filename is `88x2bu.ko` or with extra extension. 
Or if you use DKMS, use `sudo dkms remove rtl88x2bu/git --all`

However, you don't need to uninstall to try it. They are not bind to the driver, but depends on wifi control software.

Apisteftos, 2023-02-09:
I don't find any compiled file `88x2bu.ko` into the kernel directory `/lib/modules/5.15.0-60-generic/kernel/net/wireless/ ` .I did it manually `sudo cp 88x2bu.ko /lib/modules/5.15.0-60-generic/kernel/drivers/net/wireless/` and then with the command: `sudo depmod – a`.

RinCat, 2023-02-11:
That just `make install` did.

nikallian, 2023-02-11:
Hi Apisteftos

please give us the output of `sudo iw reg get` and check if it matches the country setup of your wifi router. Germany has very specific regulations about 5GHz band. Have you bought the router in the Germany or from elsewhere ?

The installation of this driver is very straight forward it must work right away. I do have the same adapter and I use it as AP in a raspberry Pi.

Apisteftos, 2023-02-12:
As I said I am connected to 2.4GHz. I can not connect to 5GHz and the signal ist about 80% while with 2.4GHz the strength is 90%. 

`global
country DE: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)
`

nikallian, 2023-02-12:
Have you made any configuration in the driver module under `/etc/modprobe.d/` ? Please check if there is any file named 88x2bu.conf. Also you can use wpa_supplicant in the command line to check if you can connect to the 5GHz band. 

You can also debug your connection using: 

`wpa_supplicant -i wlan0 -c /etc/wpa_supplicant/example5GHz.conf -d`

Apisteftos, 2023-02-14:
Let me take it from the beginning. I downloaded the wpasupplicant `sudo apt install wpasupplicant`. Constructed my 
   ` wpa_passphrase` with ` wpa_passphrase my-ESSID my-passphrase | sudo tee /etc/wpa_supplicant/example5GHz.conf ` and one more for 2.4GHz with ` wpa_passphrase my-ESSID my-passphrase | sudo tee /etc/wpa_supplicant/example24GHz.conf ` 

Two files created 

<pre>
network={
        ssid="my_SSID"
        #psk="my_passphrase"
        psk=............................................................................
}

</pre>

<pre>
network={
        ssid="my_SSID 2,4GHz"
        #psk="my_passphrase"
        psk=............................................................................
}

</pre>

The next step was the debugging the wifi connection with ` sudo wpa_supplicant -i my_wlan_interface -D wext -c /etc/wpa_supplicant/example24GHz.conf -d `

interesting points of the debugging

<pre>
TDLS: TDLS operation not supported by driver
TDLS: Driver uses internal link setup
TDLS: Driver does not support TDLS channel switching

EAPOL: SUPP_PAE entering state DISCONNECTED
EAPOL: Supplicant port status: Unauthorized

nl80211: Skip set_supp_port(unauthorized) while not associated
EAPOL: KEY_RX entering state NO_KEY_RECEIVE
EAPOL: SUPP_BE entering state INITIALIZE
EAP: EAP entering state DISABLED

nl80211: Drv Event 33 (NL80211_CMD_TRIGGER_SCAN) received for wlxd03745a447ab
nl80211: Scan timeout - try to abort it
nl80211: Abort scan
nl80211: Abort scan failed: ret=-95 (Operation not supported)
nl80211: Failed to abort scan
nl80211: Try to get scan results

RSN: PMKSA cache search - network_ctx=0x55873624a970 try_opportunistic=0 akmp=0x0
RSN: Search for BSSID 3c:a6:2f:45:c8:5a      is the mac address for 5GHz
RSN: No PMKSA cache entry found

P2P: channels: 81:1,2,3,4,5,6,7,8,9,10,11 115:36,40,44,48 116:36,44 117:40,48 128:36,40,44,48 130:36,40,44,48
wpa_driver_nl80211_disconnect(reason_code=3)

wlxd03745a447ab: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="LANwire 2,4Ghz" auth_failures=3 duration=30 reason=CONN_FAILED
wlxd03745a447ab: CTRL-EVENT-ASSOC-REJECT status_code=1
BSSID 3c:a6:2f:45:c8:59 ignore list count incremented to 3, ignoring for 60 seconds
wlxd03745a447ab: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="LANwire 2,4Ghz" auth_failures=4 duration=73 reason=CONN_FAILED
wlxd03745a447ab: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlxd03745a447ab: CTRL-EVENT-DISCONNECTED bssid=3c:a6:2f:45:c8:59 reason=3 locally_generated=1
BSSID 3c:a6:2f:45:c8:59 ignore list count incremented to 4, ignoring for 120 seconds
wlxd03745a447ab: CTRL-EVENT-DSCP-POLICY clear_all
wlxd03745a447ab: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlxd03745a447ab: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=DE

</pre>

Then I changed the configuration of both files because the encryption is ` WPA-CCMP ` of my router with ` sudo nano /etc/wpa_supplicant/example5GHz.conf ` to

<pre>
ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=netdev

network={
	ssid="my_SSID"
        #psk="my_passphrase"
        psk=............................................................................
	key_mgmt=WPA-PSK
        proto=RSN
        pairwise=CCMP
        group=CCMP
}

I think I have to disable the PMKSA but I don't know how exactly.

nikallian, 2023-02-15:
I would say to disable completely the security of your router and check again if you can connect. If your router has been set to offer WPA3 please disable it and force it to WPA2 only.

Apisteftos, 2023-02-15:
Doesn't even work without encryption or password. It's says connection failed. Just a file from the debugging with 2,4GHz channel. 

[debugging_wifi_2,4_GHz.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/10743836/debugging_wifi_2.4_GHz.txt)

RinCat, 2023-02-15:
Try disable WiFi MAC randomization.
https://wiki.archlinux.org/title/NetworkManager#Configuring_MAC_address_randomization

Apisteftos, 2023-02-15:
It was disabled but I created one more file in `/etc/NetworkManager/conf.d/wifi_rand_mac.conf`

dipol0, 2023-02-17:
You  can try check supported channels run

_iw phy_

i can connect ot my route with addition options on module load
OS - Ubunru 22.04

1. sudo touch /etc/modprobe.d/88x2bu.conf
2. add to 88x2bu.conf:

_options 88x2bu rtw_led_ctrl=1 rtw_power_mgnt=0 rtw_wireless_mode=84 rtw_channel=48_

3. run: sudo update-initramfs -u -k $(uname -r)
4. reboot

or first can try:

rmmod 88x2bu
modprobe 88x2bu rtw_led_ctrl=1 rtw_power_mgnt=0 rtw_wireless_mode=84 rtw_channel=48

*******
rtw_wireless_mode=84 - use only 5GHz
rtw_channel=48 - 5GHz 48 ch = 5240MHz (i set in configuration om my Mikrotik router)
*******

You can try play with another module parameters like as (from - https://github.com/morrownr/88x2bu-20210702/blob/main/88x2bu.conf):

# DFS Options ( rtw_dfs_region_domain )
#
# 0 = NONE (default)
# 1 = FCC
# 2 = MKK
# 3 = ETSI

# Country Code options ( rtw_country_code )
#
# Note: Allows the Country Code to be set in cases where it is unable to
# be obtained otherwise.
#
# URL: http://en.wikipedia.org/wiki/ISO_3166-1_alpha-2
#
# Example for the US: rtw_country_code=US
# Example for Panama: rtw_country_code=PA
# Example for Norway: rtw_country_code=NO
# Example for Kuwait: rtw_country_code=KW
# Example for Taiwan: rtw_country_code=TW

Apisteftos, 2023-02-19:
Yes it works now thanks @dipol0 !

So I did a lot of experiments with the configuration and it works also without any configuration but it need little bit more time. I suppose the problem was the separate SSIDs for `HomeLan 2,4GHz`  and  `HomeLan`. The char code of the SSIDs was faulty interpreted. It must be somewhere a bug in the code, maybe ` comma` of the `2,4 GHz` or the whitespace in `HomeLan 2,4GHz` .

Works:
===================

1. No configuration (but without separate SSIDs one SSID for both, channel automatic, encryption WPA2(CCMP) or WPA-WPA2) (disadvantage: takes more time to connect) 
2. options 88x2bu rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_wireless_mode=95 rtw_country_code=DE
3. options 88x2bu rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_wireless_mode=95 rtw_country_code=DE rtw_channel=52
4. options 88x2bu rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_wireless_mode=95 rtw_country_code=DE
5. options 88x2bu rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_wireless_mode=84 rtw_country_code=DE rtw_channel=52  ( 5GHz)

Doesn't Work
=================

1. options 88x2bu rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_wireless_mode=95 rtw_country_code=DE rtw_channel=6 rtw_channel=52
2. options 88x2bu rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_wireless_mode=95 rtw_country_code=DE rtw_channel=6,52
3. Separate SSIDs with same name or whitespace between the name.

dipol0, 2023-02-20:
@Apisteftos good.

[If after you be have problem with upload/download big size file size (500mb and more) over Archer try set **rtw_pwrtrim_enable=1**](https://github.com/RinCat/RTL88x2BU-Linux-Driver/issues/179#issuecomment-1436770110)

Apisteftos, 2023-02-21:
The problem with me is that the adapter doesn't connect at once, but it connect at the second try. To connect the second try, I have to do it manually. Maybe it's signal strength of the 5GHz, it's about 60%, but with 2,4GHz it's 90%.

dipol0, 2023-02-21:
@Apisteftos 
In my systmem i disable netplan and back to /etc/network/interfaces + wpa_supplicant.
Check frequenciqes supported you Archer:

#iw phy

in my case work range 5180....5700 MHz (if reg-domain=none (in mikrotik router - superchannel)):

                Frequencies:
                        *** 5180 MHz [36] (20.0 dBm)
                        * 5200 MHz [40] (20.0 dBm)
                        * 5220 MHz [44] (20.0 dBm)
                        * 5240 MHz [48] (20.0 dBm)
                        * 5260 MHz [52] (20.0 dBm) (radar detection)
                        * 5280 MHz [56] (20.0 dBm) (radar detection)
                        * 5300 MHz [60] (20.0 dBm) (radar detection)
                        * 5320 MHz [64] (20.0 dBm) (radar detection)
                        * 5500 MHz [100] (20.0 dBm) (radar detection)
                        * 5520 MHz [104] (20.0 dBm) (radar detection)
                        * 5540 MHz [108] (20.0 dBm) (radar detection)
                        * 5560 MHz [112] (20.0 dBm) (radar detection)
                        * 5580 MHz [116] (20.0 dBm) (radar detection)
                        * 5600 MHz [120] (20.0 dBm) (radar detection)
                        * 5620 MHz [124] (20.0 dBm) (radar detection)
                        * 5640 MHz [128] (20.0 dBm) (radar detection)
                        * 5660 MHz [132] (20.0 dBm) (radar detection)
                        * 5680 MHz [136] (20.0 dBm) (radar detection)
                        * 5700 MHz [140] (20.0 dBm) (radar detection)**
                        * 5720 MHz [144] (disabled)
                        * 5745 MHz [149] (disabled)
                        * 5765 MHz [153] (disabled)
                        * 5785 MHz [157] (disabled)
                        * 5805 MHz [161] (disabled)
                        * 5825 MHz [165] (disabled)
                        * 5845 MHz [169] (disabled)
                        * 5865 MHz [173] (disabled)
                        * 5885 MHz [177] (disabled)

Select freq and set it fixed on your router - for examle select 5240GHz - 48ch. If can select REG_DOMAIN - superchanel/none.

Load module with options (superchannel use - no regdomain):
rtw_dfs_region_domain=0 rtw_country_code=<YOU_COUNTRY> rtw_channel=48

in /etc/wpa_supplicant/wpa_supplicant.conf (HIDDEN+WPA2-PSK)
use bssid you 5GHz wifi you router (MAC 5GHz wifi interface on router). For example:

where BSSID - router MAC on 5ghz interfce

`
ctrl_interface_group=0
ctrl_interface=/var/run/wpa_supplicant
update_config=1
ap_scan=1
fast_reauth=1
country=<YOU_COUNTRY_CODE>
network={
    ssid="YOUNET5G"
    bssid=88:a5:c1:5f:7b:47
    scan_ssid=1
    key_mgmt=WPA-PSK
    pairwise=CCMP
    proto=RSN
    group=CCMP
    psk=<you_psk_key>
 }
`
/etc/network/interfaces:

where wlx4460f95c4250 - you wifi interface. check ifconfig/iwconfig

```
auto lo
iface lo inet loopback

auto wlx4460f95c4250
allow-hotplug wlx4460f95c4250
iface wlx4460f95c4250 inet dhcp
     wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

```
reload module with need options manually and restart networking service ot update-initramfs for  you kernel (set need options in module config) and reboot.

Apisteftos, 2023-02-21:
After `iw phy` frequencies are the same as yours.

<pre>
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CMAC (00-0f-ac:6)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CCMP-256 (00-0f-ac:10)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
		* CMAC-256 (00-0f-ac:13)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 2:
		Capabilities: 0x863
			RX LDPC
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15
		VHT Capabilities (0x03d071b2):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
			short GI (80 MHz)
			TX STBC
			SU Beamformee
			MU Beamformee
			+HTC-VHT
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 867 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 867 Mbps
		Bitrates (non-HT):
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps

WoWLAN support:
		 * wake up on anything (device continues operating normally)
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
	Device supports SAE with AUTHENTICATE command
	Device supports scan flush.
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0

</pre>

Apisteftos, 2023-02-23:
It works ok right, now, I will close the issue. I hope doesn't come back again.

closed 2023-02-23 by Apisteftos